espressif/esp-matter #145: Google Home connection issue

eos1d3, 2022-12-31:
IDF 4.4.2 / 4.4.3

Target: ESP32C3/ESP32

Google Nest Hub2: 
    System firmware version:324896
    Cast firmware: 1.56.324896

Google Home 2.62.1.15	
Google Home 2.62.80.2
Google Home 2.63.5.1-dogfood

It works with Amazon Echo 4 and can be paired and control.  But none of any combination work with Google Home. Google Home displays:

```
Connecting Device to Google Home
...
Something went Wrong
Make sure you're connected to your network and try again
```

log from ESP32C3 (from startup to Google Home pairing)

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x16b4
load:0x403cc710,len:0x930
load:0x403ce710,len:0x2d28
entry 0x403cc710
[0;32mI (30) boot: ESP-IDF v4.4.3 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 02:56:34[0m
[0;32mI (30) boot: chip revision: 3[0m
[0;32mI (32) boot.esp32c3: SPI Speed      : 80MHz[0m
[0;32mI (37) boot.esp32c3: SPI Mode       : DIO[0m
[0;32mI (41) boot.esp32c3: SPI Flash Size : 4MB[0m
[0;32mI (46) boot: Enabling RNG early entropy source...[0m
[0;32mI (52) boot: Partition Table:[0m
[0;32mI (55) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (62) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000[0m
[0;32mI (70) boot:  1 nvs              WiFi data        01 02 00010000 00006000[0m
[0;32mI (77) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000[0m
[0;32mI (85) boot:  3 otadata          OTA data         01 00 00017000 00002000[0m
[0;32mI (92) boot:  4 phy_init         RF data          01 01 00019000 00001000[0m
[0;32mI (100) boot:  5 ota_0            OTA app          00 10 00020000 001e0000[0m
[0;32mI (107) boot:  6 ota_1            OTA app          00 11 00200000 001e0000[0m
[0;32mI (115) boot:  7 fctry            WiFi data        01 02 003e0000 00006000[0m
[0;32mI (122) boot: End of partition table[0m
[0;32mI (127) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=2ec08h (191496) map[0m
[0;32mI (165) esp_image: segment 1: paddr=0004ec30 vaddr=3fc91c00 size=013e8h (  5096) load[0m
[0;32mI (166) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=f5c9ch (1006748) map[0m
[0;32mI (328) esp_image: segment 3: paddr=00145cc4 vaddr=3fc92fe8 size=02328h (  9000) load[0m
[0;32mI (330) esp_image: segment 4: paddr=00147ff4 vaddr=40380000 size=11a9ch ( 72348) load[0m
[0;32mI (347) esp_image: segment 5: paddr=00159a98 vaddr=50000010 size=00010h (    16) load[0m
[0;32mI (353) boot: Loaded app from partition at offset 0x20000[0m
[0;32mI (353) boot: Disabling RNG early entropy source...[0m
[0;32mI (368) cpu_start: Pro cpu up.[0m
[0;32mI (376) cpu_start: Pro cpu start user code[0m
[0;32mI (376) cpu_start: cpu freq: 160000000[0m
[0;32mI (376) cpu_start: Application information:[0m
[0;32mI (379) cpu_start: Project name:     light[0m
[0;32mI (384) cpu_start: App version:      v1.0[0m
[0;32mI (388) cpu_start: Compile time:     Dec 31 2022 02:56:25[0m
[0;32mI (394) cpu_start: ELF file SHA256:  dd84cad64cc3fb0f...[0m
[0;32mI (400) cpu_start: ESP-IDF:          v4.4.3[0m
[0;32mI (405) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (413) heap_init: At 3FCA5FF0 len 00036720 (217 KiB): DRAM[0m
[... 450 lines cut ...]
[0;32mI (191215) chip[TS]: Previous Last Known Good Time: 2022-12-31T02:57:26[0m
[0;32mI (191225) chip[TS]: Reverted Last Known Good Time to previous value[0m
[0;32mI (191235) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is 0 **********[0m
[0;32mI (191255) app_main: Commissioning failed, fail safe timer expired[0m
[0;31mE (191255) chip[DL]: Long dispatch time: 113 ms, for event type 32782[0m
[0;32mI (191265) chip[FS]: Fail-safe cleanly disarmed[0m
[0;32mI (191265) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)[0m
[0;32mI (191275) NimBLE: GAP procedure initiated: advertise; [0m
[0;32mI (191275) NimBLE: disc_mode=2[0m
[0;32mI (191275) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40[0m
[0;32mI (191285) NimBLE: 
[0m
[0;32mI (191305) chip[DL]: CHIPoBLE advertising started[0m
[0;32mI (191305) app_main: Commissioning window opened[0m
[0;32mI (221165) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)[0m
[0;32mI (221165) chip[DL]: Device already advertising, stop active advertisement and restart[0m
[0;32mI (221175) NimBLE: GAP procedure initiated: stop advertising.
[0m
[0;32mI (221185) NimBLE: GAP procedure initiated: advertise; [0m
[0;32mI (221195) NimBLE: disc_mode=2[0m
[0;32mI (221195) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800[0m
[0;32mI (221205) NimBLE: 
[0m

```

jonsmirl, 2022-12-31:
Google Home is currently broken from the Google side of things for all of the test device IDs.  Everyone is waiting for them to fix it.

eos1d3, 2022-12-31:
Oh... Thanks a lot for the information.

BTW, where to know when it is fixed?

jonsmirl, 2022-12-31:
watch for an update on your Google Home, the version number will change

dhrishi, 2023-01-05:
@eos1d3 @jonsmirl WIth the latest update from Google, commissioning is again successful with test devices. Please update your hub and the app and give it a try.

eos1d3, 2023-01-05:
@dhrishi Do you mean Google Home 2.62.80.2? In my area, I am still seeing Google Home 2.62.1.9 in Play Store. My Nest Hub 2 firmware is 324896. Is it OK?

dhrishi, 2023-01-05:
@eos1d3 Google Home App: 2.62.1.15 , Nest Hub: 1.56.324896, these versions work at our end. Although, the fix is on the backend side as per Google, and hence you should be able to commission successfully too (without any update). Let me know if that doesn't happen

eos1d3, 2023-01-05:
@dhrishi 

There is a good news. Google Home 2.62.1.15 finally works. So both Google Home and Alexa are working.

And I just deleted the device in Alexa and using Google Home to share the device to it, it works too! It is very nice.

And there can be a small bug in Google Home. When using Google Home, I have 2.4G and 5G enabled with different SSID names. My mobile phone is using 5G, I saw console saying it could not connect to 5G network. Of course it is normal. But when using Alaxa to pair, it is smarter and will prompt which SSID to use. I think it is Google Home's bug?

jonsmirl, 2023-01-05:
Do you have the 2.4G Network SSID/PW saved in Android?

eos1d3, 2023-01-05:
Yes. It has both sets of passwords. Once I got the problem, I changed my mobile phone to 2.4G and it works. It is strange that it should ask which SSID to use.

dhrishi, 2023-01-06:
> And there can be a small bug in Google Home. When using Google Home, I have 2.4G and 5G enabled with different SSID names. My mobile phone is using 5G, I saw console saying it could not connect to [...]

We will check this with Google.

Thanks for the confirmation. I will close this issue as the original problem is now solved.

closed 2023-01-06 by dhrishi