espressif/esp-matter #146: Pressure cluster

slompf18, 2022-12-31:
I tried to implement a pressure sensor endpoint by forking this repository. But I'm ending in Homekit saying the device is not supported. Are pressure sensors not supported in Homekit or am I missing something?

You can have a look [here](https://github.com/slompf18/esp-matter-pressure). Essentially I copied endpoint, cluster and attributes from the temperature sensor. It has the same mandatory attributes. The app I'm testing with is working fine with a temperature sensor.

jadhavrohit924, 2023-01-02:
Pressure sensor is not supproted in HomeKit.

slompf18, 2023-01-02:
Thank you for the information. Is there an overview, what is supported?

jadhavrohit924, 2023-01-03:
Apple supported devices [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#testing-your-matter-device-with-apple-home)

closed 2023-01-03 by slompf18