espressif/esp-matter #147: Can not change Passcode

a-tobias, 2023-01-02:
I`am trying to get the example light project to work with a different passcode, discriminator and hwid ... .
Basically I got it to work by using the mfg_tool (set discriminator, hwid, passcode options). Afterwards I flashed the ...-partition.bin at  address 0x10000  (nvs partition).

In sdkconfig : 
Use ESP32 Factory Data Provider (selected)
Use ESP32 Device Instance Info Provider (selected)
Chip-factory namespace : nvs

In the matter cli, when I run matter config, I can see that everything has changed according to my new settings (qrcode also) - except the pincode.
A matter config pincode results in -> Error: 45

matter config output : 
> matter config
VendorId:        65522 (0xFFF2)
ProductId:       32769 (0x8001)
HardwareVersion: 3 (0x3)
Error: 45

To create the partition with mfg_tool I used the following : 
./mfg_tool.py -cn "ESP32 Device" -v 0xFFF2 -p 0x8001 --hw-ver 0x03 --pai  \
    -k $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
    -c $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
    -cd $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der

About a tip or help I would be pleased

dhairyashah1, 2023-01-04:
Hello @a-tobias 
1. I re-created your issue on an ESP32C3 as an matter light end-device. Changes made in the app menuconfig - 
- Enabled **ESP32 Factory Data Provider** [Component config → CHIP Device Layer → Commissioning options → Use ESP32 Factory Data Provider]
- Enabled **ESP32 Device Instance Info Provider** [Component config → CHIP Device Layer → Commissioning options → Use ESP32 Device Instance Info Provider]
- Enabled **Attestation - Factory** [ Component config → ESP Matter → DAC Provider options → Attestation - Factory] { Since we are generating a custom partition using mfg_tool it needs to use proper DAC certificates } 

2. I used the same command as yours to create the partition using `mfg_tool.`

3. The updated matter config is visible in `$ESP_MATTER_PATH/tools/mfg_tool/out/<vid_pid>/<UUID>/internal/partition.csv`
for eg - 
[image]

4. The update qrcode, manualcode, discriminator, passcode/ pincode is visible in ( Use the same values for commissioning) `$ESP_MATTER_PATH/tools/mfg_tool/out/<vid_pid>/<UUID>/<UUID>-onb_codes.csv`
for eg -
[image]

5. Matter cli console output for this case - 
[image]

5. However, the **unique passcode/pincode** for production considerations is not directly known to the end device (matter light here) - [Ref_link](https://docs.espressif.com/projects/esp-matter/en/main/esp32/production.html) which says =>
`The unique setup passcode serves as the proof of possession and is also used to compute the shared secret during commissioning. The corresponding SPAKE2+ verifier of the passcode is installed on the device and not the actual passcode.` The default passcode - `20202021` is known to the device (hardcoded) for test purposes, which is not the case for custom production considerations.

dhrishi, 2023-01-05:
@a-tobias Even though, you do not see the value through the CLI command, please try commissioning using the new values and ideally it should work

a-tobias, 2023-01-05:
I additionally selected "Enabled Attestation - Factory" - I didn`t do that before.
Furthermore I had used the wrong QR code (from CLI - which is obviously the wrong one) .
Now if I use the QR-Code from ...fff2_8001/7d9c5f2c-5837-4666-bd85-dcd30d39ca32/...png  commissioning is working as expected.

The Errors in the CLi has been a little bit confusing. Thank you for the discussion and help. I close the issue.

closed 2023-01-05 by a-tobias