espressif/esp-matter #148: device name

stefano055415, 2023-01-03:
hi, how can i assign a default name to a device added to a bridge?

jadhavrohit924, 2023-01-04:
Matter bridge gives name to bridged nodes as `NODE-1, NODE-2, etc`. Do you want to set these names in matter bridge. Or you want to set name to non-devices?

For second case:-
Non-matter devices should have bridge-basic-information-cluster.

stefano055415, 2023-01-04:
I would like to assign basic information to the bridge

    node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, NULL);
    endpoint_t *aggregator = endpoint::aggregator::create(node, ENDPOINT_FLAG_NONE, NULL);
    esp_matter::start(app_event_cb);
    esp_matter_bridge::initialize(node);

and I would like to name the lights associated with the bridge

esp_matter_bridge::create_device(node, endpoint::get_id(aggregator), ESP_MATTER_EXTENDED_COLOR_LIGHT_DEVICE_TYPE_ID);

jadhavrohit924, 2023-01-05:
You will have to create node_label attribute while creating cluster [here](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.cpp#L799). 

Then use command to set node_label
`./out/debug/chip-tool bridgeddevicebasic write node-label "Light" <node-id> <enspoint-id>`

stefano055415, 2023-01-05:
The problem is that I have a dynamic bridge
The user can create or remove devices via a web interface and I would like the bridge to create a new device with a default name already

stefano055415, 2023-01-05:
when i create a new device i can give it a default name and it works with this code:

cluster_t *bridged_device_basic = cluster::get(endpoint, BridgedDeviceBasic::Id);
cluster::bridged_device_basic::attribute::create_bridged_device_basic_node_label(bridged_device_basic, name, strlen(name));

but how do i update it?

I tried with this code but it doesn't work

attribute::update(endpoint_id, BridgedDeviceBasic::Id,BridgedDeviceBasic::Attributes::NodeLabel::Id,&esp_matter_char_str(name, strlen(name)));

stefano055415, 2023-01-27:
@dhrishi 

do you have a solution to my problem?

stefano055415, 2023-01-27:
@dhrishi @jadhavrohit924 

When I use the function: create_bridged_device_basic_node_label 

the name is saved in the nvs storage and I can read it and everything works.

If I restart the device the data is no longer present in the nvs storage and therefore it is null

Is it possible that on reboot the data will be removed?

dhrishi, 2023-01-27:
@stefano055415 We will check this and update

stefano055415, 2023-01-27:
@dhrishi 
I did some more tests and the data in the storage is present before and after the reboot, but after the reboot the device no longer has the NodeLabel attribute.

if I try to get the attribute with this function return NULL

cluster_t *bridged_device_basic = cluster::get(this->dev->endpoint, BridgedDeviceBasic::Id);
attribute_t *attribute = attribute::get(bridged_device_basic, BridgedDeviceBasic::Attributes::NodeLabel::Id);

stefano055415, 2023-01-27:
@dhrishi 

the problem is in the store_val_in_nvs function 

if the type is ESP_MATTER_VAL_TYPE_CHAR_STRING

nvs_set_blob(handle, attribute_key, current_attribute->val.val.a.b, current_attribute->val.val.a.s);
 
writes only the buffer and not all the data for example type

on the next restart when I try to read the data it no longer knows what type it is

shubhamdp, 2023-01-31:
@stefano055415 Please check [get_val_from_nvs()](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.cpp#L1295), type is not saved because, type will be used from the `_attribute_t` and hence the read would be seamless. Checking why it is unable to read the exact value from NVS.

stefano055415, 2023-01-31:
@shubhamdp 
The problem is in the get_data_from_attr_val function which check the type and returns esp_matter_attr_val_type_t not handled

shubhamdp, 2023-01-31:
@stefano055415 If possible can you please share the patch that you are trying out, may be we can try to reproduce it at our end and fix it?

stefano055415, 2023-02-02:
@shubhamdp 

to replicate the problem just run the bridge example associate two lights and add the following code for each light

cluster_t *bridged_device_basic = cluster::get(endpoint, BridgedDeviceBasic::Id);
cluster::bridged_device_basic::attribute::create_bridged_device_basic_node_label(bridged_device_basic, name, strlen(name));

associate it for example with Apple Home

the two lights will see each other with the correct name, but if I restart the device the lights will lose their name

stefano055415, 2023-02-02:
@shubhamdp 

To temporarily solve the problem I removed

ATTRIBUTE_FLAG_NONVOLATILE

from the create_bridged_device_basic_node_label function

shubhamdp, 2023-02-06:
Hi @stefano055415, we found the root cause of the issue, and you were right, it was because char_string/octet_string/array was treated as an INVALID data type. We will push the fix soon and then there will be no need to remove the `ATTRIBUTE_FLAG_NONVOLATILE`.

closed 2023-02-06 by espressif-bot

dhrishi, 2023-02-06:
@stefano055415 The fix to this issue was synced now. Please give it a try and let us know if it works as expected

stefano055415, 2023-02-06:
@dhrishi 

ok perfect now it works correctly 

thanks