espressif/esp-matter #149: ESP32 RAINMAKER + MATTER CODE EXAMPLE NOT FOUND.

emangler, 2023-01-03:
NOT ABLE TO LOCATE AN ESP32 RAINMAKER + MATTER CODE EXAMPLE. WHEN WILL A ESP32 RAINMAKER + MATTER CODE EXAMPLE BE AVAILABLE?

dhrishi, 2023-01-05:
@emangler RainMaker + Matter example is under review and will be made available soon in the [esp-rainmaker](https://github.com/espressif/esp-rainmaker/) repository

emangler, 2023-01-05:
Thank you very much. I appreciate your quick response. I am looking forward to testing the RainMaker + Matter Example and incorporating it into our products.

dhrishi, 2023-01-25:
@emangler Do you need RainMaker for remote control or device management (OTA, insights etc.) or for both? It will help, if you can provide some details about your usecase.

emangler, 2023-02-06:
Hello. Sorry for the late response. Our company makes Motorized Gate Operators. You can check out our products at http://www.allomatic.netWe need RainMaker for both remote control and device management (OTA, insights etc.)Thank you.

dhrishi, 2023-08-12:
@emangler The RainMaker + Matter example is now available in the esp-rainmaker repository, here: https://github.com/espressif/esp-rainmaker/tree/master/examples/matter

closed 2023-08-12 by dhrishi