espressif/esp-matter #151: Updating matter endpoints

SYESHASWINI, 2023-01-05:
I'm currently working on matter light example, where I have some external button to operate. when I operate from the Apple home app the load connected to the buttons works accordingly. But, I'm unable to update the endpoint on the Apple home app when the button is operated. Can I know how I can make it.

jadhavrohit924, 2023-01-05:
Can you try rebooting device and kill home app and restart?

dhrishi, 2023-01-06:
@SYESHASWINI 

>Can I know how I can make it.

Apple Home subscribes to the attributes by default. So, ideally, any changes that happen locally (button in this case) on the device get reflected. We usually see that happening.

SYESHASWINI, 2023-01-09:
@jadhavrohit924 tried but didn't work.

SYESHASWINI, 2023-01-09:
@dhrishi for operating from button I'm not using any matter stack function, So I feel Its not updating. I have taken separate hardware code and merged it with matter one, then when operating from buttons it is not updating to the mobile.
Do I need to call any function to send report to mobile application as in the matter code sending report is automatically handled by the stack.

dhrishi, 2023-01-09:
@SYESHASWINI You need to update the Matter attribute when there is a change. You can use `attribute::update`.

See [this](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_driver.cpp#L55-L71 ) for reference. Here, we toggle the state of light on the push button on the devkit

SYESHASWINI, 2023-01-09:
@dhrishi tried with this, but unable to call the function or update the attribute from other file.

SYESHASWINI, 2023-01-09:
@dhrishi tried this, but unable to call the function or update the attribute apart from calling through app_driver_button_init().

dhrishi, 2023-01-09:
@SYESHASWINI Why are you unable to call the function? What is the error you are seeing?

SYESHASWINI, 2023-01-11:
@dhrishi when I call  **app_driver_button_init();** from another c file it is showing as **implicit declaration** of function and when I tried adding the header files , showing error with the header files.

SYESHASWINI, 2023-01-27:
@dhrishi The issue is fixed thank you.

SYESHASWINI, 2023-01-27:
I have use the attribute update to send report to the endpoints, and it is sending reports succesfully.

closed 2023-01-27 by SYESHASWINI