espressif/esp-matter #152: Cannot compile using Attestation Factory

shun-dgut, 2023-01-05:
Unable to compile after enabling the following configuration

Log
```
[602/1359] Performing configure step for 'chip_gn'
FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure /home/woan_qrs/Matter_Pro/master/esp-matter/examples/blemesh_bridge/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure 
cd /home/woan_qrs/Matter_Pro/master/esp-matter/examples/blemesh_bridge/build/esp-idf/chip && /home/woan_qrs/Matter_Pro/master/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/pigweed/gn --root=/home/woan_qrs/Matter_Pro/master/esp-matter/connectedhomeip/connectedhomeip/config/esp32 gen --check --fail-on-unused-args /home/woan_qrs/Matter_Pro/master/esp-matter/examples/blemesh_bridge/build/esp-idf/chip && /usr/bin/cmake -E touch /home/woan_qrs/Matter_Pro/master/esp-matter/examples/blemesh_bridge/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure
ERROR at build arg file (use "gn args <out_dir>" to edit):16:54: Build argument has no effect.
chip_use_transitional_commissionable_data_provider = false
                                                     ^----
The variable "chip_use_transitional_commissionable_data_provider" was set as a build argument
but never appeared in a declare_args() block in any buildfile.

To view all possible args, run "gn args --list <out_dir>"
[609/1359] Building C object esp-idf/nghttp/CMakeFiles/__idf_nghttp.dir/nghttp2/lib/nghttp2_session.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
[image]

[image]

shun-dgut, 2023-01-06:
> @shun-dgut We are able to compile by enabling the attestation factory options. Can you please clean and re-compile? Please close the issue if it works

I commented on the following code and it works, I suspect it's a compilation tool compatibility issue?

[image]

shun-dgut, 2023-01-06:
hardware: esp32c3
cmake version 3.22.1
CMake suite maintained and supported by Kitware (kitware.com/cmake).
ninja  version 1.10.1

dhairyashah1, 2023-01-06:
@shun-dgut 
Could you please provide the `commit/revision` of [esp-matter](https://github.com/espressif/esp-matter/), [connectedhomeip](https://github.com/espressif/connectedhomeip/tree/4088a77f557e8571a39338fad51a1d8eb0131d79) and [esp-idf](https://github.com/espressif/esp-idf) version you are using?

dhrishi, 2023-01-23:
I don't think we are able to recreate this issue. Looks like something specific to your setup? Closing.

closed 2023-01-23 by dhrishi