espressif/esp-matter #154: Register New Device Type and Cluster

PhucTa2748, 2023-01-06:
I have a device and its type is not listed in Matter Device Library doc.
How can i add (or define )new device type and cluster funtion  ? should i register it with csa ?
Thanks.

dhrishi, 2023-01-06:
@DonP2748 You can add a custom cluster and its attributes, commands. You can check the RainMaker cluster for the APIs [here](https://github.com/espressif/esp-matter/blob/main/components/esp_matter_rainmaker/esp_matter_rainmaker.cpp#L216) as a reference

Note that, other commissioners won't be able to interact/understand this cluster as this is not a standard cluster. So, you need to have your own controller/commissioner that understands it.

dhrishi, 2023-01-06:
@DonP2748 What is your device type/usecase? Is it like a standard device which is not a part of the specification yet or it's a custom or proprietory feature?

DonP2748, 2023-01-07:
> @DonP2748 You can add a custom cluster and its attributes, commands. You can check the RainMaker cluster for the APIs [here](https://github.com/espressif/esp-matter/blob/main/components/esp_matter_rainmaker/esp_matter_rainmaker.cpp#L216) as a reference

Hello @dhrishi , thanks for your quick reply. Does it  mean that my project must integrate with rainmaker ? Does espressif matter or chip have the same api ?

>@DonP2748 What is your device type/usecase? Is it like a standard device which is not a part of the specification yet or it's a custom or proprietory feature?

yeah, my device type is not common in the world, only available in my country.

dhrishi, 2023-01-10:
> Does it mean that my project must integrate with rainmaker ? Does espressif matter or chip have the same api ?

No. It was just for reference. That will help you with the APIs you can use to create custom data model (clusters, attributes, commands etc.)

DonP2748, 2023-01-12:
> One way to deal with this is to make a composite endpoint with children. The root of this could be as simple as an OnOff cluster and then one of the children is your custom cluster. Doing it this way allows Alexa/Google to still turn the device OnOff. And then your app can have greater control via the custom cluster. Or maybe use a media endpoint for the root cluster. If you use custom only Alexa/Google will have zero control.

That's great idea, i will follow it. Thank you @jonsmirl.

closed 2023-01-12 by DonP2748