espressif/esp-matter #156: BLE GAP connection terminated (con 0 reason 0x213)

WindFromTheWoods, 2023-01-06:
Hello! I'm trying to connect the light-app example to google home, but an unknown error appears during the connection. 
I have a google nest mini as a hub. I'm also from Ukraine, probably this can also interfere somehow, because there is no Google Home support in Ukraine.

````
I (2146) chip[DL]: Starting ESP WiFi layer
I (2156) wifi:mode : sta (fc:f5:c4:0e:c4:e0)
I (2156) wifi:enable tsf
W (2156) wifi:Haven't to connect to a suitable AP now!
I (2156) chip[DL]: Done driving station state, nothing else to do...
W (2176) wifi:Haven't to connect to a suitable AP now!
I (2176) chip[DL]: Done driving station state, nothing else to do...
I (2176) chip[SVR]: SetupQRCode: [MT:Y.K9042C00KA0648G00]
I (2186) chip[SVR]: Copy/paste the below URL in a browser to see the QR Code:
I (2186) chip[SVR]: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3AY.K9042C00KA0648G00
I (2206) chip[SVR]: Manual pairing code: [34970112332]
I (2216) chip[SVR]: Server initializing...
I (2216) chip[TS]: Last Known Good Time: [unknown]
I (2216) chip[TS]: Setting Last Known Good Time to firmware build time 2023-01-03T16:23:00
I (2236) chip[DMG]: AccessControl: initializing
I (2236) chip[DMG]: Examples::AccessControlDelegate::Init
I (2236) chip[DMG]: AccessControl: setting
I (2246) chip[DMG]: DefaultAclStorage: initializing
I (2246) chip[DMG]: DefaultAclStorage: 0 entries loaded
I (2276) chip[ZCL]: Using ZAP configuration...
I (2276) chip[DMG]: AccessControlCluster: initializing
I (2276) chip[ZCL]: Initiating Admin Commissioning cluster.
I (2286) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x0030', EndPoint ID: '0x00', Attribute ID: '0x0000'
I (2296) light-app-callbacks: Unhandled cluster ID: 48
I (2306) light-app-callbacks: Current free heap: 94700

I (2306) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x0004', EndPoint ID: '0x00', Attribute ID: '0x0000'
I (2326) light-app-callbacks: Unhandled cluster ID: 4
I (2326) light-app-callbacks: Current free heap: 94668

I (2336) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x0004', EndPoint ID: '0x00', Attribute ID: '0xfffc'
I (2346) light-app-callbacks: Unhandled cluster ID: 4
I (2346) light-app-callbacks: Current free heap: 94668

I (2366) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x002b', EndPoint ID: '0x00', Attribute ID: '0x0000'
I (2366) light-app-callbacks: Unhandled cluster ID: 43
I (2376) light-app-callbacks: Current free heap: 94668

I (2386) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x0004', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (2396) light-app-callbacks: Unhandled cluster ID: 4
I (2406) light-app-callbacks: Current free heap: 94668

I (2406) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x0004', EndPoint ID: '0x01', Attribute ID: '0xfffc'
I (2416) light-app-callbacks: Unhandled cluster ID: 4
I (2426) light-app-callbacks: Current free heap: 94668

I (2436) light-app-callbacks: emberAfOnOffClusterInitCallback
I (2436) app-task: Writing to OnOff cluster
I (2446) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x0006', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (2456) LEDWidget: Setting state to 0
I (2466) light-app-callbacks: Current free heap: 94668

I (2466) app-task: Writing to Current Level cluster
E (2476) app-task: Updating level cluster failed: 87
I (2476) chip[ZCL]: On/Off set value: 1 0
I (2476) chip[ZCL]: On/off already set to new value
I (2496) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x0008', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (2496) LEDWidget: Setting brightness to 1
I (2506) LEDWidget: DoSet to GPIO number 5
I (2506) light-app-callbacks: Current free heap: 94536

I (2516) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x0300', EndPoint ID: '0x01', Attribute ID: '0x0007'
I (2526) light-app-callbacks: Unhandled cluster ID: 768
I (2536) light-app-callbacks: Current free heap: 94536

I (2536) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x0300', EndPoint ID: '0x01', Attribute ID: '0x0008'
[... 419 lines cut ...]
I (283896) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (283896) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: A667913808E74C0A.
I (283906) chip[DIS]: mDNS service published: _matterc._udp
I (283986) chip[TS]: Pending Last Known Good Time: 2023-01-06T08:34:44
I (283986) chip[TS]: Previous Last Known Good Time: 2023-01-03T16:23:00
I (283986) chip[TS]: Reverted Last Known Good Time to previous value
E (284006) chip[FP]: Warning: metadata not found during delete of fabric 0x1
E (284016) chip[ZCL]: OpCreds: failed to delete fabric at index 1: d8
I (284016) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x0030', EndPoint ID: '0x00', Attribute ID: '0x0000'
I (284026) light-app-callbacks: Unhandled cluster ID: 48
I (284026) light-app-callbacks: Current free heap: 93552

I (284046) app-devicecallbacks: Current free heap: 93552

E (284046) chip[DL]: Long dispatch time: 326 ms, for event type 32782
I (284056) chip[FS]: Fail-safe cleanly disarmed
I (284056) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (284076) NimBLE: GAP procedure initiated: advertise; 
I (284076) NimBLE: disc_mode=2
I (284076) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (284086) NimBLE: 

I (284096) chip[DL]: CHIPoBLE advertising started
```
`

dhrishi, 2023-01-06:
@WindFromTheWoods Is the device commissioned earlier? In that case can you erase-flash and try again:

```
$ idf.py erase-flash build flash monitor
```

Also, which example is this? We have removed the QR code related code from the examples (as it was only for development/testing)

WindFromTheWoods, 2023-01-06:
> @WindFromTheWoods Is the device commissioned earlier? In that case can you erase-flash and try again:
> 
> ```
> $ idf.py erase-flash build flash monitor
> ```
> 
> Also, which example is this? We have removed the QR code related code from the examples (as it was only for development/testing)

i always use erase flash after commisioning. I use the example from the connecthomeid folder

WindFromTheWoods, 2023-01-06:
> i always use erase flash after commisioning. I use the example from the connecthomeid folder

/connectedhomeip/connectedhomeip/examples/lighting-app/esp32

dhrishi, 2023-01-06:
@WindFromTheWoods We will try out with the lighting-app. Can you please try the `examples/light` example? It should work fine. We have tested it with Google Home

WindFromTheWoods, 2023-01-06:
> @WindFromTheWoods We will try out with the lighting-app. Can you please try the `examples/light` example? It should work fine. We have tested it with Google Home

Ok, Thx. So, how i can add device to Google Home? Becouse i didn`t saw pair code or qr.
by CHiP-tool?

dhrishi, 2023-01-06:
@WindFromTheWoods You can scan the QR code from here: https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#commissioning (scroll down)

WindFromTheWoods, 2023-01-06:
> @WindFromTheWoods You can scan the QR code from here: https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#commissioning (scroll down)

Nothing has changed, the unknown error still appears

```
I (361794) chip[DL]: BLE GAP connection established (con 0)
I (361804) chip[DL]: CHIPoBLE advertising stopped
I (364254) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (364264) chip[BLE]: local and remote recv window sizes = 5
I (364264) chip[BLE]: selected BTP version 4
I (364274) chip[BLE]: using BTP fragment sizes rx 128 / tx 128.
I (364284) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 0 ) indicate = 1
I (364294) chip[DL]: CHIPoBLE subscribe received
I (364294) NimBLE: GATT procedure initiated: indicate; 
I (364294) NimBLE: att_handle=14

I (364324) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (364334) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (364344) chip[EM]: >>> [E:39097r M:120648669] (U) Msg RX from 0:A144582F89BABD8A [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
I (364364) chip[EM]: <<< [E:39097r M:180238095] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:21 (SecureChannel:PBKDFParamResponse)
I (364364) chip[IN]: (U) Sending msg 180238095 to IP address 'BLE'
I (364374) NimBLE: GATT procedure initiated: indicate; 
I (364374) NimBLE: att_handle=14

I (364404) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (364404) NimBLE: GATT procedure initiated: indicate; 
I (364414) NimBLE: att_handle=14

I (364434) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (364464) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (364464) chip[EM]: >>> [E:39097r M:120648670] (U) Msg RX from 0:A144582F89BABD8A [0000] --- Type 0000:22 (SecureChannel:PASE_Pake1)
I (365754) chip[EM]: <<< [E:39097r M:180238096] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:23 (SecureChannel:PASE_Pake2)
I (365754) chip[IN]: (U) Sending msg 180238096 to IP address 'BLE'
I (365764) NimBLE: GATT procedure initiated: indicate; 
I (365764) NimBLE: att_handle=14

E (365784) chip[DL]: Long dispatch time: 1316 ms, for event type 7
I (365844) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (365844) NimBLE: GATT procedure initiated: indicate; 
I (365854) NimBLE: att_handle=14

I (365984) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (365994) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (366004) chip[EM]: >>> [E:39097r M:120648671] (U) Msg RX from 0:A144582F89BABD8A [0000] --- Type 0000:24 (SecureChannel:PASE_Pake3)
I (366024) chip[EM]: <<< [E:39097r M:180238097] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (366024) chip[IN]: (U) Sending msg 180238097 to IP address 'BLE'
I (366034) NimBLE: GATT procedure initiated: indicate; 
I (366034) NimBLE: att_handle=14

I (366054) chip[SC]: SecureSession[0x3ffc8140]: Moving from state 'kEstablishing' --> 'kActive'
I (366054) chip[SVR]: Commissioning completed session establishment step
I (366064) chip[DIS]: Updating services using commissioning mode 0
I (366064) chip[DIS]: CHIP minimal mDNS started advertising.
I (366084) chip[SVR]: Device completed Rendezvous process
I (366084) app_main: Commissioning session started
I (366084) app_main: Commissioning window closed
I (366134) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (366184) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (366184) chip[EM]: >>> [E:39098r M:247433508] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
I (366204) chip[EM]: <<< [E:39098r M:148612879] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
I (366204) chip[IN]: (S) Sending msg 148612879 on secure session with LSID: 45565
I (366224) NimBLE: GATT procedure initiated: indicate; 
I (366224) NimBLE: att_handle=14

I (366374) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (366384) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (366394) chip[EM]: >>> [E:39099r M:247433509] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
I (366404) chip[EM]: <<< [E:39099r M:148612880] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
I (366414) chip[IN]: (S) Sending msg 148612880 on secure session with LSID: 45565
I (366414) NimBLE: GATT procedure initiated: indicate; 
[... 285 lines cut ...]
I (488534) chip[ZCL]: OpCreds: Fabric index 0x1 was removed
I (488534) chip[DIS]: Updating services using commissioning mode 1
I (488554) chip[DIS]: CHIP minimal mDNS started advertising.
I (488554) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (488564) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: CEF978E4C7BBF29B.
I (488584) chip[DIS]: mDNS service published: _matterc._udp
I (488664) chip[TS]: Pending Last Known Good Time: 2023-01-06T14:47:17
I (488664) chip[TS]: Previous Last Known Good Time: 2023-01-06T14:47:17
I (488664) chip[TS]: Reverted Last Known Good Time to previous value
[... cut ...]
E (488714) chip[DL]: Long dispatch time: 344 ms, for event type 32782
I (488734) chip[FS]: Fail-safe cleanly disarmed
I (488734) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (488744) NimBLE: GAP procedure initiated: advertise; 
I (488744) NimBLE: disc_mode=2
I (488744) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (488764) NimBLE: 

I (488764) chip[DL]: CHIPoBLE advertising started
I (488774) app_main: Commissioning window opened

```

WindFromTheWoods, 2023-01-06:
With apple home + HomePod all works fine.

WindFromTheWoods, 2023-01-06:
Ah, no

```
I (452279) chip[EM]: >>> [E:20937i M:20895948 (Ack:173551525)] (S) Msg RX from 1:7B9195E1F2F740A0 [F720] --- Type 0000:10 (SecureChannel:StandaloneAck)
E (452289) chip[EM]: OnMessageReceived failed, err = 70
I (452289) chip[EM]: >>> [E:20937i M:20895949 (Ack:173551525)] (S) Msg RX from 1:7B9195E1F2F740A0 [F720] --- Type 0000:10 (SecureChannel:StandaloneAck)
E (452309) chip[EM]: OnMessageReceived failed, err = 70
I (452309) chip[EM]: >>> [E:20937i M:20895950 (Ack:173551525)] (S) Msg RX from 1:7B9195E1F2F740A0 [F720] --- Type 0000:10 (SecureChannel:StandaloneAck)
E (452319) chip[EM]: OnMessageReceived failed, err = 70
I (452339) chip[EM]: >>> [E:20937i M:20895951 (Ack:173551525)] (S) Msg RX from 1:7B9195E1F2F740A0 [F720] --- Type 0000:10 (SecureChannel:StandaloneAck)
E (452349) chip[EM]: OnMessageReceived failed, err = 70

```

> With apple home + HomePod all works fine.

dhrishi, 2023-01-06:
@WindFromTheWoods When does the above happen? Does the device recover after this error or you cannot use it thereafter?

WindFromTheWoods, 2023-01-06:
> @WindFromTheWoods When does the above happen? Does the device recover after this error or you cannot use it thereafter?

In the case of Google Home, the device does not restart, but the pairing window remains closed until the Failsafe timer elapses.

In the case of Apple Home, the device also does not restart. I can manage it, but after any change it goes offline, in home it shows as "Updating".

WindFromTheWoods, 2023-01-06:
> @WindFromTheWoods Can you please upload the device side logs for both the cases (since bootup)

Okay, but let's get back to it on Tuesday when I'm at work

WindFromTheWoods, 2023-01-10:
> @WindFromTheWoods Can you please upload the device side logs for both the cases (since bootup)

try with Google Home

```
entry 0x40080694
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (29) boot: compile time 13:36:19
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (84) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (99) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (106) boot: End of partition table
I (110) boot: No factory image, trying OTA 0
I (115) boot_comm: chip revision: 1, min. application chip revision: 0
I (122) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=2e4b0h (189616) map
I (200) esp_image: segment 1: paddr=0004e4d8 vaddr=3ffbdb60 size=01b40h (  6976) load
I (203) esp_image: segment 2: paddr=00050020 vaddr=400d0020 size=e2c18h (928792) map
I (541) esp_image: segment 3: paddr=00132c40 vaddr=3ffbf6a0 size=03a40h ( 14912) load
I (548) esp_image: segment 4: paddr=00136688 vaddr=40080000 size=1ccc8h (117960) load
I (596) esp_image: segment 5: paddr=00153358 vaddr=50000000 size=00010h (    16) load
I (611) boot: Loaded app from partition at offset 0x20000
I (642) boot: Set actual ota_seq=1 in otadata[0]
I (643) boot: Disabling RNG early entropy source...
I (653) cpu_start: Pro cpu up.
I (654) cpu_start: Starting app cpu, entry point is 0x400813e4
0x400813e4: call_start_cpu1 at /home/eviiiil/matter/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (670) cpu_start: Pro cpu start user code
I (670) cpu_start: cpu freq: 160000000
I (670) cpu_start: Application information:
I (674) cpu_start: Project name:     chip-lighting-app
I (680) cpu_start: App version:      v1.0
I (685) cpu_start: Compile time:     Jan  3 2023 16:22:01
I (691) cpu_start: ELF file SHA256:  6b105e2c77cfcd6d...
I (697) cpu_start: ESP-IDF:          v4.4.2
I (702) heap_init: Initializing. RAM available for dynamic allocation:
I (709) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (715) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (721) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (727) heap_init: At 3FFD3348 len 0000CCB8 (51 KiB): DRAM
I (733) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (740) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (746) heap_init: At 4009CCC8 len 00003338 (12 KiB): IRAM
I (754) spi_flash: detected chip: generic
I (757) spi_flash: flash io: dio
I (764) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (795) light-app: ==================================================
I (795) light-app: chip-esp32-light-example starting
I (805) light-app: ==================================================
I (985) system_api: Base MAC address is not set3, stack:6656, core=0
> I (985) system_api: read default base MAC address from EFUSE
I (995) wifi:wifi firmware version: eeaa27d
I (1005) wifi:wifi certification version: v7.0
I (1005) wifi:config NVS flash: enabled
I (1015) wifi:config nano formating: disabled
I (1015) wifi:Init data frame dynamic rx buffer num: 32
I (1015) wifi:Init management frame dynamic rx buffer num: 32
I (1025) wifi:Init management short buffer num: 32
[... 519 lines cut ...]
I (259335) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 00A07DFF2855D17C.
I (259345) chip[DIS]: mDNS service published: _matterc._udp
I (259425) chip[TS]: Pending Last Known Good Time: 2023-01-10T07:49:31
I (259425) chip[TS]: Previous Last Known Good Time: 2023-01-03T16:23:00
I (259425) chip[TS]: Reverted Last Known Good Time to previous value
E (259445) chip[FP]: Warning: metadata not found during delete of fabric 0x1
E (259455) chip[ZCL]: OpCreds: failed to delete fabric at index 1: d8
I (259455) light-app-callbacks: PostAttributeChangeCallback - Cluster ID: '0x0030', EndPoint ID: '0x00', Attribute ID: '0x0000'
I (259465) light-app-callbacks: Unhandled cluster ID: 48
I (259465) light-app-callbacks: Current free heap: 93584

I (259485) app-devicecallbacks: Current free heap: 93584

E (259485) chip[DL]: Long dispatch time: 324 ms, for event type 32782
I (259495) chip[FS]: Fail-safe cleanly disarmed
I (259495) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (259515) NimBLE: GAP procedure initiated: advertise; 
I (259515) NimBLE: disc_mode=2
I (259515) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (259525) NimBLE: 

I (259525) chip[DL]: CHIPoBLE advertising started
I (259535) app-devicecallbacks: Current free heap: 93584

```

WindFromTheWoods, 2023-01-10:
With Apple Home, and i try change brightness 

```

entry 0x40080694
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (29) boot: compile time 13:36:19
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (84) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (99) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (106) boot: End of partition table
I (110) boot: No factory image, trying OTA 0
I (115) boot_comm: chip revision: 1, min. application chip revision: 0
I (203) esp_image: segment 2: paddr=00050020 vaddr=400d0020 size=e2c18h (928792) mapd
I (541) esp_image: segment 3: paddr=00132c40 vaddr=3ffbf6a0 size=03a40h ( 14912) load
I (547) esp_image: segment 4: paddr=00136688 vaddr=40080000 size=1ccc8h (117960) load
I (596) esp_image: segment 5: paddr=00153358 vaddr=50000000 size=00010h (    16) load
I (611) boot: Loaded app from partition at offset 0x20000
I (643) boot: Set actual ota_seq=1 in otadata[0]
I (643) boot: Disabling RNG early entropy source...
I (654) cpu_start: Pro cpu up.
I (655) cpu_start: Starting app cpu, entry point is 0x400813e4
0x400813e4: call_start_cpu1 at /home/eviiiil/matter/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (671) cpu_start: Pro cpu start user code
I (671) cpu_start: cpu freq: 160000000
I (671) cpu_start: Application information:
I (675) cpu_start: Project name:     chip-lighting-app
I (681) cpu_start: App version:      v1.0
I (686) cpu_start: Compile time:     Jan  3 2023 16:22:01
I (692) cpu_start: ELF file SHA256:  6b105e2c77cfcd6d...
I (698) cpu_start: ESP-IDF:          v4.4.2
I (703) heap_init: Initializing. RAM available for dynamic allocation:
I (710) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (716) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (722) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (728) heap_init: At 3FFD3348 len 0000CCB8 (51 KiB): DRAM
I (734) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (741) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (747) heap_init: At 4009CCC8 len 00003338 (12 KiB): IRAM
I (755) spi_flash: detected chip: generic
I (758) spi_flash: flash io: dio
I (765) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (796) light-app: ==================================================
I (796) light-app: chip-esp32-light-example starting
I (806) light-app: ==================================================
I (986) system_api: Base MAC address is not set3, stack:6656, core=0
I (986) system_api: read default base MAC address from EFUSE
> I (1006) wifi:wifi firmware version: eeaa27d
I (1006) wifi:wifi certification version: v7.0
I (1006) wifi:config NVS flash: enabled
I (1006) wifi:config nano formating: disabled
I (1006) wifi:Init data frame dynamic rx buffer num: 32
I (1026) wifi:Init management frame dynamic rx buffer num: 32
I (1026) wifi:Init management short buffer num: 32
I (1026) wifi:Init dynamic tx buffer num: 32
I (1036) wifi:Init static rx buffer size: 1600
I (1036) wifi:Init static rx buffer num: 10
[... 1537 lines cut ...]
I (249836) chip[DMG]: Refresh Subscribe Sync Timer with min 0 seconds and max 4 seconds
I (250226) chip[IN]: (S) Sending msg 258665978 on secure session with LSID: 1268
I (250916) chip[EM]: >>> [E:63004i M:149717357 (Ack:258665978)] (S) Msg RX from 1:7B9195E1F2F740A0 [F720] --- Type 0001:01 (IM:StatusResponse)
I (250916) chip[IM]: Received status response, status is 0x00
I (250926) chip[EM]: <<< [E:63004i M:258665979 (Ack:149717357)] (S) Msg TX to 1:7B9195E1F2F740A0 [F720] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (250946) chip[IN]: (S) Sending msg 258665979 on secure session with LSID: 1268
I (250946) chip[EM]: >>> [E:63004i M:149717357 (Ack:258665978)] (S) Msg RX from 1:7B9195E1F2F740A0 [F720] --- Type 0001:01 (IM:StatusResponse)
I (250966) chip[EM]: <<< [E:63004i M:258665980 (Ack:149717357)] (S) Msg TX to 1:7B9195E1F2F740A0 [F720] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (250976) chip[IN]: (S) Sending msg 258665980 on secure session with LSID: 1268
I (250996) chip[EM]: >>> [E:63004i M:149717358 (Ack:258665978)] (S) Msg RX from 1:7B9195E1F2F740A0 [F720] --- Type 0000:10 (SecureChannel:StandaloneAck)
E (251006) chip[EM]: OnMessageReceived failed, err = 70
I (251206) chip[EM]: >>> [E:63004i M:149717357 (Ack:258665978)] (S) Msg RX from 1:7B9195E1F2F740A0 [F720] --- Type 0001:01 (IM:StatusResponse)
I (251206) chip[EM]: <<< [E:63004i M:258665981 (Ack:149717357)] (S) Msg TX to 1:7B9195E1F2F740A0 [F720] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (251216) chip[IN]: (S) Sending msg 258665981 on secure session with LSID: 1268
I (253836) chip[DMG]: Refresh subscribe timer sync after 4 seconds
I (253846) chip[EM]: <<< [E:63005i M:258665982] (S) Msg TX to 1:7B9195E1F2F740A0 [F720] --- Type 0001:05 (IM:ReportData)
I (253846) chip[IN]: (S) Sending msg 258665982 on secure session with LSID: 1268
[... cut ...]
I (253886) chip[EM]: >>> [E:63005i M:149717359 (Ack:258665982)] (S) Msg RX from 1:7B9195E1F2F740A0 [F720] --- Type 0001:01 (IM:StatusResponse)
I (253886) chip[IM]: Received status response, status is 0x00
I (253896) chip[EM]: <<< [E:63005i M:258665983 (Ack:149717359)] (S) Msg TX to 1:7B9195E1F2F740A0 [F720] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (253916) chip[IN]: (S) Sending msg 258665983 on secure session with LSID: 1268
I (257866) chip[DMG]: Refresh subscribe timer sync after 4 seconds

```

WindFromTheWoods, 2023-01-12:
@dhrishi

dhrishi, 2023-01-12:
@WindFromTheWoods Looks like you are still using the lighting-app. As discussed above can you please try with the light example in this repository?
Also, if possible, please upload/attach the subsequent console logs instead of pasting them in the comment :-)

dhairyashah1, 2023-01-12:
@WindFromTheWoods 
Could you please retry with this [`esp-matter light example`](https://github.com/espressif/esp-matter/tree/main/examples/light)?

WindFromTheWoods, 2023-01-12:
@dhairyashah1 @dhrishi 

I (796) light-app: chip-esp32-light-example starting
isn`t that it

WindFromTheWoods, 2023-01-12:
[logs.txt](https://github.com/espressif/esp-matter/files/10400426/logs.txt)

dhairyashah1, 2023-01-13:
@WindFromTheWoods 
Based on your logs, commissioning seems to follow the standard flow and fails due to `exceeding the failsafetime`. However, did you follow [`these steps`](https://support.google.com/googlenest/answer/13127223?hl=en#:~:text=Add%20your%20Matter%2Denabled%20device%20to%20the%20Google%20Home%20app,-Open%20the%20Google&text=New%20device%20Matter%2Denabled%20device,add%20your%20Matter%2Denabled%20device.)? 

This says that - 
1. You need a google account 
2. Latest version of the google home app 
3. Google Play services version 22.48.14 or later and etc, 

Could you check against the requirements and get back whether they are satisfied for you in Ukraine? This could be the reason for failure. For me, the device is getting commissioned and working fine using all - `chip-tool, google-home and apple-home` for [`esp-matter light example`](https://github.com/espressif/esp-matter/tree/main/examples/light) and also [`connecctedhomeip/examples/lighting-app/esp32)`](https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/esp32).

dhairyashah1, 2023-01-18:
@WindFromTheWoods did you manage to verify and follow the above steps?

closed 2023-01-19 by WindFromTheWoods