espressif/esp-matter #157: Does this sdk support esp32c6？

Firesuiry, 2023-01-07:
Does this sdk support esp32c6？

dhrishi, 2023-01-10:
@Firesuiry ESP-IDF v5.1 release (not released yet) will introduce initial support for ESP32-C6. So, the answer for now is that the Matter SDK doesn't support ESP32-C6. It will be added later.

closed 2023-01-10 by dhrishi

dhrishi, 2023-01-10:
More details about ESP32-C6 support status -> https://github.com/espressif/esp-idf/issues/10423

mgoeller, 2023-03-28:
According to the support status thread WiFi and Thread are now supported and even coexistence. 
`Coexistence of Wi-Fi and 802.15.4 - Supported`
What does that say regarding support of this SDK?

dhrishi, 2023-03-28:
@mgoeller The latest `main` branch of esp-matter now supports ESP32-C6 (preview) with ESP-IDF release/v5.1 branch. Please see [this](https://github.com/espressif/esp-matter/#supported-esp-idf-and-connectedhomeip-versions)