espressif/esp-matter #158: read seruc====][

shun-dgut, 2023-01-09:
basic read  serial-number  0x7283 0x00
[1673256155.527357][68074:68079] CHIP:TOO: Sending command to node 0x7283
[1673256155.527568][68074:68079] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000007283]
[1673256155.527583][68074:68079] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1673256155.527602][68074:68079] CHIP:CTL: Found an existing secure session to [1:0000000000007283]!
[1673256155.527612][68074:68079] CHIP:CTL: OperationalSessionSetup[1:0000000000007283]: State change 1 --> 5
[1673256155.527639][68074:68079] CHIP:TOO:      cluster 0x0000_0028, attribute: 0x0000_000F, endpoint 0
[1673256155.527651][68074:68079] CHIP:TOO: Sending ReadAttribute to:
[1673256155.527663][68074:68079] CHIP:DMG: SendReadRequest ReadClient[0x7fa13000e580]: Sending Read Request
[1673256155.527815][68074:68079] CHIP:EM: <<< [E:6436i M:10625817] (S) Msg TX to 1:0000000000007283 [13D9] --- Type 0001:02 (IM:ReadRequest)
[1673256155.527837][68074:68079] CHIP:IN: (S) Sending msg 10625817 on secure session with LSID: 42613
[1673256155.527946][68074:68079] CHIP:DMG: MoveToState ReadClient[0x7fa13000e580]: Moving to [AwaitingIn]
[1673256155.669595][68074:68079] CHIP:EM: >>> [E:6436i M:71438956 (Ack:10625817)] (S) Msg RX from 1:0000000000007283 [13D9] --- Type 0001:05 (IM:ReportData)
[1673256155.669634][68074:68079] CHIP:EM: Found matching exchange: 6436i, Delegate: 0x7fa13000e590
[1673256155.669655][68074:68079] CHIP:EM: Rxd Ack; Removing MessageCounter:10625817 from Retrans Table on exchange 6436i
[1673256155.669694][68074:68079] CHIP:DMG: ReportDataMessage =
[1673256155.669706][68074:68079] CHIP:DMG: {
[1673256155.669714][68074:68079] CHIP:DMG:      AttributeReportIBs =
[1673256155.669726][68074:68079] CHIP:DMG:      [
[1673256155.669733][68074:68079] CHIP:DMG:              AttributeReportIB =
[1673256155.669745][68074:68079] CHIP:DMG:              {
[1673256155.669754][68074:68079] CHIP:DMG:                      AttributeStatusIB =
[1673256155.669773][68074:68079] CHIP:DMG:                      {
[1673256155.669782][68074:68079] CHIP:DMG:                              AttributePathIB =
[1673256155.669793][68074:68079] CHIP:DMG:                              {
[1673256155.669803][68074:68079] CHIP:DMG:                                      Endpoint = 0x0,
[1673256155.669811][68074:68079] CHIP:DMG:                                      Cluster = 0x28,
[1673256155.669826][68074:68079] CHIP:DMG:                                      Attribute = 0x0000_000F,
[1673256155.669835][68074:68079] CHIP:DMG:                              }
[1673256155.669847][68074:68079] CHIP:DMG:
[1673256155.669861][68074:68079] CHIP:DMG:                              StatusIB =
[1673256155.669872][68074:68079] CHIP:DMG:                              {
[1673256155.669885][68074:68079] CHIP:DMG:                                      status = 0x86 (UNSUPPORTED_ATTRIBUTE),
[1673256155.669895][68074:68079] CHIP:DMG:                              },
[1673256155.669904][68074:68079] CHIP:DMG:
[1673256155.669913][68074:68079] CHIP:DMG:                      },
[1673256155.669927][68074:68079] CHIP:DMG:
[1673256155.669933][68074:68079] CHIP:DMG:              },
[1673256155.669946][68074:68079] CHIP:DMG:
[1673256155.669953][68074:68079] CHIP:DMG:      ],
[1673256155.669963][68074:68079] CHIP:DMG:
[1673256155.669972][68074:68079] CHIP:DMG:      SuppressResponse = true,
[1673256155.669981][68074:68079] CHIP:DMG:      InteractionModelRevision = 1
[1673256155.669986][68074:68079] CHIP:DMG: }
[1673256155.670056][68074:68079] CHIP:TOO: Response Failure: IM Error 0x00000586: General error: 0x86 (UNSUPPORTED_ATTRIBUTE)
[1673256155.670130][68074:68079] CHIP:EM: <<< [E:6436i M:10625818 (Ack:71438956)] (S) Msg TX to 1:0000000000007283 [13D9] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1673256155.670146][68074:68079] CHIP:IN: (S) Sending msg 10625818 on secure session with LSID: 42613
[1673256155.670221][68074:68079] CHIP:EM: Flushed pending ack for MessageCounter:71438956 on exchange 6436i
[1673256155.670239][68074:68074] CHIP:TOO: Run command failure: IM Error 0x00000586: General error: 0x86 (UNSUPPORTED_ATTRIBUTE)
[1673256176.005910][68074:68078] CHIP:DL: Got IP address on interface: wlp0s20f3 IP: 192.168.0.165
[1673256176.006090][68074:68079] CHIP:DL: HandlePlatformSpecificBLEEvent 32770
[1673256286.597302][68074:68078] CHIP:DL: Got IP address on interface: wlp0s20f3 IP: 192.168.0.165
[1673256286.597465][68074:68079] CHIP:DL: HandlePlatformSpecificBLEEvent 32770

dhrishi, 2023-01-09:
@shun-dgut As per the Matter specification, SerialNumber is an optional attribute in the Basic Information Cluster. We add only the mandatory ones by default. If you want the SerialNumber, you can add the attribute using the data model APIs

shun-dgut, 2023-01-10:
Thanks，Can you tell me where to add it？

jadhavrohit924, 2023-01-10:
@shun-dgut You have to create serialNumber attribute [here](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.cpp#L215) using `create_serial_number` API. Same as we created Node Label.

jadhavrohit924, 2023-01-11:
If you set serial number here^ it will not reflect as `serial_number` attribute is handled internally in connectedhomeip [see here](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_attribute.h#L78)

You can set serial number using [mfg-tool](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool) for that you will need to enable some configs through `idf.py menuconfig`.

> Enabled ESP32 Factory Data Provider [Component config → CHIP Device Layer → Commissioning options → Use ESP32 Factory Data Provider]
> Enabled ESP32 Device Instance Info Provider [Component config → CHIP Device Layer → Commissioning options → Use ESP32 Device Instance Info Provider]
> Enabled Attestation - Factory [ Component config → ESP Matter → DAC Provider options → Attestation - Factory] { Since we are generating a custom partition using mfg_tool it needs to use proper DAC certificates }

Then flash bin generated via mfg-tool.
Then you will see serial_number set by you.

jonsmirl, 2023-01-11:
How about an option in mfg_tool to set the serial_number to the ESP32 chip's internal MAC address?

shubhamdp, 2023-01-16:
@jonsmirl mfg_tool generates binaries and do not interact with the device to fetch the mac address.
This utility generates "N" partitions, (default is 1) so either we would need N devices connected to the host and script has to provide N ports from where to read the mac address.

And serial number is an optional attribute

shubhamdp, 2023-01-16:
I think we can use https://github.com/project-chip/connectedhomeip/blob/master/scripts/tools/generate_esp32_chip_factory_bin.py for such purpose

dhrishi, 2023-01-16:
@shun-dgut Please close the issue if it works fine

dhrishi, 2023-01-23:
Closing. Please feel free to re-open if you are unable to generate and read the serial number

closed 2023-01-23 by dhrishi