espressif/esp-matter #159: Unable to generate correct QR code

shun-dgut, 2023-01-10:
custom flow  does not know why it is specified，without specifying ble

[image]

dhairyashah1, 2023-01-10:
@shun-dgut 
The `custom flow` is defined and governed by the `commissioning mode` parameter. 
[image]

Custom flow value field can have three values - `0, 1 or 2`. This is mentioned in the `Matter_Core_V1.0_Specification` -
[image]

shun-dgut, 2023-01-10:
> @shun-dgut The `custom flow` is defined and governed by the `commissioning mode` parameter. [image]
> 
> Custom flow value field can have three values - `0, 1 or 2`. This is mentioned in the `Matter_Core_V1.0_Specification` - [image]
> 
> Since you have set `commissioning mode -> 2` what exactly do you want/desire to do with the custom flow?

Thanks ，If I want to set Discovery Bitmask what parameters do I need

[image]

dhairyashah1, 2023-01-10:
@shun-dgut 
1. The discovery bitmask can be set using the `--rendezvous` option present in the command,
[image]

2. Usage - 
`  ./chip-tool payload generate-qrcode [--existing-payload] [--discriminator] [--setup-pin-code] [--version] [--vendor-id] [--product-id] [--commissioning-mode] [--allow-invalid-payload] [--rendezvous] [--tlvBytes]`

3. 
[image]

closed 2023-01-11 by shun-dgut