espressif/esp-matter #161: Single Node with multiple Endpoints vs. Bridge with multiple Nodes

mgoeller, 2023-01-11:
We have a physical device acting as a Gateway between a bus-based system (multiple lights, blinds, doors etc.) and Matter.

Should we implement our Gateway as a single node with an endpoint for each light, blind, door. Or is better to implement a bridge with one bridged-node per light, blind, door?

wqx6, 2023-01-12:
@mgoeller Hi, the SPEC has been updated after blog. Now the SPEC defines a device type `aggregator` (formerly `bridge` device type) to manage the bridged nodes. And the aggregator endpoint should be the parent endpoint of the bridged-endpoints that it manages.
[image]
In our example, we create Endpoint1 as the aggregator device, and then create its child endpoints for the bridged-devices.

Also you can add the `aggregator` device type to the root node Endpoint0 if there is only one aggregator endpoint. But it might cause some understandings if there is other aggregator endpoints which mange other bridged nodes. In this situation, the root node's part list should include all the endpoints except itself, but the aggregator endpoint's part list should only include its child endpoints.

jadhavrohit924, 2023-01-12:
> E (3258) esp_matter_bridge: Could not find endpoint index

Can you please share where you called `app_bridge_create_bridged_device`?

jonsmirl, 2023-01-15:
The bridge is designed to have devices come and go and to forward notifications to that effect. If you use multiple endpoints it is assumed the devices are static. But in general the two approaches are very similar.

mgoeller, 2023-01-18:
@wqx6 and @jonsmirl: 
Thank you, the differences between bridge vs. node with multiple endpoints is now clear. We also have seen that the Apple Home app handles a node with multiple endpoints as an accessory with multiple services. Even though you can separate them as individual tiles, you can't move them in different rooms from each other.

We are used from HomeKit, that an empty bridge (in matter that would be: an aggregator with multiple endpoints) pairs completely fine, and when dynamically accessories / endpoints later on they just start to show up. A Matter empty aggregator shows up in HomeKit as an unsupported home accessory. Which is probably an unwanted behaviour, right?

@jadhavrohit924: We create an aggregator with multiple accessories by using the Zigbee bridge example from esp-matter and modifying the provided APIs to fit our use-case:

Node with parent aggregator:
```
node::config_t node_config;
node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
endpoint_t *aggregator = endpoint::aggregator::create(node, ENDPOINT_FLAG_NONE, NULL);
aggregator_endpoint_id = endpoint::get_id(aggregator);
```
Starting matter:
`esp_matter::start(app_event_cb);`

Then, we basically use the following code:
```
esp_matter_bridge::initialize(node);
esp_matter_bridge::create_device(node, aggregator_endpoint_id, ESP_MATTER_ON_OFF_LIGHT_DEVICE_TYPE_ID);
```
but we execute those functions using the helper functions defined in  `app_bridge_device.cpp`:
`app_bridge_initialize(node)` 

Google and Apple Home applications see the compiled Matter aggregator with 1 simple Lightbulb. Google Home can't pair at all, Apple Home pairs successfully, but has "No Response" immediately after. Tried with multiple devices, and different Hubs are all online. Also the ESP32 is rebooting during the pairing process with Apple Home:

```
I (63625) matter_init: Commissioning complete
I (63625) NimBLE: GAP procedure initiated: stop advertising.
I (63635) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19
I (63645) chip[DL]: CHIPoBLE unsubscribe received
I (63645) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40056eed  PS      : 0x00060130  A0      : 0x8202ee0a  A1      : 0x3fcd82c0  
A2      : 0x3fca5334  A3      : 0x00000007  A4      : 0x00000007  A5      : 0x00060b23  
A6      : 0x00060b20  A7      : 0x00000001  A8      : 0x00000007  A9      : 0x00000000  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00060b20  A13     : 0x00000000  
A14     : 0x02c9a930  A15     : 0x00ffffff  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000007  LBEG    : 0x4202f06e  LEND    : 0x4202f07b  LCOUNT  : 0x00000002  
0x4202f06e: ble_hs_hci_evt_dispatch_find at /Volumes/case-sensitive/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:138
 (inlined by) ble_hs_hci_evt_process at /Volumes/case-sensitive/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:896

0x4202f07b: ble_hs_hci_evt_process at /Volumes/case-sensitive/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:899

Backtrace: 0x40056eea:0x3fcd82c0 |<-CORRUPTED

ELF file SHA256: 6a90cf3da474d92f

Rebooting...
```

Any hints?

mgoeller, 2023-01-18:
I used to get a single node, with a single endpoint (dimmable light) to work with Google and Apple Home a few days ago. The exact same device (did not flash it) since then only works with Apple Home. With Google Home I can't pair anymore somehow. The Google Home wasn't updated, my Google phone wasn't updated (intentionally, Pixel 7 Pro), and I rebooted both (Google Nest Mini and the phone) multiple times.

I get the same error 0x213 as in this thread: [](https://github.com/espressif/esp-matter/issues/156)

jonsmirl, 2023-01-18:
IOS and Android continuously break their ability to commission devices. This has been going on for a year. You just have to live with it. Matter is not really stable yet.  When Google breaks I move over to IOS, if they are both broken I use chiptool from the command line.

Just keep updating everything and it will start working again sooner or latter. For example Google left Matter development in a non-working state for about two weeks over Christmas.

dhrishi, 2023-01-19:
> With Google Home I can't pair anymore somehow. The Google Home wasn't updated, my Google phone wasn't updated (intentionally, Pixel 7 Pro), and I rebooted both (Google Nest Mini and the phone) multiple times.

@mgoeller Please check if your device needs to be registered with Google developer console. You can check the table "Table: Vendor ID (VID) and device type pairing scenarios" here: https://developers.home.google.com/matter/get-started

dhrishi, 2023-01-19:
>Also the ESP32 is rebooting during the pairing process with Apple Home:

@mgoeller For the crash you've shared, can you please confirm your ESP-IDF version? Also, please see that your IDF submodules (specifically nimble) are updated

mgoeller, 2023-01-20:
> @mgoeller Please check if your device needs to be registered with Google developer console. You can check the table "Table: Vendor ID (VID) and device type pairing scenarios" here: https://developers.home.google.com/matter/get-started
Device Vendor ID is 0xFFF1 and Device Product ID is 0x8000. And not registered in the Google Developer Console. But I could commission the Matter accessory a few days ago. Just now I registered the Developer Console. How can I check I did this successfully?

> @mgoeller For the crash you've shared, can you please confirm your ESP-IDF version? Also, please see that your IDF submodules (specifically nimble) are updated
I'm using ESP-IDF 4.4.3 with Connected Home IP Tag 1.0.0.2 (as recommended). If I test with ESP-IDF 5.0 compiling fails.

mgoeller, 2023-01-24:
After registering my device with Google developer console (remember, it used to work before) again, pairing is now working and the device is responsive.

However since yesterday, Apple Home shows "No Response" after pairing. I completely reset / reflashed the device, using the same firmware binary file like a week ago (that used to work with Apple Home). HomeHub is active, same Apple ID and I tried with both: iOS16.2 and iOS16.3

Any tips?

dhrishi, 2023-01-25:
>After registering my device with Google developer console (remember, it used to work before) again, pairing is now working and the device is responsive.

The behavior was changed sometime back and now it requires registration

>HomeHub is active, same Apple ID and I tried with both: iOS16.2 and iOS16.3

We will try with the latest update with Apple Home and update.

mgoeller, 2023-01-25:
Today I was able to successfully pair with both platforms again. So far so good.

I am now running an OnOff Light, a dimmable Light, a Shade, and a Temperature Sensor in one node with multiple endpoints. This works.

So back to my problems, I have the following problems runnings this as a bridge (allowing me to move the devices into separate rooms in Apple Home):

1.  I can't pair an empty bridge in Apple Home (unsupported device, even when dynamically adding further endpoints later on). This was possible with a pure HomeKit bridge. Shouldn't that be possible in Matter as well?
2. I am missing several device types in esp_matter_bridge.cpp. For example: shades, door_lock etc. Only Lights available. Can these please be added. When reading through the issues, in particular there seems to be a problem with [a Bridge and Shades](https://github.com/espressif/esp-matter/issues/168#issue-1549513374).

jonsmirl, 2023-01-25:
I agree that the bridge examples need to be expanded  Start with this one:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_sensor_model/sensor_client/README.md

Also needs some demo code about getting secure devices onto the mesh.

closed 2023-05-31 by mgoeller