espressif/esp-matter #162: Google: CASE failed to match destination ID with local fabrics

jonsmirl, 2023-01-15:
After commissioning I can't get it to set up a CASE session. This log from my app, but the light example doesn't work either (it doesn't print the log).  Is that %st1 in the IPV6 address right?

I tried: V1.0.0.1, V1.0.0.2 and master. All have same failure.

~~~
I (66193) app_main: Free 63304
I (66693) app_main: Free 63304
I (66973) chip[EM]: >>> [E:37713r M:27138964] (U) Msg RX from 0:84674CAF4DDC3D9A [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (66973) chip[IN]: CASE Server received Sigma1 message. Starting handshake. EC 0x3fc9af90
I (66983) chip[IN]: CASE Server disabling CASE session setups
I (66993) chip[EM]: <<< [E:37713r M:162108684 (Ack:27138964)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (67003) chip[IN]: (U) Sending msg 162108684 to IP address 'UDP:[FE80::1E53:F9FF:FEB3:3B54%st1]:5540'
I (67013) chip[SC]: Received Sigma1 msg
E (67023) chip[SC]: CASE failed to match destination ID with local fabrics
I (67023) chip[EM]: <<< [E:37713r M:162108685 (Ack:27138964)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (67043) chip[IN]: (U) Sending msg 162108685 to IP address 'UDP:[FE80::1E53:F9FF:FEB3:3B54%st1]:5540'
E (67053) chip[IN]: CASE Session establishment failed: 10
I (67053) chip[IN]: CASE Server enabling CASE session setups
E (67063) chip[EM]: OnMessageReceived failed, err = 10
I (67173) chip[EM]: >>> [E:37713r M:27138965 (Ack:162108685)] (U) Msg RX from 0:84674CAF4DDC3D9A [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (67193) app_main: Free 63304
I (67693) app_main: Free 63304
I (68193) app_main: Free 63304
I (68693) app_main: Free 63304
~~~

jonsmirl, 2023-01-16:
Here is same interaction from IOS which works:
~~~
I (68239) chip[EM]: >>> [E:44841r M:912572] (U) Msg RX from 0:D09E1D45C484C3EA [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (68239) chip[IN]: CASE Server received Sigma1 message. Starting handshake. EC 0x3fc9af90
I (68249) chip[IN]: CASE Server disabling CASE session setups
I (68249) chip[EM]: <<< [E:44841r M:22442098 (Ack:912572)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (68269) chip[IN]: (U) Sending msg 22442098 to IP address 'UDP:[FE80::1887:636C:8349:54E8%st1]:62777'
I (68279) chip[SC]: Received Sigma1 msg
I (68289) chip[EM]: <<< [E:44841r M:22442099 (Ack:912572)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:33 (SecureChannel:CASE_Sigma2Resume)
I (68289) chip[IN]: (U) Sending msg 22442099 to IP address 'UDP:[FE80::1887:636C:8349:54E8%st1]:62777'
I (68339) chip[EM]: >>> [E:44841r M:912573 (Ack:22442099)] (U) Msg RX from 0:D09E1D45C484C3EA [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (68339) chip[SC]: Success status report received. Session was established
I (68359) chip[SC]: SecureSession[0x3fc99768]: Moving from state 'kEstablishing' --> 'kActive'
I (68359) chip[IN]: CASE Session established to peer: <204B50A898A38F65, 1>
I (68369) chip[IN]: CASE Server enabling CASE session setups
~~~

jonsmirl, 2023-01-16:
Google suggestions did not help
https://issuetracker.google.com/issues/265528483

jonsmirl, 2023-01-16:
Factory reset of Home display will clear this.

dhrishi, 2023-01-16:
@jonsmirl So it worked fine after factory reset of the Hub? What changed on the Google and/or the device side that you encountered the issue? Version update or anything else?

jonsmirl, 2023-01-16:
Factory reset of the Hub. Second version update after Christmas did this. I suspect they changed the SSL keys they had stored in the hub internally and renumbered them.   I only figured this out because someone from Nordic solved it in their issue tracker.

dhrishi, 2023-01-17:
Closing as this is solved

closed 2023-01-17 by dhrishi