espressif/esp-matter #163: Some warnings in light example

eos1d3, 2023-01-16:
Just cloned this respository and built light example. Build is successful but there are some warnings:

Are these from `connectedhomeip` and can these be fixed?

```
/Users/andy/esp/esp-matter/components/esp_matter/esp_matter_event.cpp:21:20: warning: 'TAG' defined but not used [-Wunused-variable]
 static const char *TAG = "esp_matter_event";
                    ^~~
[517/632] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/EventManagement.cpp.obj
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/EventManagement.cpp: In member function 'CHIP_ERROR chip::app::EventManagement::LogEventPrivate(chip::app::EventLoggingDelegate*, const chip::app::EventOptions&, chip::EventNumber&)':
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/EventManagement.cpp:410:23: warning: 'writer' is used uninitialized in this function [-Wuninitialized]
     CircularTLVWriter checkpoint = writer;
                       ^~~~~~~~~~
[545/632] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/attribute-storage.cpp.obj
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/attribute-storage.cpp: In function 'void emberAfEndpointConfigure()':
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/attribute-storage.cpp:117:40: warning: comparison is always true due to limited range of data type [-Wtype-limits]
     static_assert(FIXED_ENDPOINT_COUNT <= std::numeric_limits<decltype(ep)>::max(),
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/attribute-storage.cpp:142:21: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     for (ep = 0; ep < FIXED_ENDPOINT_COUNT; ep++)
[578/632] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/Users/and...connectedhomeip/connectedhomeip/src/app/clusters/door-lock-server/door-lock-server.cpp.ob
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/door-lock-server/door-lock-server.cpp: In member function 'bool DoorLockServer::HandleRemoteLockOperation(chip::app::CommandHandler*, const chip::app::ConcreteCommandPath&, chip::app::Clusters::DoorLock::DlLockOperationType, RemoteLockOpHandler, const chip::Optional<chip::Span<const unsigned char> >&)':
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/door-lock-server/door-lock-server.cpp:3348:38: warning: '*((void*)& pinCredIdx +2)' may be used uninitialized in this function [-Wmaybe-uninitialized]
         foundCred[0].credentialIndex = pinCredIdx.Value();
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
[583/632] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/Users/and...atter/connectedhomeip/connectedhomeip/src/app/clusters/level-control/level-control.cpp.ob
In file included from /Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/ConcreteAttributePath.h:23,
                 from /Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/cluster-objects.h:27,
                 from /Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/attributes/Accessors.h:27,
                 from /Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/level-control/level-control.cpp:23:
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/lib/core/Optional.h: In function 'void emberAfOnOffClusterLevelControlEffectCallback(chip::EndpointId, bool)':
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/lib/core/Optional.h:72:13: warning: '*((void*)& transitionTime +2)' may be used uninitialized in this function [-Wmaybe-uninitialized]
             new (&mValue.mData) T(other.mValue.mData);
             ^~~
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/level-control/level-control.cpp:1074:40: note: '*((void*)& transitionTime +2)' was declared here
     app::DataModel::Nullable<uint16_t> transitionTime;
                                        ^~~~~~~~~~~~~~
[595/632] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/Users/and...eip/src/app/clusters/operational-credentials-server/operational-credentials-server.cpp.ob
In file included from /Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/ConcreteAttributePath.h:23,
                 from /Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/cluster-objects.h:27,
                 from /Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/attributes/Accessors.h:27,
                 from /Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/operational-credentials-server/operational-credentials-server.cpp:25:
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/lib/core/Optional.h: In function 'bool emberAfOperationalCredentialsClusterCSRRequestCallback(chip::app::CommandHandler*, const chip::app::ConcreteCommandPath&, const chip::app::Clusters::OperationalCredentials::Commands::CSRRequest::DecodableType&)':
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/lib/core/Optional.h:72:13: warning: '*((void*)& fabricIndexForCsr +1)' may be used uninitialized in this function [-Wmaybe-uninitialized]
             new (&mValue.mData) T(other.mValue.mData);
             ^~~
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/operational-credentials-server/operational-credentials-server.cpp:1054:31: note: '*((void*)& fabricIndexForCsr +1)' was declared here
         Optional<FabricIndex> fabricIndexForCsr;
                               ^~~~~~~~~~~~~~~~~
[617/632] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/window-covering-server/window-covering-server.cpp.obj
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/window-covering-server/window-covering-server.cpp: In function 'bool chip::app::Clusters::WindowCovering::IsPercent100thsValid(chip::Percent100ths)':
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/window-covering-server/window-covering-server.cpp:42:51: warning: comparison is always false due to limited range of data type [-Wtype-limits]
 #define CHECK_BOUNDS_INVALID(MIN, VAL, MAX) ((VAL < MIN) || (VAL > MAX))
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/window-covering-server/window-covering-server.cpp:43:45: note: in expansion of macro 'CHECK_BOUNDS_INVALID'
 #define CHECK_BOUNDS_VALID(MIN, VAL, MAX) (!CHECK_BOUNDS_INVALID(MIN, VAL, MAX))
                                             ^~~~~~~~~~~~~~~~~~~~
/Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/window-covering-server/window-covering-server.cpp:341:9: note: in expansion of macro 'CHECK_BOUNDS_VALID'
     if (CHECK_BOUNDS_VALID(WC_PERCENT100THS_MIN_OPEN, percent100ths, WC_PERCENT100THS_MAX_CLOSED))
         ^~~~~~~~~~~~~~~~~~
[628/632] Generating ld/sections.ld
warning: ENABLE_ESP32_LOCATIONCAPABILITY (defined at /Users/andy/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:660) has leading or trailing whitespace in its prompt
[631/632] Generating binary image from built executable
```

shripad621git, 2023-01-17:
/Users/andy/esp/esp-matter/components/esp_matter/esp_matter_event.cpp:21:20: warning: 'TAG' defined but not used [-Wunused-variable]
 static const char *TAG = "esp_matter_event";
This warning is from esp-matter.
Other warnings are from connnectedhomeip.

shripad621git, 2023-01-31:
We have fixed the warning from esp-matter.

Most of the warnings from  connectedhomeip are initialization warnings. For fixing them,we need to initialize the attributes which are Optional and Nullable.

But,initializing Optional and Nullable is wrong. The whole point of these classes is to be able to distinguish whether a given value is specified or not, so initializing them to 0 will likely change the behaviour of the code.

The one single warning which can be removed is the trailing whitespace warning for ENABLE_ESP32_LOCATIONCAPABILITY  in Kconfig file and that has been fixed.

eos1d3, 2023-01-31:
Understood your points. Thanks!

closed 2023-01-31 by eos1d3