espressif/esp-matter #164: "Unable to create file. File name too long"

abrender, 2023-01-16:
After following the instructions at [https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#getting-the-repositories](https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#getting-the-repositories) I try to build the example light application but it fails with:

```
username:examples/light$ idf.py build
...
[708/1303] Performing build step for 'chip_gn'
FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a /home/username/esp/esp-matter/examples/light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build /home/username/esp/esp-matter/examples/light/build/esp-idf/chip/lib/libCHIP.a 
cd /home/username/esp/esp-matter/examples/light/build/esp-idf/chip && ninja esp32
[1/729] stamp obj/mbedtls/mbedtls.stamp
[2/729] stamp obj/third_party/connectedhomeip/src/controller/data_model/data_model_codegen_codegen._metadata_path_list.txt.stamp
[3/729] stamp obj/third_party/connectedhomeip/src/lib/dnssd/platform_header.stamp
[4/729] stamp obj/third_party/connectedhomeip/src/lib/support/logging_constants.stamp
[5/729] ACTION //third_party/connectedhomeip/src/ble:gen_ble_buildconfig(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
[6/729] stamp obj/third_party/connectedhomeip/src/lib/support/attributes.stamp
[7/729] ACTION //third_party/connectedhomeip/src/app:gen_app_buildconfig(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
[8/729] ACTION //third_party/connectedhomeip/src/inet:gen_inet_buildconfig(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
[9/729] ACTION //third_party/connectedhomeip/src/lib/core:gen_chip_buildconfig(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
[10/729] stamp obj/third_party/connectedhomeip/src/inet/gen_inet_buildconfig.stamp
[11/729] stamp obj/third_party/connectedhomeip/src/ble/gen_ble_buildconfig.stamp
[12/729] stamp obj/third_party/connectedhomeip/src/app/gen_app_buildconfig.stamp
[13/729] stamp obj/third_party/connectedhomeip/src/lib/core/gen_chip_buildconfig.stamp
[14/729] stamp obj/third_party/connectedhomeip/src/inet/inet_buildconfig.stamp
[15/729] stamp obj/third_party/connectedhomeip/src/ble/ble_buildconfig.stamp
[16/729] stamp obj/third_party/connectedhomeip/src/app/app_buildconfig.stamp
[17/729] stamp obj/third_party/connectedhomeip/src/lib/core/chip_buildconfig.stamp
[18/729] stamp obj/third_party/connectedhomeip/src/platform/logging/headers.stamp
ninja: error: WriteFile(__third_party_connectedhomeip_src_setup_payload_gen_additional_data_payload_buildconfig___third_party_connectedhomeip_config_esp32_toolchain_esp32__rule.rsp): Unable to create file. File name too long
ninja: build stopped: .
...
```

Are there any recommendations for how to avoid this issue? Thanks!

jadhavrohit924, 2023-01-17:
Could you please provide the steps that you have followed?? 
What is your working environment windows/linux/mac??
Also try clean build by removing build, sdkconfig, sdkconfig.old.

dhrishi, 2023-01-19:
@abrender Can you please provide the requested details?

shubhamdp, 2023-01-19:
@abrender export.sh scripts must be sourced and not executed
> ./export.sh

`source ./export.sh` or `. ./export.sh`

shubhamdp, 2023-01-19:
Can you please tell us which host you are using"

abrender, 2023-01-19:
> export.sh scripts must be sourced and not executed

Yep, sorry for the typo in my response. I sourced it correctly when actually running.

Host is Ubuntu 22.04.1 LTS (64 bit x86), 24G RAM (32G swap), CPU is Intel(R) Core(TM) i7-6770HQ.

I've switched to the "shallow clone" method and still have the same error.

I've created `reproduce.sh` and attached the logs:

```
abrender@d2:~/esp/matter-test$ cat reproduce.sh 
#!/bin/bash

set -e  # Exit when any command fails.
set -x  # echo the commands being run.

git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf; git checkout v4.4.2; git submodule update --init --recursive;
./install.sh
. ./export.sh
cd ..

git clone --depth 1 https://github.com/espressif/esp-matter.git
cd esp-matter
git submodule update --init --depth 1
./connectedhomeip/connectedhomeip/scripts/checkout_submodules.py --platform esp32 --shallow
./install.sh
. ./export.sh

cd examples/light
idf.py build

abrender@d2:~/esp/matter-test$ ./reproduce.sh > /tmp/reproduce-logs.txt 2>&1
abrender@d2:~/esp/matter-test$
```

The raw logs can be seen here: https://gist.githubusercontent.com/abrender/3ca49ff2cc06d7e3be5b61a30074f791/raw/8f7dd37e409f8a92642babd21153665ea3f8aef1/reproduce-logs.txt

Please let me know if you can't access that gist.

Thanks!

shubhamdp, 2023-01-20:
Thanks for confirming the host. I assume that you have already installed the prerequisites as mentioned [here](https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#host-setup).

Also, can you please specify the version of ninja.
```
which -a ninja
<for_each> --version
```

abrender, 2023-01-27:
Hi @shubhamdp,

Ninja 1.10.1 is installed:

```
abrender@d2:~/esp/matter-test$ ninja --version
1.10.1
abrender@d2:~/esp/matter-test$ which -a ninja
/usr/bin/ninja
/bin/ninja
abrender@d2:~/esp/matter-test$ ls -ld /bin/ninja
-rwxr-xr-x 1 root root 248056 Aug 30  2020 /bin/ninja
abrender@d2:~/esp/matter-test$ ls -ld /usr/bin/ninja
-rwxr-xr-x 1 root root 248056 Aug 30  2020 /usr/bin/ninja
abrender@d2:~/esp/matter-test$ 
```

> I assume that you have already installed the prerequisites as mentioned [here](https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#host-setup).

Yep! 

```
$ sudo apt-get install git gcc g++ pkg-config libssl-dev libdbus-1-dev \
     libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev \
     python3-pip unzip libgirepository1.0-dev libcairo2-dev libreadline-dev
[sudo] password for abrender: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
g++ is already the newest version (4:11.2.0-1ubuntu1).
gcc is already the newest version (4:11.2.0-1ubuntu1).
libavahi-client-dev is already the newest version (0.8-5ubuntu5).
libcairo2-dev is already the newest version (1.16.0-5ubuntu2).
libgirepository1.0-dev is already the newest version (1.72.0-1).
libreadline-dev is already the newest version (8.1.2-1).
pkg-config is already the newest version (0.29.2-1ubuntu3).
ninja-build is already the newest version (1.10.1-1).
python3-pip is already the newest version (22.0.2+dfsg-1).
git is already the newest version (1:2.34.1-1ubuntu1.6).
libdbus-1-dev is already the newest version (1.12.20-2ubuntu4.1).
libglib2.0-dev is already the newest version (2.72.4-0ubuntu1).
libssl-dev is already the newest version (3.0.2-0ubuntu1.7).
python3-dev is already the newest version (3.10.6-1~22.04).
unzip is already the newest version (6.0-26ubuntu3.1).
python3-venv is already the newest version (3.10.6-1~22.04).
The following packages were automatically installed and are no longer required:
  apache2-bin libaprutil1-dbd-sqlite3 libaprutil1-ldap libflashrom1 libftdi1-2
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
```

dhrishi, 2023-02-08:
@abrender Did it work at your end?

jlines, 2023-02-11:
I am encountering the same issue as abrender when running the reproduce script he posted.

[26/729] stamp obj/third_party/connectedhomeip/src/lib/support/gen_chip_version.stamp
[27/729] stamp obj/third_party/connectedhomeip/src/platform/logging/headers.stamp
ninja: error: WriteFile(__third_party_connectedhomeip_src_setup_payload_gen_additional_data_payload_buildconfig___third_party_connectedhomeip_config_esp32_toolchain_esp32__rule.rsp): Unable to create file. File name too long
ninja: build stopped: .
[708/1300] Building C object esp-idf/d...CMakeFiles/__idf_driver.dir/uart.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

My OS info
% uname -a
Linux linux 5.15.0-60-generic #66-Ubuntu SMP Fri Jan 20 14:29:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

Ninja info 
jlines@linux ‹ main › : ~/dev/esp-matter/examples/light
[1] % ninja --version
1.10.1
(base)
jlines@linux ‹ main › : ~/dev/esp-matter/examples/light
[0] % which -a ninja
/usr/bin/ninja
/bin/ninja
/home/jlines/dev/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/pigweed//ninja

dhrishi, 2023-10-14:
Unfortunately, we haven't been seeing this issue. Can @abrender or @jlines update if this is fixed at your end? If yes, how? :-)

abrender, 2023-10-14:
I have given up on this, please feel free to close this issue.

dhrishi, 2024-01-31:
If anyone is encountering this, please comment on this issue and provide details that can help us reproduce it. Closing the issue for now

closed 2024-01-31 by dhrishi

Genewolf, 2024-10-29:
Hello,
I see that this issue is still on (I think) and I got the same, but not on Ubuntu like abrender.
Instead, I'm on Windows 10, and when I use the command:
`git clone --recursive https://github.com/espressif/esp-idf.git`
it copy everything great until a certain point where I can see this type of error:
```
error: unable to create file source/third_party/freertos/portable/GCC/RISC-V/chip_specific_extensions/Pulpino_Vega_RV32M1RM/freertos_risc_v_chip_specific_extensions.h: Filename too long
error: unable to create file source/third_party/freertos/portable/GCC/RISC-V/chip_specific_extensions/RISCV_MTIME_CLINT_no_extensions/freertos_risc_v_chip_specific_extensions.h: Filename too long
error: unable to create file source/third_party/freertos/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions/freertos_risc_v_chip_specific_extensions.h: Filename too long
error: unable to create file source/third_party/freertos/portable/IAR/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions/freertos_risc_v_chip_specific_extensions.h: Filename too long
error: unable to create file source/third_party/freertos/portable/ThirdParty/Community-Supported-Ports/GCC/RISC-V/chip_specific_extensions/THEAD_RV32/freertos_risc_v_chip_specific_extensions.h: Filename too long
fatal: cannot create directory at 'source/third_party/sidewalk/apps/common/sidewalk_sdk/sal/common/sall/include/export/halo/lib/hall/appv1/network_coverage/network_diag': Filename too long
fatal: Unable to checkout '374a26a45a5b05cd87c62d9a5da04d9e6d0ed319' in submodule path 'connectedhomeip/connectedhomeip/third_party/openthread/ot-ti/third_party/ti_simplelink_sdk/repo_cc13xx_cc26xx'
fatal: Failed to recurse into submodule path 'connectedhomeip/connectedhomeip/third_party/openthread/ot-ti'
fatal: Failed to recurse into submodule path 'connectedhomeip/connectedhomeip'```
I tried to clone the git directly at C:/ but it's still too long, and even after trying to enable long paths on windows system it doesn't solve the problem.
So basically right now I can't use the esp-matter SDK for my projects, I must use the original connectedhomeip git, which happen to not have this problem (I think? At least I didn't saw it when cloning)

For the details (for reproduce it) I don't really do much manipulation, it seems like just cloning the git create this error.
Hope I can help one way or another.