espressif/esp-matter #166: Build/Project variables not defined

eos1d3, 2023-01-18:
From [Build/Project Variables](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#build-project-variables), it says `PROJECT_NAME` are `PROJECT_DIR` are available.

> `PROJECT_DIR`: Absolute path of the project directory containing the project CMakeLists. Same as the CMAKE_SOURCE_DIR variable.

I am modifying `light` example and moving `device_hal` to project directory, and found the variables are not defined yet. But `CMAKE_SOURCE_DIR` and `CMAKE_CURRENT_LIST_DIR` (undocumented in docs) are defined. But I prefer to use `PROJECT_DIR` as it makes more sense here.

Is this a bug?

shubhamdp, 2023-01-19:
`PROJECT_DIR` is defined and available in your cmake files? Can you share the snippet of what you are trying to do?

You can try adding the below to your `CMakeLists.txt` to make sure that the variable is set.
```
message(WARNING "***PROJECT_DIR:${PROJECT_DIR} ***")
```

As per the doc that you mentioned it is and idf build property to correct way to retrieve it would be
```
idf_build_get_property(project_dir, PROJECT_DIR)
message(WARNING "***PROJECT_DIR:${project_dir} ***")
```

eos1d3, 2023-01-21:
@shubhamdp 

At the beginning of CMakeLists.txt

```
cmake_minimum_required(VERSION 3.5)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
idf_build_get_property(project_dir, PROJECT_DIR)
message(WARNING "***PROJECT_DIR:${project_dir} ***")

```

And I still get"
```
CMake Warning at CMakeLists.txt:6 (message):
  ***PROJECT_DIR: ***
```

What is missing?

shubhamdp, 2023-01-27:
I tried above from `examples/light/CMakeLists.txt` and `examples/light/main/CMakeLists.txt` files.

Are you trying this in from some other place? Can you also share the project directory structure may be that would help me recreate this issue?

eos1d3, 2023-01-27:
@shubhamdp Hi, I just build the original Matter `light` example on VSCode version: 1.74.3 (Universal), macOS 12.6.3. And added these two lines to CMakeLists.txt

```
idf_build_get_property(project_dir, PROJECT_DIR)
message(WARNING "***PROJECT_DIR:${project_dir} ***")
```

PROJECT_DIR is empty and I still get

```
[0/1] Re-running CMake...
CMake Warning at CMakeLists.txt:35 (message):
  ***PROJECT_DIR: ***

```

The CMakeLists.txt is:

```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

if(NOT DEFINED ENV{ESP_MATTER_PATH})
    message(FATAL_ERROR "Please set ESP_MATTER_PATH to the path of esp-matter repo")
endif(NOT DEFINED ENV{ESP_MATTER_PATH})

if(NOT DEFINED ENV{ESP_MATTER_DEVICE_PATH})
    if("${IDF_TARGET}" STREQUAL "esp32" OR "${IDF_TARGET}" STREQUAL "")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32_devkit_c)
    elseif("${IDF_TARGET}" STREQUAL "esp32c3")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32c3_devkit_m)
    elseif("${IDF_TARGET}" STREQUAL "esp32h2")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32h2_devkit_c)
    elseif("${IDF_TARGET}" STREQUAL "esp32s3")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32s3_devkit_c)
    else()
        message(FATAL_ERROR "Unsupported IDF_TARGET")
    endif()
endif(NOT DEFINED ENV{ESP_MATTER_DEVICE_PATH})

set(PROJECT_VER "v1.0")
set(PROJECT_VER_NUMBER 1)

set(ESP_MATTER_PATH $ENV{ESP_MATTER_PATH})
set(MATTER_SDK_PATH ${ESP_MATTER_PATH}/connectedhomeip/connectedhomeip)
set(ZAP_GENERATED_PATH ${CMAKE_CURRENT_LIST_DIR}/main/zap-generated)

# This should be done before using the IDF_TARGET variable.
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
include($ENV{ESP_MATTER_DEVICE_PATH}/esp_matter_device.cmake)

idf_build_get_property(project_dir, PROJECT_DIR)
message(WARNING "***PROJECT_DIR:${project_dir} ***")

set(EXTRA_COMPONENT_DIRS
    "${ESP_MATTER_PATH}/examples/common"
    "${MATTER_SDK_PATH}/config/esp32/components"
    "${ESP_MATTER_PATH}/components"
    "${ESP_MATTER_PATH}/device_hal/device"
    ${extra_components_dirs_append})

project(light)

idf_build_set_property(CXX_COMPILE_OPTIONS "-std=gnu++14;-Os;-DCHIP_HAVE_CONFIG_H" APPEND)
idf_build_set_property(C_COMPILE_OPTIONS "-Os" APPEND)
# For RISCV chips, project_include.cmake sets -Wno-format, but does not clear various
# flags that depend on -Wformat
idf_build_set_property(COMPILE_OPTIONS "-Wno-format-nonliteral;-Wno-format-security;-Wformat=0" APPEND)

```

eos1d3, 2023-01-27:
Using terminal to build the `light` examples without using VSCode, project_dir is also not defined.

```
Executing action: all (aliases: build)
Running ninja in directory /Users/andy/esp/esp-matter/examples/light/build
Executing "ninja all"...
[0/1] Re-running CMake...
CMake Warning at CMakeLists.txt:35 (message):
  ***PROJECT_DIR: ***

```

shubhamdp, 2023-01-27:
Okay, can you try moving these lines after `project(light)`. These variables are these variables are updated after that.

eos1d3, 2023-01-27:
Just found out the following:

```
idf_build_get_property(project_dir, PROJECT_DIR)
project(light)
message(WARNING "***PROJECT_DIR:${project_dir} ***")
```

To get the IDF variables, `idf_build_get_property` and `project` must be called first. And their order is not important.  This also works:
```
project(light)
idf_build_get_property(project_dir, PROJECT_DIR)
message(WARNING "***PROJECT_DIR:${project_dir} ***")
```

So the whole CmakeLists.txt have to be reordered in order to use IDF variables. And using CMake builtin variables (i.e. CMAKE_SOURCE_DIR)  are easier without huge change.

Thanks!