espressif/esp-matter #167: Light example without RGB

eos1d3, 2023-01-18:
Hi, Just trying to make a color temperature changing light project without RGB color attribute. I have already removed the following in `app_main.cpp` But Alexa still shows `Color Capabilities` with `Color` and `Color Temperature`.

```
    /* Add additional features to the node */
    cluster_t *cluster = cluster::get(endpoint, ColorControl::Id);
    cluster::color_control::feature::hue_saturation::config_t hue_saturation_config;
    hue_saturation_config.current_hue = DEFAULT_HUE;
    hue_saturation_config.current_saturation = DEFAULT_SATURATION;
    cluster::color_control::feature::hue_saturation::add(cluster, &hue_saturation_config);
```

What is the correct way to remove RGB attribute?

jadhavrohit924, 2023-01-19:
Hi @eos1d3 
Do you want to make color temperature light without color?

jadhavrohit924, 2023-01-19:
Color temperature light has color control cluster which has primaryX, primaryY mandatory attributes which adds color capabilities.

eos1d3, 2023-01-19:
@jadhavrohit924 Right, I am making a project with color temperature light without using RGB. I use warm LEDs with 2700K and 6500K together. By changing the ratio of each, color temperature is changed.

Google and Alexa have color temperature light only. I made this with Rainmaker and color attribute never appears.

I tried EMBER_ZCL_COLOR_MODE_CURRENT_X_AND_CURRENT_Y but color attribute is still there.

```
extern "C" void app_main()
{
    esp_err_t err = ESP_OK;

    /* Initialize the ESP NVS layer */
    nvs_flash_init();

    /* Initialize driver */
    app_driver_handle_t light_handle = app_driver_light_init();
    app_driver_handle_t button_handle = app_driver_button_init();
    app_reset_button_register(button_handle);

    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

    color_temperature_light::config_t light_config;
    light_config.on_off.on_off = DEFAULT_POWER;
    light_config.on_off.lighting.start_up_on_off = nullptr;
    light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.lighting.start_up_current_level = DEFAULT_BRIGHTNESS;

    // use EMBER_ZCL_COLOR_MODE_CURRENT_X_AND_CURRENT_Y cannot help
    light_config.color_control.color_mode = EMBER_ZCL_COLOR_MODE_COLOR_TEMPERATURE;

    // removing the following line cannot help
    light_config.color_control.enhanced_color_mode = EMBER_ZCL_COLOR_MODE_COLOR_TEMPERATURE;

    light_config.color_control.color_temperature.startup_color_temperature_mireds = nullptr;
    endpoint_t *endpoint = color_temperature_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);

    /* These node and endpoint handles can be used to create/add other endpoints and clusters. */
    if (!node || !endpoint) {
        ESP_LOGE(TAG, "Matter node creation failed");
    }

    light_endpoint_id = endpoint::get_id(endpoint);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id);

    /* Add additional features to the node */
   //  cluster_t *cluster = cluster::get(endpoint, ColorControl::Id);
   //  cluster::color_control::feature::hue_saturation::config_t hue_saturation_config;
   //  hue_saturation_config.current_hue = DEFAULT_HUE;
   //  hue_saturation_config.current_saturation = DEFAULT_SATURATION;
   //  cluster::color_control::feature::hue_saturation::add(cluster, &hue_saturation_config);

    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }

    /* Starting driver with default values */

    app_driver_light_set_defaults(light_endpoint_id);

#if CONFIG_ENABLE_CHIP_SHELL
    esp_matter::console::diagnostics_register_commands();
    esp_matter::console::wifi_register_commands();
    esp_matter::console::init();
#endif
}
```

[image]

jadhavrohit924, 2023-01-19:
@eos1d3 By just removing hue and saturation, I have check it with Apple and Google. It worked as expected. Looks like there is some problem with Alexa.
[image]
[image]

eos1d3, 2023-01-19:
@jadhavrohit924 Thanks a lot! Google Home could not add new device with Test Device ID last week. The problem was fixed but it comes back 2 days ago. Can you add new device now? I can only use Alexa to test right now. (Just shared test device from Alexa to Google Home, it still failed ("Something went wrong" on screen.)

And what you did is to remove these lines, right?

```
   /* Add additional features to the node */
   cluster_t *cluster = cluster::get(endpoint, ColorControl::Id);
   cluster::color_control::feature::hue_saturation::config_t hue_saturation_config;
   hue_saturation_config.current_hue = DEFAULT_HUE;
   hue_saturation_config.current_saturation = DEFAULT_SATURATION;
   cluster::color_control::feature::hue_saturation::add(cluster, &hue_saturation_config);
```

jadhavrohit924, 2023-01-19:
@eos1d3 
> And what you did is to remove these lines, right?
Right.

> Can you add new device now? 
My Google home app version is 2.64.23.1

eos1d3, 2023-01-19:
Just update to 2.63.1.12, the latest I can find in Play Store. It still fails to add.

I find your version in apkmirror. Will try to download it. Thanks!!

eos1d3, 2023-01-19:
Using 2.64.23.1 still cannot add new device to Google Home.

Last time, this problem was caused by their server end and not the Google Home version. I could add successfully last week with older version after they fixed the problem. Now the same error screen appears.

jadhavrohit924, 2023-01-19:
Try factory resetting Nest Hub. It may work!!

eos1d3, 2023-01-19:
I do not register there and it worked last week.

I will factory reset it first. If it won't work, I will try register there.

Thanks!

dhrishi, 2023-01-19:
Please close the issue if the original concern is addressed.

eos1d3, 2023-01-19:
@dhrishi week try above solutions for pairing. The color temperature issue fixed.

Thanks!

closed 2023-01-19 by eos1d3