espressif/esp-matter #168: window covering - apple home

stefano055415, 2023-01-19:
I'm trying to add window_coverings to a bridge

I added in esp_matter_bridge.cpp

case ESP_MATTER_WINDOW_COVERING_DEVICE_TYPE_ID: {
        window_covering_device::config_t window_covering_device_config;
        bridged_device->endpoint =
            window_covering_device::add(bridged_device->endpoint, &window_covering_device_config);
        break;
    }

and with this code in Apple Home it works to open and close, but not to adjust the position

dev = esp_matter_bridge::create_device(node, endpoint::get_id(aggregator), ESP_MATTER_WINDOW_COVERING_DEVICE_TYPE_ID);

cluster_t *cluster = cluster::get(dev->endpoint, WindowCovering::Id);
		cluster::window_covering::feature::lift::config_t lift;
		cluster::window_covering::feature::tilt::config_t tilt;
		cluster::window_covering::feature::position_aware_lift::config_t position_aware_lift;
		cluster::window_covering::feature::position_aware_tilt::config_t position_aware_tilt;
		cluster::window_covering::feature::absolute_position::config_t absolute_position;
		cluster::window_covering::feature::lift::add(cluster, &lift);
		cluster::window_covering::feature::tilt::add(cluster, &tilt);
		cluster::window_covering::feature::position_aware_lift::add(cluster, &position_aware_lift);
		cluster::window_covering::feature::position_aware_tilt::add(cluster, &position_aware_tilt);
		cluster::window_covering::feature::absolute_position::add(cluster, &absolute_position);

		esp_matter::endpoint::enable(dev->endpoint);

If i try to use the same code without bridge  work fine

closed 2023-01-27 by stefano055415

mgoeller, 2023-01-27:
Did you manage to get it working with a bridge?

stefano055415, 2023-01-27:
@mgoeller  
It works now, I haven't changed anything, it was probably an Apple home problem
also tested with google home and it works fine

Add this code to esp_matter_bridge.ccp:

case ESP_MATTER_WINDOW_COVERING_DEVICE_TYPE_ID: {
        window_covering_device::config_t window_covering_device_config;
        bridged_device->endpoint =
            window_covering_device::add(bridged_device->endpoint, &window_covering_device_config);
        break;
    }

and in your main file

device = esp_matter_bridge::create_device(node, endpoint::get_id(aggregator), ESP_MATTER_WINDOW_COVERING_DEVICE_TYPE_ID);

cluster_t *cluster = cluster::get(device->endpoint, WindowCovering::Id);
		cluster::window_covering::feature::lift::config_t lift;
		cluster::window_covering::feature::position_aware_lift::config_t position_aware_lift;
		cluster::window_covering::feature::absolute_position::config_t absolute_position;
		cluster::window_covering::feature::lift::add(cluster, &lift);
		cluster::window_covering::feature::position_aware_lift::add(cluster, &position_aware_lift);
		cluster::window_covering::feature::absolute_position::add(cluster, &absolute_position);

esp_matter::endpoint::enable(device->endpoint);

dhrishi, 2023-01-27:
Thanks for the feedback @stefano055415. This is very useful :-)

JonathanWitthoeft, 2023-01-30:
@stefano055415 I have implemented the same code and I am getting "No response" for each of my bridged accessories in the Home app. Using google home it says there is a it says there is a network error. Is this still working for you?
esp-idf @ v4.3.3
esp-matter @ e5d7cac

jzwyssig, 2023-03-29:
I added a bridged window covering and it works with Google Home, but still not with Apple's Home app (HomePod & iPad are up to date with version 16.4). Strangely, the iPad shows the correct percentage of the roller shade and it is even possible to adjust the level of the shade and it is correctly synced with Google. Therefore, I guess it is still an Apple Home problem. Does it work for you?