espressif/esp-matter #169: Apple Home :  color presets in the phone do not give a callback

vvrec, 2023-01-23:
Color presets in the phone do not give a callback, and if they do, then the wrong one - attribute_id == ColorTemperatureMireds. 
Is this a known bug or am I doing something wrong?...

shubhamdp, 2023-01-23:
Hi @vvrec, could you provide more details, like what example you are trying, esp-idf and esp-matter commit SHA, target device.

vvrec, 2023-01-23:
idf (4.4.3), esptool.py v3.3.2, last update esp-matter, esp32, esp32-c3 . 
Light example with no modifications , lighting-app example in connectedhomeip folder.
Same result. 

I can't catch callback hue and saturation when switch color presets -  is returned only color temperature with the outdated value. There is also no feedback in the application interface on the phone. Everything works correctly only when I make color in colorwheel ui directly.

[image]

PSONALl, 2023-01-24:
@vvrec We have color temperature device type in light example and ColorTemperatureMireds is a mandatory attribute for color-temperature light. Attribute change callback for ColorTemperatureMireds gets called for shades in color-temperature light. 
This seems to be an issue with the preset colors which we also have observed that, the colors from preset colors in Apple Home UI are not getting correctly reflected in code

Whereas I recieve color temperature mireds callback when the white, sky blue and orange shades are reflected on led also 
I can also recieve hue saturation callbacks when corresponding colors are reflected 

So It seems to be the issue with Apple Home UI for preset colors, we are checking with Apple about this issue

vvrec, 2023-01-24:
I did the forced saturation and hue update directly from the ColorTemperatureMireds callback. At least it was possible to work with this.

But when I touch the presets in the UI, sometimes I get the correct double callback (hue, saturation) and sometimes just the temperature. Randomly. 
 
In addition, when only the temperature callback is caught - there are also two options: the outdated temperature value is caught, or the value that correctly matches the color in the preset interface. also Randomly. 

I also noticed that the presets work more correctly when they have the most saturated colors.

PSONALl, 2023-01-25:
> But when I touch the presets in the UI, sometimes I get the correct double callback (hue, saturation) and sometimes just the temperature. Randomly.
> In addition, when only the temperature callback is [...]

@vvrec Thanks for reporting your observations, I have noticed the same results, If you will remove the hue saturation feature from light then presets work correctly with all colors supported by color temperature light

vvrec, 2023-01-31:
> Thanks for reporting your observations, I have noticed the same results, If you will remove the hue saturation feature from light then presets work correctly with all colors supported by color temperature light

Yes - without colorwheel works correctly.
So - this is not a problem with the matter framework and it remains only to wait for a fix from Apple?

VaishaliAvhale, 2023-11-02:
I tried it just now using iPadOS 17.1, and HomePod 17.1 (21K69), and it's working fine on my end.