espressif/esp-matter #170: Help on `spi_flash_mmap.h` not found?

loicmorvan, 2023-01-23:
I'm on macOS Monterey 12.6.2 (21G320).
I'm trying to build the example `light` after having followed the setup guide found [here](https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html).

It seems almost everything works as expected, the build task does many things, but ends up with this error:

```
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/DiagnosticDataProviderImpl.cpp:38:10: fatal error: spi_flash_mmap.h: No such file or directory
   38 | #include "spi_flash_mmap.h"
      |          ^~~~~~~~~~~~~~~~~~
```

Can you help me understand what's happening here? 😄

shubhamdp, 2023-01-23:
Hi @loicmorvan Can you share the esp-idf and esp-matter commit id that you are using

dhrishi, 2023-01-23:
@loicmorvan This SDK works with v4.4.3 release of ESP-IDF. v5.0 release isn't supported yet. Please try with v4.4.3 and you should not see the error.
Reference documentation: https://github.com/espressif/esp-matter#supported-esp-idf-and-connectedhomeip-versions

closed 2023-01-23 by loicmorvan