espressif/esp-matter #171: err = b from CHIP on attributes

jonsmirl, 2023-01-23:
Any idea what causes this? It is transitory and will go away after a while.

I (87517) chip[IM]: Received status response, status is 0x00
I (87527) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0005's Attribute 0xFFFC is 0 **********
I (87537) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0005's Attribute 0xFFFD is 4 **********
I (87547) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0005's Attribute 0x0000 is 0 **********
I (87557) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0005's Attribute 0x0001 is 0 **********
I (87567) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0005's Attribute 0x0002 is 0 **********
I (87577) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0005's Attribute 0x0003 is 0 **********
I (87587) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0005's Attribute 0x0004 is 0 **********
E (87607) chip[DMG]: Error retrieving data from clusterId: 0x0000_0005, err = b

shubhamdp, 2023-01-27:
This error may be caused when reading Attribute `0x0005` of scene cluster which is optional in specification.
I tried to recreate this using `./chip-tool scenes read-by-id "0xfffc,0xfffd,1,2,3,4,5" 6 1` but I could not.

Can you check if adding the optional attribute solves this for Google home?
```
cluster_t *scene_cluster = cluster::get(endpoint, Scenes::Id);
attribute_t *last_configured_by = attribute::create(scene_cluster, 0x0005, ATTRIBUTE_FLAG_NULLABLE, esp_matter_nullable_uint64(nullable<uint64_t>()));
```

jonsmirl, 2023-01-28:
I get many, many of these err b
~~~
I (108127) app_main: Free 63284
I (108627) app_main: Free 63284
I (109127) app_main: Free 63304
I (109487) chip[EM]: >>> [E:61309r M:23965279] (U) Msg RX from 0:649B121CB1D7A4A8 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (109487) chip[IN]: CASE Server received Sigma1 message. Starting handshake. EC 0x3fc9af90
I (109497) chip[IN]: CASE Server disabling CASE session setups
I (109507) chip[EM]: <<< [E:61309r M:56296145 (Ack:23965279)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (109517) chip[IN]: (U) Sending msg 56296145 to IP address 'UDP:[FE80::1E53:F9FF:FEB3:3B54%st1]:5540'
I (109527) chip[SC]: Received Sigma1 msg
I (109537) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x00000000810609B2
I (110037) chip[EM]: <<< [E:61309r M:56296146 (Ack:23965279)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
I (110037) chip[IN]: (U) Sending msg 56296146 to IP address 'UDP:[FE80::1E53:F9FF:FEB3:3B54%st1]:5540'
I (110047) chip[SC]: Sent Sigma2 msg
E (110057) chip[DL]: Long dispatch time: 568 ms, for event type 3
I (110067) app_main: Free 58972
I (110107) chip[EM]: >>> [E:61309r M:23965280 (Ack:56296146)] (U) Msg RX from 0:649B121CB1D7A4A8 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (110207) chip[EM]: >>> [E:61309r M:23965281 (Ack:56296146)] (U) Msg RX from 0:649B121CB1D7A4A8 [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
I (110207) chip[EM]: <<< [E:61309r M:56296147 (Ack:23965281)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (110227) chip[IN]: (U) Sending msg 56296147 to IP address 'UDP:[FE80::1E53:F9FF:FEB3:3B54%st1]:5540'
I (110237) chip[SC]: Received Sigma3 msg
E (110247) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
I (111267) NimBLE: GAP procedure initiated: advertise; 
I (111267) NimBLE: disc_mode=0
I (111267) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (111277) NimBLE: 

I (111277) chip[EM]: <<< [E:61309r M:56296148 (Ack:23965281)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (111297) chip[IN]: (U) Sending msg 56296148 to IP address 'UDP:[FE80::1E53:F9FF:FEB3:3B54%st1]:5540'
I (111307) chip[SC]: SecureSession[0x3fc998d8]: Moving from state 'kEstablishing' --> 'kActive'
I (111317) chip[IN]: CASE Session established to peer: <0000000097985A03, 1>
I (111327) chip[IN]: CASE Server enabling CASE session setups
E (111327) chip[DL]: Long dispatch time: 1124 ms, for event type 3
I (111337) chip[EM]: >>> [E:61310r M:169145165] (S) Msg RX from 1:0000000097985A03 [DC52] --- Type 0001:03 (IM:SubscribeRequest)
I (111347) chip[DMG]: Final negotiated min/max parameters: Min = 0s, Max = 120s
I (111357) chip[EM]: >>> [E:61309r M:23965282 (Ack:56296148)] (U) Msg RX from 0:649B121CB1D7A4A8 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (111367) NimBLE: GAP procedure initiated: stop advertising.

I (111377) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x001D's Attribute 0xFFFC is 0 **********
E (111387) chip[DMG]: Error retrieving data from clusterId: 0x0000_001F, err = b
I (111397) chip[EM]: <<< [E:61310r M:85848652 (Ack:169145165)] (S) Msg TX to 1:0000000097985A03 [DC52] --- Type 0001:05 (IM:ReportData)
I (111407) chip[IN]: (S) Sending msg 85848652 on secure session with LSID: 41766
I (111417) app_main: Free 60416
I (111427) chip[EM]: >>> [E:61310r M:169145166 (Ack:85848652)] (S) Msg RX from 1:0000000097985A03 [DC52] --- Type 0001:01 (IM:StatusResponse)
I (111437) chip[IM]: Received status response, status is 0x00
I (111447) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x001F's Attribute 0xFFFC is 0 **********
I (111457) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0028's Attribute 0xFFFC is 0 **********
I (111467) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0028's Attribute 0xFFFD is 1 **********
I (111477) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0028's Attribute 0x0005 is  **********
E (111487) chip[DMG]: Error retrieving data from clusterId: 0x0000_0028, err = b
I (111497) chip[EM]: <<< [E:61310r M:85848653 (Ack:169145166)] (S) Msg TX to 1:0000000097985A03 [DC52] --- Type 0001:05 (IM:ReportData)
I (111507) chip[IN]: (S) Sending msg 85848653 on secure session with LSID: 41766
I (111537) chip[EM]: >>> [E:61310r M:169145167 (Ack:85848653)] (S) Msg RX from 1:0000000097985A03 [DC52] --- Type 0001:01 (IM:StatusResponse)
I (111537) chip[IM]: Received status response, status is 0x00
I (111547) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0xFFFC is 0 **********
I (111557) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0xFFFD is 1 **********
I (111567) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is 0 **********
E (111577) chip[DMG]: Error retrieving data from clusterId: 0x0000_0030, err = b
I (111587) chip[EM]: <<< [E:61310r M:85848654 (Ack:169145167)] (S) Msg TX to 1:0000000097985A03 [DC52] --- Type 0001:05 (IM:ReportData)
I (111597) chip[IN]: (S) Sending msg 85848654 on secure session with LSID: 41766
I (111637) chip[EM]: >>> [E:61310r M:169145168 (Ack:85848654)] (S) Msg RX from 1:0000000097985A03 [DC52] --- Type 0001:01 (IM:StatusResponse)
[... cut ...]
I (113537) chip[EM]: <<< [E:61310r M:85848672 (Ack:169145185)] (S) Msg TX to 1:0000000097985A03 [DC52] --- Type 0001:04 (IM:SubscribeResponse)
I (113547) chip[IN]: (S) Sending msg 85848672 on secure session with LSID: 41766
I (113567) app_main: Free 61516
I (113577) chip[EM]: >>> [E:61310r M:169145186 (Ack:85848672)] (S) Msg RX from 1:0000000097985A03 [DC52] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (114067) app_main: Free 63304
I (114567) app_main: Free 63304
I (115067) app_main: Free 63304
I (115567) app_main: Free 63304
I (116067) app_main: Free 63304
I (116567) app_main: Free 63304
~~~

jonsmirl, 2023-01-28:
Another interaction from Google Home that does not generate any error b
So what is the pattern?
~~~
 (109903) app_main: Free 63140
I (110363) chip[EM]: >>> [E:61317r M:38411751] (S) Msg RX from 1:0000000097985A03 [DC52] --- Type 0001:08 (IM:InvokeCommandRequest)
I (110363) esp_matter_command: Received command 0x0004 for enpoint 0x0001's cluster 0x00000008
I (110373) chip[ZCL]: 0x3c12fb40MOVE_TO_LEVEL_WITH_ON_OFF 4 null 0 0
I (110383) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x0000 is 1 **********
I (110393) chip[ZCL]: Setting on/off to 0x3c12fcb0 due to level change
I (110403) chip[ZCL]: On/Off set value: 1 1
I (110403) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 0 **********
I (110413) chip[ZCL]: Toggle on/off from 0 to 1
I (110423) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0xFFFC is 1 **********
I (110433) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x4001 is 0 **********
I (110443) chip[ZCL]: On Command - OffWaitTime :  0
I (110443) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x4002 is 0 **********
I (110463) chip[ZCL]: On/Toggle Command - Stop Timer
I (110463) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x4000 is 1 **********
I (110473) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 1 **********
I (110483) app_main: Free 59956
I (110493) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0005's Attribute 0x0003 is 0 **********
I (110503) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0xFFFC is 3 **********
I (110513) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x4000 is 1 **********
I (110523) chip[EM]: <<< [E:61317r M:151345120 (Ack:38411751)] (S) Msg TX to 1:0000000097985A03 [DC52] --- Type 0001:09 (IM:InvokeCommandResponse)
I (110533) chip[IN]: (S) Sending msg 151345120 on secure session with LSID: 60740
E (110543) chip[DL]: Long dispatch time: 183 ms, for event type 3
I (110553) chip[EM]: >>> [E:61318r M:38411752] (S) Msg RX from 1:0000000097985A03 [DC52] --- Type 0001:08 (IM:InvokeCommandRequest)
I (110563) esp_matter_command: Received command 0x0001 for enpoint 0x0001's cluster 0x00000006
I (110573) chip[ZCL]: On/Off set value: 1 1
I (110573) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 1 **********
I (110583) chip[ZCL]: On/off already set to new value
I (110593) chip[EM]: <<< [E:61318r M:151345121 (Ack:38411752)] (S) Msg TX to 1:0000000097985A03 [DC52] --- Type 0001:09 (IM:InvokeCommandResponse)
I (110603) chip[IN]: (S) Sending msg 151345121 on secure session with LSID: 60740
I (110623) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0005's Attribute 0x0003 is 0 **********
I (110623) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 1 **********
I (110633) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x4000 is 1 **********
I (110653) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x4002 is 0 **********
I (110663) chip[EM]: <<< [E:14107i M:151345122] (S) Msg TX to 1:0000000097985A03 [DC52] --- Type 0001:05 (IM:ReportData)
I (110673) chip[IN]: (S) Sending msg 151345122 on secure session with LSID: 60740
I (110683) chip[DMG]: Refresh Subscribe Sync Timer with min 0 seconds and max 120 seconds
I (110683) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x0000 is 1 **********
I (110693) chip[ZCL]: Event: move from 1
I (110703) chip[ZCL]:  to 4 
I (110703) chip[ZCL]: (diff +1)
I (110713) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x0000 is 4 **********
I (110723) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x000F is 0 **********
I (110733) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0005's Attribute 0x0003 is 0 **********
I (110743) chip[ZCL]: Setting on/off to 0x3c12fcb0 due to level change
I (110753) chip[ZCL]: On/Off set value: 1 1
I (110753) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 1 **********
I (110763) chip[ZCL]: On/off already set to new value
I (110773) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x0001 is 0 **********
I (110783) chip[EM]: >>> [E:61317r M:38411753 (Ack:151345120)] (S) Msg RX from 1:0000000097985A03 [DC52] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (110793) chip[EM]: >>> [E:61318r M:38411754 (Ack:151345121)] (S) Msg RX from 1:0000000097985A03 [DC52] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (110813) chip[EM]: >>> [E:14107i M:38411755 (Ack:151345122)] (S) Msg RX from 1:0000000097985A03 [DC52] --- Type 0001:01 (IM:StatusResponse)
I (110823) chip[IM]: Received status response, status is 0x00
I (110833) chip[EM]: <<< [E:14107i M:151345123 (Ack:38411755)] (S) Msg TX to 1:0000000097985A03 [DC52] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (110843) chip[IN]: (S) Sending msg 151345123 on secure session with LSID: 60740
I (110853) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0005's Attribute 0x0003 is 0 **********
I (110863) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x0000 is 4 **********
[... cut ...]
I (111533) chip[EM]: >>> [E:14110i M:38411760 (Ack:151345130)] (S) Msg RX from 1:0000000097985A03 [DC52] --- Type 0001:01 (IM:StatusResponse)
I (111543) chip[IM]: Received status response, status is 0x00
I (111553) chip[EM]: <<< [E:14110i M:151345131 (Ack:38411760)] (S) Msg TX to 1:0000000097985A03 [DC52] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (111563) chip[IN]: (S) Sending msg 151345131 on secure session with LSID: 60740
I (112033) app_main: Free 63148
I (112533) app_main: Free 63148
I (113033) app_main: Free 63148
I (113533) app_main: Free 63128

~~~

jonsmirl, 2023-01-28:
Why this one, it read a single attribute...
~~~
I (112187) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x001D's Attribute 0xFFFC is 0 **********
E (112197) chip[DMG]: Error retrieving data from clusterId: 0x0000_001D, err = b
~~~

shripad621git, 2023-02-15:
It is normal to encounter `CHIP_ERROR_NO_MEMORY` during chunked reads.

When commissioner performs a wildcard read,it reads all the attributes from all the clusters. Due to this the chunk allocated to read the data gets filled up and remaining attributes are read in the next chunk.

Here's the related discussion on the same issue in connectedhomeip repository https://github.com/project-chip/connectedhomeip/issues/24481

closed 2023-02-16 by dhrishi