espressif/esp-matter #172: removing device from Google home

jonsmirl, 2023-01-24:
When I remove a device form Google Home and the commissioned fabric count goes to zero, it should reenter BLE advertising mode so that it can be commissioned again.

dhrishi, 2023-01-25:
@jonsmirl The spec does not mandate the device to enter the commissioning mode after it is removed from the last fabric. It will restart the announcement (on the IP network) after reboot. If you want, you can use the open commissioning window API to restart the announcement in your app.

jonsmirl, 2023-01-25:
The only place is was commissioned was on Google Home.  Does the open commissioning window API work if the device is not on any fabric?

What I did was commission a device with Google Home, then I removed the device from Google Home and lost access to it. How do I regain access? Is it in a mode similar to BLE advertising, but on the wifi network?

dhrishi, 2023-01-25:
>Does the open commissioning window API work if the device is not on any fabric?

It should, ideally. In the factory reset state, after bootup, it opens the commissioning window.

>What I did was commission a device with Google Home, then I removed the device from Google Home and lost access to it. How do I regain access? Is it in a mode similar to BLE advertising, but on the wifi network?

The device has already the network credentials installed. So, it announces its presence over the IP network. The commissioners have to check on all available transports and hence the original QR code (that mentions device is on BLE) should work fine in this case too.

You can try out the steps below to quickly verify that:
1. Commission to Google Home using the QR code
2. Uncommission the device from the Google Home
3. Reboot the device (as we do not begin advertisement automatically)
4. Commission to Google Home using the same QR code (this commissioning would be over the IP network and not BLE)

jonsmirl, 2023-01-25:
Why the need for the reboot?

dhrishi, 2023-01-25:
As mentioned above, currently, we do not restart announcement for commissioning after the last fabric is removed. It will be done on reboot

jonsmirl, 2023-01-25:
That has side effect of making the human go find the device and hit a button on it.

dhrishi, 2023-01-25:
Yes. That is the current implementation and the comment was for quickly verifying it. Or else

>If you want, you can use the open commissioning window API to restart the announcement in your app.

jonsmirl, 2023-01-27:
I verified that this works with the reboot in the middle. This still needs some fixing, customers are not going to like that reboot.

dhrishi, 2023-02-08:
@jonsmirl We've added the support for this. Please pull the latest codebase and refer the light example: https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L63-L78

Please give it a try and close the issue if it works as expected

dhrishi, 2023-02-13:
Fixed with https://github.com/espressif/esp-matter/commit/3e8ea13b09b0e328c785a36c77f5da5f5de82b02

closed 2023-02-13 by dhrishi