espressif/esp-matter #173: ESP32S3 with W5500 LAN Chipset

mgoeller, 2023-01-24:
Our demo board with W5500 and the built-in Ethernet library from ESP-IDF works perfectly.

However if we start matter, it fails to initialise the chip stack:
```E (831) esp_matter_core: Failed to initialize CHIP stack
E (831) esp_matter_core: Error initializing matter
E (831) matter_init: Matter start failed: -1```

Any idea why that could be? Is there any way to run matter just for pairing and control, without commissioning a network connection?

jonsmirl, 2023-01-24:
Matter is loading into SRAM and you have run out. We are also struggling with forcing Matter to load into PSRAM.

dhrishi, 2023-01-25:
@mgoeller Can you try moving the NimBLE (BLE host) and mbedTLS to PSRAM. This should free up some internal memory. Steps:

```
idf.py menuconfig

Component config -> ESP32S3-Specific -> Support for external, SPI-connected RAM

Component config -> Bluetooth -> NimBLE Options -> Memory allocation strategy -> External SPIRAM

Component config -> mbedTLS -> Memory allocation strategy -> External SPIRAM
```

> Is there any way to run matter just for pairing and control, without commissioning a network connection?

Do you mean not doing the "network/Wi-Fi connection" and carrying out commissioning on the IP network directly?

jonsmirl, 2023-01-25:
That is not enough to fix this issue. You need to also move BSS / DATA into PSRAM.

mgoeller, 2023-01-25:
> Do you mean not doing the "network/Wi-Fi connection" and carrying out commissioning on the IP network directly?

Correct. That is exactly what I am looking for. However I don't think the problem has to do with PSRAM. Because the WiFi example runs fine.

The problem is rather:
In my lan_start() routine I called the function esp_netif_init(), which should only be called once. This function is also called in PlatformManagerImpl.cpp via Matter. Once out-commenting the line in my code, then calling esp_matter::start() first, and afterwards initialising ethernet via my routine, I successfully get an IP. This only works when wifi is deactivated in the matter settings in menuconfig.

Even though in my understanding, Bluetooth is not needed when using a fixed Ethernet connection (since no comissioning of the WiFi settings), I can't disable Bluetooth in menuconfig. Otherwise the code doesn't compile.

Leaving Bluetooth activated, I **can** pair the accessory with Google Home (via Bluetooth), however get an immediate "No Response" in the App afterwards. With Apple Home pairing doesn't work at all.

dhrishi, 2023-01-30:
>Even though in my understanding, Bluetooth is not needed when using a fixed Ethernet connection (since no comissioning of the WiFi settings), I can't disable Bluetooth in menuconfig. Otherwise the code doesn't compile.

@mgoeller We have fixed the issue. Please pull the latest codebase and then disable Bluetooth from menuconfig. You should be able to compile and you should have significant amount of memory.

closed 2023-03-01 by dhrishi