espressif/esp-matter #174: Alexa and Switch example not able to pair

eos1d3, 2023-01-24:
Using `light_switch` example with ESP32. Echo 4 gen (latest firmware).
After device is trying to connect to the given WiFi, Alexa app finally shows "Something went wrong". Tests had be done many times without any success. 

But using the same environment, but with Google Home (latest) and Nest Hub 2 (latest firmware), there is no pairing issue.

And using `light` example with the same Alexa configuration, it never has any pairing problem.

This is the log:

```
I (58495) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (58495) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (58515) chip[EM]: >>> [E:36728r M:124065800] (S) Msg RX from 1:FFFFFFFB00000000 [14EC] --- Type 0001:08 (IM:InvokeCommandRequest)
I (58525) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is 14 **********
I (58535) chip[EM]: <<< [E:36728r M:6607458] (S) Msg TX to 1:FFFFFFFB00000000 [14EC] --- Type 0001:09 (IM:InvokeCommandResponse)
I (58545) chip[IN]: (S) Sending msg 6607458 on secure session with LSID: 36530
I (58555) NimBLE: GATT procedure initiated: indicate; 
I (58565) NimBLE: att_handle=14

I (58635) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (58635) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (58645) chip[EM]: >>> [E:36729r M:124065801] (S) Msg RX from 1:FFFFFFFB00000000 [14EC] --- Type 0001:08 (IM:InvokeCommandRequest)
I (58655) chip[NP]: ESP NetworkCommissioningDelegate: SSID: 999A
W (58665) wifi:Haven't to connect to a suitable AP now!
I (58665) chip[DL]: WiFi station mode change: Enabled -> Disabled
I (58725) chip[DL]: WiFi station mode change: Disabled -> Enabled
W (58725) wifi:Haven't to connect to a suitable AP now!
I (58725) chip[DL]: Attempting to connect WiFi station interface
I (58735) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (58745) chip[DL]: Done driving station state, nothing else to do...
W (58755) wifi:Haven't to connect to a suitable AP now!
I (58755) chip[DL]: Attempting to connect WiFi station interface
E (58755) wifi:sta is connecting, return error
E (58765) chip[DL]: esp_wifi_connect() failed: ESP_ERR_WIFI_CONN
W (58765) wifi:Haven't to connect to a suitable AP now!
I (58775) chip[DL]: Attempting to connect WiFi station interface
E (58785) wifi:sta is connecting, return error
E (58785) chip[DL]: esp_wifi_connect() failed: ESP_ERR_WIFI_CONN
W (58795) wifi:Haven't to connect to a suitable AP now!
E (58795) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (61145) NimBLE: GATT procedure initiated: indicate; 
I (61145) NimBLE: att_handle=14

I (61285) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (62115) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (62115) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (62115) wifi:Haven't to connect to a suitable AP now!
I (62135) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
I (62135) chip[DL]: Next WiFi station reconnect in 100 ms
I (62145) chip[DL]: Done driving station state, nothing else to do...
W (62145) wifi:Haven't to connect to a suitable AP now!
E (62155) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (62155) wifi:Haven't to connect to a suitable AP now!
E (62175) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (62245) wifi:Haven't to connect to a suitable AP now!
I (62245) chip[DL]: Attempting to connect WiFi station interface
I (62245) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (62265) chip[DL]: Done driving station state, nothing else to do...
W (62265) wifi:Haven't to connect to a suitable AP now!
E (62275) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (63855) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (65625) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (65625) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (65625) wifi:Haven't to connect to a suitable AP now!
I (65645) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
I (65645) chip[DL]: Next WiFi station reconnect in 100 ms
I (65655) chip[DL]: Done driving station state, nothing else to do...
W (65655) wifi:Haven't to connect to a suitable AP now!
E (65665) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (65665) wifi:Haven't to connect to a suitable AP now!
[... 133 lines cut ...]
I (89595) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (90245) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (90245) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (90245) wifi:Haven't to connect to a suitable AP now!
I (90265) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
I (90265) chip[DL]: Done driving station state, nothing else to do...
W (90275) wifi:Haven't to connect to a suitable AP now!
E (90275) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (90285) wifi:Haven't to connect to a suitable AP now!
E (90285) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (92095) NimBLE: GATT procedure initiated: indicate; 
I (92095) NimBLE: att_handle=14

I (92155) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (94635) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (97135) NimBLE: GATT procedure initiated: indicate; 
I (97135) NimBLE: att_handle=14

I (97245) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (99765) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (102265) NimBLE: GATT procedure initiated: indicate; 
I (102265) NimBLE: att_handle=14

I (102325) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
```

eos1d3, 2023-01-24:
Same problem for ESP32C3.

I (29700) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (29700) NimBLE: GATT procedure initiated: indicate; 
I (29710) NimBLE: att_handle=14

I (29920) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (29920) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (29940) chip[EM]: >>> [E:36746r M:62411075] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (29950) esp_matter_command: Received command 0x0000 for enpoint 0x0000's cluster 0x00000030
I (29950) chip[FS]: GeneralCommissioning: Received ArmFailSafe (600s)
I (29960) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is 3 **********
I (29980) chip[EM]: <<< [E:36746r M:266285657] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (29990) chip[IN]: (S) Sending msg 266285657 on secure session with LSID: 18543
I (29990) NimBLE: GATT procedure initiated: indicate; 
I (30000) NimBLE: att_handle=14

I (30060) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (30060) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (30070) chip[EM]: >>> [E:36747r M:62411076] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (32570) NimBLE: GATT procedure initiated: indicate; 
I (32570) NimBLE: att_handle=14

I (32620) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (33460) chip[DL]: WIFI_EVENT_SCAN_DONE
I (33460) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is 21 **********
I (33470) chip[EM]: <<< [E:36747r M:266285658] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (33480) chip[IN]: (S) Sending msg 266285658 on secure session with LSID: 18543
I (33490) NimBLE: GATT procedure initiated: indicate; 
I (33500) NimBLE: att_handle=14

I (33660) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (33660) NimBLE: GATT procedure initiated: indicate; 
I (33670) NimBLE: att_handle=14

I (34020) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (34020) NimBLE: GATT procedure initiated: indicate; 
I (34030) NimBLE: att_handle=14

I (34380) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (34380) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (34390) NimBLE: GATT procedure initiated: indicate; 
I (34390) NimBLE: att_handle=14

I (34560) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (34560) NimBLE: GATT procedure initiated: indicate; 
I (34570) NimBLE: att_handle=14

I (34690) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (37030) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (39530) NimBLE: GATT procedure initiated: indicate; 
I (39530) NimBLE: att_handle=14

I (39600) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (42160) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (44660) NimBLE: GATT procedure initiated: indicate; 
I (44660) NimBLE: att_handle=14

I (44770) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (47240) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (49750) NimBLE: GATT procedure initiated: indicate; 
I (49750) NimBLE: att_handle=14

I (49810) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (52280) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (54790) NimBLE: GATT procedure initiated: indicate; 
I (54790) NimBLE: att_handle=14

I (54850) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (55210) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
[... 351 lines cut ...]
I (70600) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 5941805496A814EC-00BBD543277CE8E0.
I (70610) chip[DIS]: mDNS service published: _matter._tcp
I (70610) ROUTE_HOOK: Hook already installed on netif, skip...
I (70630) chip[SVR]: Cannot load binding table: a0
I (69290) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 1 ) indicate = 0
I (70640) chip[DL]: CHIPoBLE unsubscribe received
I (70640) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
I (70510) ROUTE_HOOK: Received RIO
I (70650) ROUTE_HOOK: prefix FD9A:2BCC:F4AD:1:: lifetime 1800

I (70640) chip[BLE]: Releasing end point's BLE connection back to application.
I (70670) chip[EM]: >>> [E:18466r M:57412326 (Ack:113195853)] (U) Msg RX from 0:39F280C47BA7AA2B [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (71300) chip[EM]: >>> [E:18467r M:231786078] (S) Msg RX from 1:0098D2FE954E9A81 [14EC] --- Type 0001:02 (IM:ReadRequest)
I (71310) chip[EM]: <<< [E:18467r M:177165391 (Ack:231786078)] (S) Msg TX to 1:0098D2FE954E9A81 [14EC] --- Type 0001:05 (IM:ReportData)
I (71320) chip[IN]: (S) Sending msg 177165391 on secure session with LSID: 18544
[... cut ...]
I (72490) app_main: Interface IP Address Changed
I (72490) chip[DIS]: Updating services using commissioning mode 0
I (72500) chip[DIS]: CHIP minimal mDNS started advertising.
I (72510) chip[DIS]: Advertise operational node 5941805496A814EC-00BBD543277CE8E0
I (72520) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 5941805496A814EC-00BBD543277CE8E0.
I (72540) chip[DIS]: mDNS service published: _matter._tcp
I (72540) ROUTE_HOOK: Hook already installed on netif, skip...

dhrishi, 2023-01-25:
@eos1d3 The `light_switch` example has the OnOff cluster client which can be used only after it is bound to a Matter node with an OnOff cluster server. In general, the ecosystems do not support the binding feature right now. And hence do not support this device type too, resulting into this error.

Please close the issue if your question is answered.

eos1d3, 2023-01-25:
@dhrishi Hi, I know what you are talking about current binding issue. Google also states it is not yet available. 

I am not talking about this issue. What I mentioned is that Alexa cannot ADD the switch device while Google Home can add it without error. 

[image]

shubhamdp, 2023-01-27:
@eos1d3 For first scenario, device is trying to connect to `SSID: 999A` and it is unable to find that AP, concluding based on following is logged multiple times.
```
W (76285) wifi:Haven't to connect to a suitable AP now!
```

For second case (esp32c3), CASE session is establishment was successful, and I am not seeing commissioning complete event though, complete logs would help.
```
I (70530) chip[IN]: CASE Session established to peer: <0098D2FE954E9A81, 1>
```

It would be helpful if you paste the logs in a file and attach that file or use the `<details>` and `<summary>` tags in markdown.

eos1d3, 2023-01-27:
@shubhamdp I just retested again using the same configuration. (Echo 4 gen, ESP32 DevKit_V4)

1. Rebuild (set-target, build, flash, monitor) Matter example, `light_switch`
2. Use Alexa (Android) to add the device
3. "Something went wrong" again

Then:
1. Rebuild Matter example, `light`
2. Use Alexa (Android) to add the device
3. The device is added successfully

Same router, same ESP32 board, same WiFi SSID, same locations for router and device, same mobile phone. So it is obvious that `light_switch` has problem.

Actually `light` example works well with Alexa. I have some other boards and none has this problem.

For second scenario, it is the complete log and there is no more message for a while. I can test it again if needed. But I think fixing the first issue will solve all. Both are related to the same problem.

PSONALl, 2023-01-27:
@eos1d3 I have observerd the same issue because device types with client clusters are not supported yet by Amazon

eos1d3, 2023-01-27:
So just close this issue and wait for Alexa update. Many thanks to all!

closed 2023-01-27 by eos1d3