espressif/esp-matter #175: Alexa & Google Home connection issue

stefano055415, 2023-01-25:
I have a problem connecting with google home and alexa

example/light:
It works with Apple
It works with Alexa
It doesn't work with Google home

example/ble_bridge:
It works with Apple
It doesn't work with Alexa
It doesn't work with Google home

this is the log with google home:

I (25157) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (25167) NimBLE: GATT procedure initiated: indicate; 
I (25167) NimBLE: att_handle=14

I (25257) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (26907) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 1 ) indicate = 0
I (26917) chip[DL]: CHIPoBLE unsubscribe received
I (26917) chip[BLE]: Releasing end point's BLE connection back to application.
I (26917) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
W (28637) ROUTE_HOOK: Ignore invalid ICMP packet
W (49317) ROUTE_HOOK: Ignore invalid ICMP packet
I (140437) chip[FS]: Fail-safe timer expired
E (140437) chip[SVR]: Failsafe timer expired
I (140437) chip[SC]: SecureSession[0x3fca1300]: Moving from state 'kActive' --> 'kPendingEviction'
E (140447) chip[SVR]: Commissioning failed (attempt 1): 32
I (140457) chip[DIS]: Updating services using commissioning mode 1
I (140457) chip[DIS]: CHIP minimal mDNS started advertising.
I (140467) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (140477) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: CE9DC169F755744F.
I (140497) chip[DIS]: mDNS service published: _matterc._udp
E (140497) chip[ZCL]: OpCreds: Got FailSafeTimerExpired
E (140497) chip[ZCL]: OpCreds: Proceeding to FailSafeCleanup on fail-safe expiry!
I (140507) chip[TS]: Pending Last Known Good Time: 2023-01-20T12:14:13
I (140527) chip[TS]: Previous Last Known Good Time: 2023-01-20T12:14:13
I (140527) chip[TS]: Reverted Last Known Good Time to previous value
I (140527) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is 0 **********
I (140547) APP DEVICE: Commissioning failed, fail safe timer expired
E (140547) chip[DL]: Long dispatch time: 111 ms, for event type 32782
I (140557) chip[FS]: Fail-safe cleanly disarmed
I (140567) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (140567) NimBLE: GAP procedure initiated: advertise; 
I (140577) NimBLE: disc_mode=2
I (140577) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (140587) NimBLE: 

I (140597) chip[DL]: CHIPoBLE advertising started
I (140597) APP DEVICE: Commissioning window opened

this is the log with alexa:

I (252497) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (253547) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 1 ) indicate = 0
I (253547) chip[DL]: CHIPoBLE unsubscribe received
I (253547) chip[BLE]: Releasing end point's BLE connection back to application.
I (253557) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
I (284327) chip[EM]: >>> [E:58382r M:40566430] (U) Msg RX from 0:49C6A3B66B1AAE84 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (284327) chip[IN]: CASE Server received Sigma1 message. Starting handshake. EC 0x3fca2b28
I (284337) chip[IN]: CASE Server disabling CASE session setups
I (284347) chip[EM]: <<< [E:58382r M:62254211 (Ack:40566430)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (284357) chip[IN]: (U) Sending msg 62254211 to IP address 'UDP:192.168.86.95%st1:5541'
I (284367) chip[SC]: Received Sigma1 msg
I (284387) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x00BCCFE7A9785320
I (285007) chip[EM]: <<< [E:58382r M:62254212 (Ack:40566430)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
I (285017) chip[IN]: (U) Sending msg 62254212 to IP address 'UDP:192.168.86.95%st1:5541'
I (285027) chip[SC]: Sent Sigma2 msg
E (285027) chip[DL]: Long dispatch time: 701 ms, for event type 3
I (285037) chip[EM]: >>> [E:58382r M:40566431 (Ack:62254212)] (U) Msg RX from 0:49C6A3B66B1AAE84 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (285257) chip[EM]: >>> [E:58382r M:40566432 (Ack:62254212)] (U) Msg RX from 0:49C6A3B66B1AAE84 [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
I (285257) chip[EM]: <<< [E:58382r M:62254213 (Ack:40566432)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (285277) chip[IN]: (U) Sending msg 62254213 to IP address 'UDP:192.168.86.95%st1:5541'
I (285287) chip[SC]: Received Sigma3 msg
E (285297) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
I (286577) chip[EM]: <<< [E:58382r M:62254214 (Ack:40566432)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (286587) chip[IN]: (U) Sending msg 62254214 to IP address 'UDP:192.168.86.95%st1:5541'
I (286597) chip[SC]: SecureSession[0x3fca13b8]: Moving from state 'kEstablishing' --> 'kActive'
I (286597) chip[IN]: CASE Session established to peer: <00BCC5F600DC0B91, 1>
I (286607) chip[IN]: CASE Server enabling CASE session setups
E (286617) chip[DL]: Long dispatch time: 1361 ms, for event type 3
I (286627) chip[EM]: >>> [E:58382r M:40566433 (Ack:62254214)] (U) Msg RX from 0:49C6A3B66B1AAE84 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (312327) chip[FS]: Fail-safe timer expired
E (312327) chip[SVR]: Failsafe timer expired
I (312327) chip[SC]: SecureSession[0x3fca1300]: Moving from state 'kActive' --> 'kPendingEviction'
E (312327) chip[SVR]: Commissioning failed (attempt 2): 32
I (312337) chip[DIS]: Updating services using commissioning mode 1
I (312347) chip[DIS]: CHIP minimal mDNS started advertising.
I (312367) chip[DIS]: Advertise operational node 6F66ABD517E8AB02-00BCCFE7A9785320
I (312367) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 6F66ABD517E8AB02-00BCCFE7A9785320.
I (312377) chip[DIS]: mDNS service published: _matter._tcp
I (312377) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (312387) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: CE9DC169F755744F.
I (312417) chip[DIS]: mDNS service published: _matterc._udp
E (312417) chip[ZCL]: OpCreds: Got FailSafeTimerExpired
E (312417) chip[ZCL]: OpCreds: Proceeding to FailSafeCleanup on fail-safe expiry!
I (312427) chip[SC]: SecureSession[0x3fca13b8]: Moving from state 'kActive' --> 'kPendingEviction'
E (312437) chip[FP]: Reverting pending fabric data for fabric 0x1
E (312537) chip[FP]: Warning: metadata not found during delete of fabric 0x1
I (312757) chip[FP]: Fabric (0x1) deleted.
I (312757) chip[ZCL]: OpCreds: Fabric index 0x1 was removed
I (312757) chip[DIS]: Updating services using commissioning mode 1
I (312757) chip[DIS]: CHIP minimal mDNS started advertising.
I (312807) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (312807) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: CE9DC169F755744F.
I (312827) chip[DIS]: mDNS service published: _matterc._udp
I (313057) chip[TS]: Pending Last Known Good Time: 2023-01-25T14:43:39
I (313057) chip[TS]: Previous Last Known Good Time: 2023-01-20T12:14:13
I (313057) chip[TS]: Reverted Last Known Good Time to previous value
E (313077) chip[FP]: Warning: metadata not found during delete of fabric 0x1
E (313137) chip[ZCL]: OpCreds: failed to delete fabric at index 1: d8
I (313137) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is 0 **********
I (313147) APP DEVICE: Commissioning failed, fail safe timer expired
E (313157) chip[DL]: Long dispatch time: 830 ms, for event type 32782
I (313157) chip[FS]: Fail-safe cleanly disarmed
I (313167) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (313177) NimBLE: GAP procedure initiated: advertise; 
I (313177) NimBLE: disc_mode=2
I (313187) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (313197) NimBLE: 

I (313197) chip[DL]: CHIPoBLE advertising started
I (313197) APP DEVICE: Commissioning window opened

jonsmirl, 2023-01-27:
Go into Google Home app on phone. Select your Google Home device. On the device page drop the menu and select reboot. That will trigger the download of updates to the Google Home device.

If that doesn't work, factory reset your Google Home device. Factory reset will clears out all of the old commissioning attempts you have tried.

Also, you need at entry for your device in the Google Matter developers console.

shubhamdp, 2023-01-27:
@stefano055415 Could you post the complete logs when Google Home commissioner was used? Also, it will be helpful if you paste them into a file and attach that file here.

Also, as mentioned by @jonsmirl you need to sign up for google developer console, create a project, and an matter integration for your device.

stefano055415, 2023-01-27:
@jonsmirl @shubhamdp 

Ok i added a project in google console and now commissioning works but i get this error:

CASE failed to match destination ID with local fabrics

i have already tried to reset the hub as mentioned [here](https://github.com/espressif/esp-matter/issues/162), but the problem remains

stefano055415, 2023-01-27:
I reinstalled the GooglePlayService and now it works
thank you

closed 2023-01-27 by stefano055415