espressif/esp-matter #176: Can't get attribute update for device type door lock

mgoeller, 2023-01-25:
I am trying to implement a simple node with one endpoint of type door_lock. According to this issue [here](https://github.com/espressif/esp-matter/issues/127#issuecomment-1358805407), I set the return value to true and therefore get a functioning lock accessory in Google Home. 

```
door_lock::config_t config;
config.door_lock.lock_state = 1;
endpoint_t *endpoint = door_lock::create(node, &config, ENDPOINT_FLAG_NONE, context);
```

However the door_lock is not triggering the attribute callback so I can unlock/lock the physical lock by GPIO. Why is that not implemented in esp_matter (as defined by CHIP) as in the other examples such as lights.

jadhavrohit924, 2023-01-31:
@mgoeller Returning true there won’t work. You will need to define your callbacks as we did [here](https://github.com/project-chip/connectedhomeip/blob/master/examples/platform/esp32/lock/ZclCallbacks.cpp#L50) to update attribute.

nkpm, 2023-07-21:
I implemented all the callbacks like shown in the example.

Using Homekit with homeapp via appleTV:
- Unlocking the door_lock works fine.
- Locking the door_lock does not "finish". There's a spinner in the home UI which does not stop spinning. The lock is visually locked but the status text keeps saying "locking" and does not end with saying "locked" as it does with a working lock (via homebrige not matter).
- When shutting down and starting the ESP32 it also seems like homekit does not get the status of the lock.

Do I miss a "get status" callback? Is there one?

nkpm, 2023-07-21:
ok i solved it myself, I found the example in another issue: https://github.com/espressif/esp-matter/issues/127#issuecomment-1358808454 and implemented it like follows in the main.cpp:

`bool emberAfPluginDoorLockOnDoorLockCommand(chip::EndpointId endpointId, const Optional<chip::ByteSpan> & pinCode,
                                            DlOperationError & err) {

DoorLockServer::Instance().SetLockState(endpointId, DlLockState::kLocked);

return true;
}

bool emberAfPluginDoorLockOnDoorUnlockCommand(chip::EndpointId endpointId, const Optional<chip::ByteSpan> & pinCode,
                                              DlOperationError & err) {

DoorLockServer::Instance().SetLockState(endpointId, DlLockState::kUnlocked);

return true;
}`

jadhavrohit924, 2023-07-26:
Right @nkpm, this is wat I suggedted in https://github.com/espressif/esp-matter/issues/176#issuecomment-1409829200

jadhavrohit924, 2023-07-26:
@mgoeller Please close the issue, if problem is solved.

nkpm, 2023-07-26:
@jadhavrohit924 yes thanks a lot! What I was failing at were the 

`DoorLockServer::Instance().SetLockState...`

parts.

closed 2023-07-26 by mgoeller