espressif/esp-matter #178: Backing up fabric root keys

jonsmirl, 2023-01-28:
By default fabric root keys for a commissioner are kept in local storage on the device. That's bad news if the device dies because then you lose access to your fabric. So I'm trying to figure out how to store the root key in the cloud.

First off, is it possible to create a new commissioner while only using the cloud root key once when the commissioner is created? 
Next, can I create a second commissioner on the same fabric also using the cloud root key only once? and then be able to locally commission from both devices (assuming they assign disjoint nodeIds).

If the answer to both of those is yes, then how do I set this up? When you create a commissioner you can supply...

     * If the keypairDelegate is provided, then the rootCertificate, nodeOperationalCertificate, and
     * ipkEpochKey must also be specified. If no operational credentials are specified here, then an
     * ephemeral signing configuration will be generated for you.

     * @param[in] keypairDelegate a pointer to a Java KeypairDelegate implementation.
     * @param[in] rootCertificate an X.509 DER-encoded trusted root certificate for this node
     * @param[in] intermediateCertificate an X.509 DER-encoded intermediate certificate for this node
     * @param[in] nodeOperationalCertificate an X.509 DER-encoded operational certificate for this node
     * @param[in] ipkEpochKey the IPK epoch key to use for this node

In the source there is just an interface description for KeypairDelegate, no code for an implementation, but I can probably figure it out.
https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/src/chip/devicecontroller/KeypairDelegate.java

So which keypair does KeypairDelegate operate on? if I call KeypairDelegate.getPublicKey() which public key do I get? NOC or root?

So are rootCertificate and nodeOperationalCertificate just the public keys in certificate format? And then keypairDelegate() is operations on the NOC key.

And I have no clue what to do with ipkEpoch (Identity Protection Key)

Is this sequence correct?

1) Cloud, make root key pair 
2) Cloud, make certificate for it
3) Device, get root certificate from cloud
4) Device, Generate NOC keypair in trustzone
5) Device, Generate CSR
6) Device, Send CSR to cloud, get back NOCcertificate
7) Device, pass in rootCertificate and nodeOperationalCertificate to the commissioner
8) Device, Commissioner uses keypairDelegate to sign using local NOC key

jonsmirl, 2023-01-28:
BTW, I am trying to figure this out on my phone first since it is easier. When I get the basics figured out, the target is a ESP32S3 based commissioner.

wqx6, 2023-02-07:
> is it possible to create a new commissioner while only using the cloud root key once when the commissioner is created?

Yes, It is possible that all the devices(commissioner and commissionee) in one fabric using the same cloud RCAC.

> can I create a second commissioner on the same fabric also using the cloud root key only once?

Yes, the second commissioner can be created but one commissioner in one fabric is enough. We don't need to add another commissioner.

> So which keypair does KeypairDelegate operate on? if I call KeypairDelegate.getPublicKey() which public key do I get? NOC or root?

The KeypairDelegate input is just a delegate for a P256keypair for use within Java environment, it was used to generate Key pairs. And the key pairs are used to generated the certs of the NOCChain.

> So are rootCertificate and nodeOperationalCertificate just the public keys in certificate format?

Yes, the public key of the certificate is in the certificate.

> Is this sequence correct?

The device is always unable to connect to the cloud if it is not commissioned.
I suggest that you use the following sequence:
1. Cloud, make root key pair
2. Cloud, make RCAC
3. Commissioner, get the RCAC from cloud
4. Commissioner, generate commissioner's NOC keypair and CSR
5. Commissioner, Send CSR to cloud, get back commissioner's NOC, ICAC(Optional)
6. Device, generate device's NOC keypair and CSR
7. Device, send CSR to the commissioner, and the commissioner sends it to cloud
8. Device, receive RCAC and device's NOC, ICAC(optional) from the commissioner.

dhrishi, 2023-02-22:
@jonsmirl Can you please close this if your questions are answered?

closed 2023-02-22 by jonsmirl