espressif/esp-matter #179: How to tell ESP32 to open commissioning window for Multi-Admin?

tobyloki, 2023-01-28:
Hello. I have an ESP32 that has already been commissioned to a Matter fabric via an Android app. I want to also commission the ESP32 to my Raspberry Pi (which can run chip-tool to collect device metrics). I see this can be made possible by telling the ESP32 to open up its commissioning window again according to https://blog.espressif.com/matter-multi-admin-identifiers-and-fabrics-a291371af365. I wanted to ask how it would be possible to accomplish this programmatically, using the light example (https://github.com/espressif/esp-matter/tree/main/examples/light)?

jonsmirl, 2023-01-28:
It would probably be easier to commission the device to the rasppi using chip-tool first. Then send the open-commissioning-window from the rasppi to the device. Android will see the commissioning window open on the LAN and ask if you want to add it to Google Home.  Copy the passcode from chip-tool to Android and it should work.

To do it in the other direction you are going to need to write an Android app. You would start from this, and then add code to open the commissioning window. https://github.com/google-home/sample-app-for-matter-android

You can't do this automatically. The second time you open a commissioning window it will generate a passcode. You need to copy that passcode to the other device.

tobyloki, 2023-01-29:
So far I've followed the steps here (https://developers.home.google.com/codelabs/matter-sample-app#6) for the Android app to tell the ESP32 to open its commissioning window. I am now stuck trying to run the below command on the RPI from (https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md) because it requires a payload (i.e. manual pairing code). The Android app only provides a SETUP_PIN_CODE and DISCRIMINATOR, but does not print out the manual pairing code. How can I get the manual pairing code?

`chip-tool pairing code <payload> <node_id> --commissioner-name <commissioner_name>`

jonsmirl, 2023-01-29:
You are going to have to change the app code to print it out. It is a parameter in the Intent that GPS sends to the Android app.

jonsmirl, 2023-01-29:
It is in the logs.

  fun multiadminCommissioning(intent: Intent, context: Context) {
    Timber.d("multiadminCommissioning: starting")
    _commissionDeviceStatus.postValue(TaskStatus.InProgress)

    val sharedDeviceData = SharedDeviceData.fromIntent(intent)
    Timber.d("multiadminCommissioning: sharedDeviceData [${sharedDeviceData}]")
    Timber.d("multiadminCommissioning: manualPairingCode [${sharedDeviceData.manualPairingCode}]")

jonsmirl, 2023-01-29:
1) Go into the Google Home app.
2) Hold icon of device to get device control screen
3) Tap the Settings icon
4) Tap Link matter apps/services
5) Tap Link apps and services
6) slide the row of icons at the bottom over to expose the last ones
7) Select 'Use pairing code'

That opens the commissioning window and puts the code on the screen.
Enter that code manually on the RaspPi

tobyloki, 2023-01-29:
The pairing code works. However, my chip-tool on the RPI is giving me this error which prevents the commissioning from completing. How can I fix it?

The error I get if I use a new commissioner name (`chip-tool pairing code 13 00030406855 --commissioner-name 6`)
`CHIP:TOO: Run command failure: ../../src/controller/CHIPDeviceController.cpp:1363: CHIP Error 0x0000000B: No memory`

The error I get if I use the same commissioner name from before (`chip-tool pairing code 13 00030406855 --commissioner-name 4`)
`CHIP:TOO: Run command failure: ../../src/controller/CHIPDeviceController.cpp:1367: CHIP Error 0x0000007E: Trying to add a NOC for a fabric that already exists`

jonsmirl, 2023-01-29:
I think there are only 4 or 5 commissioner slots and they are all likely full from your previous attempts.  Do erase_flash on esp32 and start over. 

The way you are doing this is not actually how it should be done. You should make your RaspPi into a Matter Controller.  Then use your phone to commission the RaspPi Controller onto the same fabric as the phone.  The Controller can then proceed to do all of the monitoring stuff. I'm not sure how to run a controller from chip-tool. When chip-tool starts it generates it own keys, it does not wait for someone else to supply them. You can be a controller from a Linux app.

Note Commissioner and Controller are two separate things. A Commissioner just brings new devices onto the fabric and installs keys into them. The Controller actually sets the devices up and does subscriptions, etc.  You don't really want the RaspPi to be a Commissioner, it should be a Controller for your use-case.  

Typically Commissioners are also Controllers, but Controllers do not have be Commissioners. For example your phone needs to be a Commissioner so that you can carry it around within BLE range of the devices. It makes no sense for the RaspPi to be a Commissioner since you can't carry it around. Instead it should just be a Controller and then use the phone to commission it onto your fabric. Once it has been commissioned onto the fabric it can use MDNS to browse for all of the Matter devices active on that same fabric. That will let you get the nodeIds for them.

dhrishi, 2023-01-30:
@tobyloki If I understand your usecase, you commission the device through the Android app. Then open the commissioning window from the app. The generated payload from the Android app is passed through the `pairing code` command from the RPi. Is the commissioning successful and does the RPi get added as the 2nd admin?

And then you are trying with different commissioner name using the same RPi, in a way different instances? Or am I missing anything?

dhrishi, 2023-02-02:
@tobyloki Please provide updates, if any. Or close the issue if your question is answered.

dhrishi, 2023-02-13:
Closing this. Please re-open to share any update or issue.

closed 2023-02-13 by dhrishi