espressif/esp-matter #180: mfg_tool and certs, how to use?

jonsmirl, 2023-01-28:
I made a partition with certs...
./mfg_tool.py -cn "My bulb" -v 0xFFF2 -p 0x8001 --pai \
   -k path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
   -c path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
   -cd path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der

I erased flash
I flashed my app

I put it on the device...
esptool.py -p <serial_port> write_flash <address> path/to/<uuid>-partition.bin

I rebooted and monitored. 

It doesn't seem to use the cert partition since it still works with my old QRCode. So how do I tell it to use the new cert partition?

After I get this functioning, how to I get access to the DAC public key, and then ask the ESP to sign things using the DAC private key? I want to use the DAC key as my ID to access Google Firebase.

We need an example for mfg_tool, copying the partition, etc. Just using the demo certificates.  I would recommend disabling the internal certificate support and forcing all of the examples to use mfg_tool and sec_cert partition (with the same demo certificates).

jonsmirl, 2023-01-28:
Also, doesn't mfg_tool need to put the deviceId and vendorId into the DAC certificate when it is generated?

jonsmirl, 2023-01-29:
Do mfg_tool need to use the DS peripheral and write the efuses?

dhairyashah1, 2023-01-30:
@jonsmirl 

1. Regarding creating and flashing a custom partition using `mfg_tool` -
     a. Did you refer to the example usage mentioned [here](https://github.com/espressif/esp-matter/tree/208ffca11d74517a864ada9ac900971655075493/tools/mfg_tool) ?
     b. The **following config** in `menuconfig` needs to be enabled for custom payload and data providers- 
         i. Enable [**`ESP32 Factory Data Provider`**](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/factory_data.md) [Component config → CHIP Device Layer → Commissioning options → Use ESP32 Factory Data Provider]
        ii. Enable **`ESP32 Device Instance Info Provider`** [Component config → CHIP Device Layer → Commissioning options → Use ESP32 Device Instance Info Provider]
        iii. Enable **`Attestation - Factory`** [ Component config → ESP Matter → DAC Provider options → Attestation - Factory] 
    c. The above config may not have been enabled, thus it was still working with old QR code.

2. Ideally you should read the **certificate** for getting access to the `DAC public key` by reading the `ESP32Config::kConfigKey_DACPublicKey` field from [here](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32FactoryDataProvider.cpp#L149).
3. You can get the `Device Attestation PAI certificate` from [here](https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/DeviceAttestationCredsProvider.h#L68)
4. Reference for `signing with Device Attestation key `can be found [here](https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/DeviceAttestationCredsProvider.h#L90)
5. Sample usage of `mfg_tool` is mentioned [here](https://github.com/espressif/esp-matter/tree/208ffca11d74517a864ada9ac900971655075493/tools/mfg_tool). We will soon update the `README.md` with example steps for creating and uploading custom partition.
6. Custom partition / configuration corresponds to using **Customer Certificates**. We can't use the `demo certificates` for the same.

jonsmirl, 2023-01-30:
Yes, I have been following the mfg_tool instructions. 

Can ESP DS hardware directly protect the DAC ECDSA private key?

How can I ask it to sign something using the DAC private key?  I want to also use the DAC key for Google authorization so that I don't need to deal with two keys on the device.

AdityaHPatwardhan, 2023-01-31:
Hi @jonsmirl The DS peripheral present on the current chipsets only supports the RSA algorithm.
It would require hardware level changes for supporting ECDSA-based signing with DS peripheral. These hardware changes might be possible in the upcoming espressif chipsets.
As for the DS peripheral that is present, it cannot be used for the ECDSA key.

In order to involve hardware in the protection of the DAC private key, we are currently in process of developing a security workflow that uses HMAC-based ECDSA key generation. The firmware level support is already merged in the [esp_secure_cert_mgr](https://github.com/espressif/esp_secure_cert_mgr) through https://github.com/espressif/esp_secure_cert_mgr/commit/97086f8f16b3b7988989f440fa0f3307fb65ab3e

In this case, an HMAC key shall be stored in the eFuse under R/W protection and that shall be used along with the HMAC peripheral and SHA peripheral to generate an ECDSA key at run-time. This way the key is not present in the flash at all.
Once the secure boot is enabled only the trusted firmware can be executed on the device and hence only trusted firmware can access the HMAC peripheral to generate the ECDSA private key.

shubhamdp, 2023-01-31:
> How can I ask it to sign something using the DAC private key? I want to also use the DAC key for Google authorization so that I don't need to deal with two keys on the device.

You can use the [device attestation credentials provider](https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/DeviceAttestationCredsProvider.h) to sign the message.
```
#include <credentials/DeviceAttestationCredsProvider.h>
chip::Credentials::GetDeviceAttestationCredentialsProvider()->SignWithDeviceAttestationKey(msg, sign);
```

For getting DAC public key, you get the DAC certificate and extract the public key from there.
```
#include <credentials/DeviceAttestationCredsProvider.h>
#include <crypto/CHIPCryptoPAL.h>

chip::Credentials::GetDeviceAttestationCredentialsProvider()-> GetDeviceAttestationCert(outSpan);
chip::Crypto::ExtractPubkeyFromX509Cert(certSpan, pubkey);
```

Reference:
https://github.com/project-chip/connectedhomeip/blob/210ae33e6c369abd899649742ae0dbc2b12239d7/src/crypto/CHIPCryptoPAL.h#L1504

jonsmirl, 2023-01-31:
Is this good enough to meet the chip spec? No clause requiring hardware protected keys like Amazon has?

Another solution would be to lobby the CHIP people to also accept RSA certificates for the DAC.

dhrishi, 2023-05-30:
Fixed with 9fc215451f57898187185bbe56a913455ebf022e

closed 2023-05-30 by dhrishi

thisal29, 2023-07-18:
Hi, 

I have been following this thread which was very useful in the task that I'm currently engaged in. 

I'm trying to use DAC private and public keys to sign LoRa messages. 

I'm able to sign the messages with,

`chip::Credentials::GetDeviceAttestationCredentialsProvider()->SignWithDeviceAttestationKey(msg, sign);`

I also understand how to extract the DAC public key.

Will anyone be able to help understand how to use the DAC public key to verify a message that came from another device?

Thanks!

jonsmirl, 2023-07-18:
When the message from the other device arrives it will be encrypted with the public key. To decrypt it you need to use the private key via an API. I don't think I used the correct key name earlier, you don't want to be using the Attestation key, you want to be using the Fabric key. So you would decrypt with the private Fabric key.

Basic things you need.
1) A root certificate authority
2) Two keys issued by that authority
3) Each device has a key pair. 
4) Exchange public keys and the key chain. You verify by checking the common root authority between your local key and the remote one.
4) To send a message, encode it using remote public key
5) To receive a message, decode it using local private key

This is the process that happens when you commission a device. Each fabric has a different root certificate authority.

BTW, the right way to access Firebase is via a JWT token.