espressif/esp-matter #182: Error while building zigbee_bridge example

SYESHASWINI, 2023-01-30:
I'm getting error while build zigbee_bridge matter examples. I'm using esp32c3 module.

this is my log: 
/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/PlatformManagerImpl.cpp:41:10: fatal error: spi_flash_mmap.h: No such file or directory
   41 | #include "spi_flash_mmap.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
[21/235] c++ obj/third_party/connectedhomeip/src/platform/ESP32/nimble/ESP32.BLEManagerImpl.cpp.oninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
HINT: Please make sure that the header name is correct. Also please check if you've specified all component dependencies with 'idf_component_register(REQUIRES ...)'. If the component is not present then it should be added by the IDF Component Manager. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the header file has been removed, renamed or relocated - refer to the migration guide for more information.
ninja failed with exit code 1, output of the command is in the /home/ubuntu/esp/esp-matter/examples/zigbee_bridge/build/log/idf_py_stderr_output_6886 and /home/ubuntu/esp/esp-matter/examples/zigbee_bridge/build/log/idf_py_stdout_output_6886

wqx6, 2023-01-30:
For Zigbee Bridge example, ESP-IDF release/v5.0 branch at [commit 20949d44](https://github.com/espressif/esp-idf/tree/20949d44) should be used. Could you use IDF on commit 20949d44 to compile the bridge example and test?

shubhamdp, 2023-01-30:
@SYESHASWINI can you please tell the IDF version/commit that you are using?

are you trying to build the example with IDF v5.0? Currently we only support v4.4.3.

SYESHASWINI, 2023-01-30:
yes, I'm using esp-idf version v5.0 and the log belongs to the same version.

dhrishi, 2023-01-30:
@SYESHASWINI Can you use the specific commit on `release/v5.0` branch as mentioned by @wqx6 instead of ESP-IDF v5.0 release. That should solve your problem.

dhrishi, 2023-01-31:
@SYESHASWINI Please close this issue if it works as expected

SYESHASWINI, 2023-01-31:
@dhrishi I installed idf from the commit 20949d44, but still I'm getting the error.

Log:
-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/esp-matter/examples/light/build/CMakeFiles/CMakeOutput.log".
HINT: The component mdns has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details.
cmake failed with exit code 1, output of the command is in the /home/ubuntu/esp-matter/examples/light/build/log/idf_py_stderr_output_22729 and /home/ubuntu/esp-matter/examples/light/build/log/idf_py_stdout_output_22729

dhrishi, 2023-01-31:
@SYESHASWINI Can you clean and re-compile your application:
```
$ rm -rf sdkconfig sdkconfig.old build/
$ idf.py build
```

Ideally it should work as the examples get compiled in our CI on all commits, before they are pushed to Github.

SYESHASWINI, 2023-01-31:
@dhrishi  executed the commands, still getting the same error

dhrishi, 2023-01-31:
@SYESHASWINI You were trying the zigbee_bridge example previously, right? It requires the specific commit on IDF `release/v5.0` branch

>See also "/home/ubuntu/esp-matter/examples/light/build/CMakeFiles/CMakeOutput.log".

Although, this shows that you are trying the light example? Light example is based on IDF v4.4.3 and it won't work on the commit on IDF `release/v5.0` branch.

Please check the documentation here: https://github.com/espressif/esp-matter/#supported-esp-idf-and-connectedhomeip-versions

SYESHASWINI, 2023-01-31:
I also checked with zigbee_bridge example.

log:
Make Error at /home/ubuntu/esp-idf/tools/cmake/component.cmake:312 (message):
  Include directory '/home/ubuntu/esp-idf/components/mbedtls/mbedtls/include'
  is not a directory.
Call Stack (most recent call first):
  /home/ubuntu/esp-idf/tools/cmake/component.cmake:486 (__component_add_include_dirs)
  /home/ubuntu/esp-idf/components/mbedtls/CMakeLists.txt:18 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/esp-matter/examples/zigbee_bridge/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/ubuntu/esp-matter/examples/zigbee_bridge/build/log/idf_py_stderr_output_4175 and /home/ubuntu/esp-matter/examples/zigbee_bridge/build/log/idf_py_stdout_output_4175

dhrishi, 2023-01-31:
@SYESHASWINI Please attach/upload a complete log file containing: showing IDF and esp-matter version, activating the environment (export.sh), clean and re-compile the zigbee_bridge example

SYESHASWINI, 2023-01-31:
@dhrishi I'm attaching the log, please have a look at it.
[esp_log.txt](https://github.com/espressif/esp-matter/files/10543353/esp_log.txt)

shubhamdp, 2023-01-31:
@SYESHASWINI, can you try the following
```
# open a new terminal
cd esp-idf
. ./export.sh

cd esp-matter
. ./export.sh

cd examples/zigbee_bridge
idf.py fullclean
idf.py build
```

In logs I see, you did not sourced the export.sh in esp-matter repository.

shubhamdp, 2023-01-31:
Also, can you confirm one more thing, did you update the submodules after checking out esp-idf commit `20949d44`? If not can you please do the following
```
cd esp-idf
git submodule update --init --recursive
```

SYESHASWINI, 2023-01-31:
tried the above command and this is how I got the log.
[esp-idf.txt](https://github.com/espressif/esp-matter/files/10543837/esp-idf.txt)

the procedure I followed to install IDF is I downloaded the commit release/v5.0 and installed by running ./install.sh

wqx6, 2023-01-31:
The submodule update command gets the following error message in your log
```
fatal: not a git repository (or any of the parent directories): .git
```
Please check whether the `~/esp-idf` is a git repository.

dhrishi, 2023-02-01:
@SYESHASWINI There seem to be a few issues as per your attached `esp-idf.txt`:

1. There's already a folder `esp-idf` within which you are cloning `esp-idf`. And before doing `git submodule update --init --recursive` you are not entering the newly created `esp-idf` folder that was just cloned. What is the outer esp-idf folder?
2. Also, you are running the `install.sh` from the outer `esp-idf` folder whereas you should run that from the newly cloned folder
3. The log does not show that you are using the specific commit of ESP-IDF and just using the master branch

Can you please remove the outer esp-idf folder (if it is a stale one) and try out everything in a clean way

SYESHASWINI, 2023-02-01:
Cleared the folder and installed again and exported. still getting error attaching the log of it.
Thank You.
[idf.txt](https://github.com/espressif/esp-matter/files/10555255/idf.txt)

dhrishi, 2023-02-01:
@SYESHASWINI Why did you do the `git switch release/v5.0`? You moved away from the checked out commit. Also, you should update the submodules after checking out the required commit.

Can you please follow all the steps here: https://docs.espressif.com/projects/esp-matter/en/latest/esp32h2/developing.html#getting-the-repositories and confirm that everything from the setup is as expected?

`zigbee_bridge` example gets compiled through our CI before any commit goes out. So, ideally, it should compile fine

SYESHASWINI, 2023-02-01:
It has compiled fine. Thank You.

SYESHASWINI, 2023-02-01:
It has compiled fine. Thank you

closed 2023-02-01 by SYESHASWINI