espressif/esp-matter #183: Cannot use openocd to debug example light using JLINK and esp32-wroom-32

eme-wesleywong, 2023-01-30:
**OS**
Linux

**Operating System version**
Ubuntu 22.04

**Visual Studio Code version**
1.74.3

**ESP-IDF version**
4.4.3

**Python version**
3.8.7

Hardware is ESP-WROOM-32 with Segger JLINK.
For blink example in esp-idf, command line debug can be done using below commands.
```
openocd -f interface/jlink.cfg -f target/esp32.cfg
xtensa-esp32-elf-gdb -x gdbinit build/blink.elf
```

However, I cannot use the same method to debug matter example light in esp-matter.

Commands
```
openocd -f interface/jlink.cfg -f target/esp32.cfg
xtensa-esp32-elf-gdb -x gdbinit build/light.elf
```

openocd log
```
openocd -f interface/jlink.cfg -f target/esp32.cfg
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 9600 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link V10 compiled Sep 24 2021 16:39:44
Info : Hardware version: 10.10
Info : VTarget = 3.328 V
Info : clock speed 9600 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: timed out while waiting for target halted
Error executing event gdb-attach on target esp32.cpu0:

Warn : No symbols for FreeRTOS!
Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Info : Halt timed out, wake up GDB.

```

gdb log
```
xtensa-esp32-elf-gdb -x gdbinit build/light.elf
GNU gdb (crosstool-NG esp-2021r2-patch5) 9.2.90.20200913-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/light.elf...
gdbinit:1: Error in sourced command file:
Remote connection closed
(gdb) 
```

gdbinit
```
target remote :3333
set remote hardware-watchpoint-limit 2
mon reset halt
flushregs
thb app_main
c
```

igrr, 2023-01-30:
I have ran into this as well, when trying to run the light example for the first time.

The root cause is that the default GPIO used for controlling the LED is GPIO12:
https://github.com/espressif/esp-matter/blob/5ff7d0d7fa9e93d248b557f0c4eee2a7df5410d8/device_hal/device/esp32_devkit_c/device.c#L18

which is also one of the JTAG pins on ESP32. Once the code has reconfigured this pin, JTAG doesn't work anymore.

eme-wesleywong, 2023-01-30:
Thanks for the quick reply, @igrr . I tried the solution and it works.

closed 2023-01-30 by eme-wesleywong

igrr, 2023-01-30:
I think we should actually consider changing this default LED pin, since it's likely that more people who have a habit of always using JTAG will run into this. DevKit-C doesn't have a fixed LED pin at all, IIRC, so this shouldn't be a problem to use a different pin.

dhrishi, 2023-01-30:
@igrr We will update the code to use GPIO 5 as the default pin, as done in the ESP-IDF blink example.

dhrishi, 2023-02-01:
Fixed with https://github.com/espressif/esp-matter/commit/fcd058c2e7ea3839729383c4c2ea250338a50894