espressif/esp-matter #184: how to set window_covering device calibration mode

shun-dgut, 2023-01-30:
How do I set the Needs Calibration flag for the window_covering device, and also how can I test that the Needs Calibration flag for the device is set correctly?

shun-dgut, 2023-01-30:
```
window_covering_device::config_t window_covering_conf;
window_covering_conf.window_covering.type = 0x04;
window_covering_conf.window_covering.mode = 0x01;
bridged_device->endpoint = window_covering_device::add(bridged_device->endpoint, &window_covering_conf);
```

jadhavrohit924, 2023-01-31:
@shun-dgut Spec say 1 bit should be set of `mode`. Can you try setting mode value to `0x02`.

dhrishi, 2023-02-08:
@shun-dgut Please close this issue if your question is answered

dhrishi, 2023-02-13:
Closing this. Feel free to re-open in case of any issue

closed 2023-02-13 by dhrishi