espressif/esp-matter #185: The dac generated using the generated tool cannot be configured by chiptool, but the one using the default code can.

shun-dgut, 2023-01-30:
Using the generated dac can be added via home, but cannot be configured by chiptool

Using the code default dac can be added by home and can be configured by chiptool.

The logs that cannot be configured are as follows

[chiptool log.txt](https://github.com/espressif/esp-matter/files/10535424/chiptool.log.txt)

shubhamdp, 2023-01-30:
```
Failed in verifying 'Attestation Information' command received from the device: err 604
```
https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/attestation_verifier/DeviceAttestationVerifier.h#L72

Looks like the vendor Id in DAC, CD, and the one reported by basic cluster is not matching.

You will have to make sure the VID and PID in all these three must match if you are using different VID/PID

shubhamdp, 2023-01-30:
You can use https://github.com/project-chip/connectedhomeip/blob/master/src/tools/chip-cert/README.md#gen-cd command to generate the test certification declaration with desired VID/PID

shun-dgut, 2023-01-31:
> You can use https://github.com/project-chip/connectedhomeip/blob/master/src/tools/chip-cert/README.md#gen-cd command to generate the test certification declaration with desired VID/PID

use cmd gen dac 
```
./mfg_tool.py -cn "xxxx" -v 0xFFF2 -p 0x8001 --pai \
   -k ~/Matter_Pro/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
   -c ~/Matter_Pro/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
   -cd ~/Matter_Pro/esp-matter/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der
```

shubhamdp, 2023-01-31:
Can you please share the `sdkconfig` generated for this build, since you are reading stuff from the factory partition, below options shall be enabled.
```
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_FACTORY_PARTITION_DAC_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
```

Can you also specify the address that was used to flash the factory partition? It should be the address of value specified in `CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL` option.

shun-dgut, 2023-01-31:
> Can you please share the `sdkconfig` generated for this build, since you are reading stuff from the factory partition, below options shall be enabled.
>
> ``` [...]
[image]
[image]

I've set these parameters, but it still doesn't work.

I can add devices using homekit

shubhamdp, 2023-01-31:
Hi, can you enable `CONFIG_FACTORY_PARTITION_DAC_PROVIDER=y` and `CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y` options?

First one is for reading DACs from the factory partition and second one is for reading vid/pid from the factory partition.

Also, I assume you are flashing the generated partition at the address of `fctry` partition.

shun-dgut, 2023-02-01:
> Hi, can you enable `CONFIG_FACTORY_PARTITION_DAC_PROVIDER=y` and `CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y` options?
>
> First one is for reading DACs from the factory partition and second [...]

Thanks，CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER = y is Ok

closed 2023-02-01 by shun-dgut