espressif/esp-matter #186: Home App is not able to find my matter device

aidnani, 2023-01-30:
From the last time I tried doing this.
- I got an Apple TV4, 
- upgraded that to 16.3
- added the matter profile to Apple TV4
- Apple TV4 and iPhone are on the same WiFi network.
- Powered my matter device ESP32S3, and enabled ble adv on it.
- When I try to add the device using Home App on iPhone I get an error (see attached) 
[image]

The matter device shows the following when my iPhone connects to it,

`> matter ble adv state
BLE advertising is enabled
Done
> I (1075665) chip[DL]: BLE GAP connection established (con 1)
I (1075665) chip[DL]: CHIPoBLE advertising stopped
I (1076525) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (1076525) chip[BLE]: local and remote recv window sizes = 5
I (1076535) chip[BLE]: selected BTP version 4
I (1076535) chip[BLE]: using BTP fragment sizes rx 244 / tx 244.
I (1076705) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 1 ) indicate = 1
I (1076705) NimBLE: GATT procedure initiated: indicate; 
I (1076715) NimBLE: att_handle=14

I (1076705) chip[DL]: CHIPoBLE subscribe received
I (1076765) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (1076765) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (1076785) chip[EM]: >>> [E:9803r M:191229861] (U) Msg RX from 0:1A84708F5F040C74 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
I (1079285) NimBLE: GATT procedure initiated: indicate; 
I (1079285) NimBLE: att_handle=14

I (1079315) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (1081805) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (1084305) NimBLE: GATT procedure initiated: indicate; 
I (1084305) NimBLE: att_handle=14

I (1084355) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (1086845) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (1089345) NimBLE: GATT procedure initiated: indicate; 
I (1089345) NimBLE: att_handle=14

I (1089395) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (1091885) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (1094385) NimBLE: GATT procedure initiated: indicate; 
I (1094385) NimBLE: att_handle=14

I (1094435) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (1096925) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (1099425) NimBLE: GATT procedure initiated: indicate; 
I (1099425) NimBLE: att_handle=14

I (1099475) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (1101965) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (1104465) NimBLE: GATT procedure initiated: indicate; 
I (1104465) NimBLE: att_handle=14

I (1104515) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (1106525) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 1 ) indicate = 0
I (1106525) chip[DL]: CHIPoBLE unsubscribe received
I (1106535) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
I (1106525) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (1106555) NimBLE: GAP procedure initiated: advertise; 
I (1106555) NimBLE: disc_mode=2
I (1106565) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (1106575) NimBLE: 

I (1106575) chip[DL]: CHIPoBLE advertising started
I (1106585) chip[BLE]: Releasing end point's BLE connection back to application.
`

What am I doing wrong?

aidnani, 2023-01-30:
I just did a git pull on the repo, and it wouldn't complete the install.sh script anymore.

`Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (3.0s)
  Setting up Python environment.....[-]
ERROR at //BUILD.gn:122:3: Unknown function.
  pw_python_pip_install("pip_install_matter_packages") {
  ^--------------------
['gn', 'gen', '/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']

Traceback (most recent call last):
  File "/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 310, in install_packages
    subprocess.check_call(gn_cmd,
  File "/usr/local/Cellar/python@3.9/3.9.8/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gn', 'gen', '/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 795, in <module>
    sys.exit(main())
  File "/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 787, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 457, in setup
    result = step(spin)
  File "/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 606, in virtualenv
    if not virtualenv_setup.install(
  File "/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 342, in install
    install_packages(gn_target)
  File "/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 317, in install_packages
    raise subprocess.CalledProcessError(err.returncode, err.cmd,
subprocess.CalledProcessError: Command '['gn', 'gen', '/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/Ajay.Idnani/Documents/Code/Matter/connectedhomeip/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.`

jonsmirl, 2023-01-30:
you need to install the gn command

https://gn.googlesource.com/gn/

aidnani, 2023-01-30:
Thank you. I actually trashed my folder and recloned, that solved the install problem. I am still unable to pair with iPhone (using AppleTV4) - the original problem.

aidnani, 2023-01-30:
I did the following and tried again.

idf.py erase-flash build flash monitor

and this time I was able to add the Matter accessory to HomeApp and control it. You can close this issue. (Although I would like to understand what is it that erase-flash does that solved the problem. I didn't have any WiFi configured on the device before.

jonsmirl, 2023-01-30:
Commissioned nodeIDs are remembered in flash. When you erase_flash you got rid of the old commissioning info.  Your likely problem, the esp device was commissioned onto a different nodeID than the phone thought it was on.  When you erased and recommissioned the phone an ESP device got onto the same page.

You need to erase_flash each time you recommission a device.

closed 2023-01-31 by dhrishi