espressif/esp-matter #189: typo in fan_control adding attribute?

YOGYUI, 2023-01-31:
[esp_matter_attribute.cpp](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_attribute.cpp)
line 1220 ~ 1224
```
attribute_t *create_percent_current(cluster_t *cluster, uint8_t value)
{
    return esp_matter::attribute::create(cluster, FanControl::Attributes::PercentSetting::Id, ATTRIBUTE_FLAG_NONE,
                                         esp_matter_uint8(value));
}
```
I think 2nd argument or "create" function should be replaced by "FanControl::Attributes::PercentCurrent::Id".
Is that right?

dhrishi, 2023-01-31:
@YOGYUI Thanks for pointing it out. Yes, that's a typo. We will fix it soon

dhrishi, 2023-02-01:
Thanks @YOGYUI. This is now fixed with https://github.com/espressif/esp-matter/commit/674e41928241bb7225927e55b321a5e06d8a2abd

closed 2023-02-01 by dhrishi