espressif/esp-matter #194: building BLE example out of tree

jonsmirl, 2023-02-01:
It should be possible to copy the BLE example out of tree and still build it.

So I would propose moving the platform directory out of the blemesh example and up to the top level of esp-matter. Call it ble-platform or similar.  But then how do you use environment variables in sdkconfig variable CONFIG_CHIP_EXTERNAL_PLATFORM_DIR to point to it?
~~~
CONFIG_CHIP_EXTERNAL_PLATFORM_DIR="$ESP_MATTER_PATH/examples/blemesh_bridge/platform/ESP32_custom"

ERROR at build arg file (use "gn args <out_dir>" to edit):15:28: Undefined identifier in string expansion.
chip_platform_target = "//$ESP_MATTER_PATH/examples/blemesh_bridge/platform/ESP32_custom"
                           ^--------------
"ESP_MATTER_PATH" is not currently in scope.
~~~

jonsmirl, 2023-02-02:
I understand this problem better now....

1) I should not be duplicating the blemesh platform directory in my project.
2) I don't believe the platform directory should be part of the blemesh example. Instead move it up to the root of the project and rename it ble-platform (or similar). This directory is not really part of the blemesh example, instead it is a common piece of infrastructure needed by any project using BLE after Matter starts.
3) Change sdkconfig to CONFIG_CHIP_EXTERNAL_PLATFORM_DIR="../../../../ble-platform/ESP32_custom"  This is not a relative path as it appears, instead it is appended onto the esp-matter chip path, so it needs to climb up out of the CHIP project.

These changes follow the rule that if I delete the esp-matter examples directory, then my external project should still be able to build. Now you can also copy the blemesh_bridge example out of tree and it will still build. The needed symlinks are in the ble-platform which was moved out of the project.

BTW, it you really wanted to use an environment variable in that path the syntax is $ENV{ESP_MATTER_PATH} instead of $ESP_MATTER_PATH

dhrishi, 2023-02-02:
>So I would propose moving the platform directory out of the blemesh example and up to the top level of esp-matter. Call it ble-platform or similar.

IMO, the custom platform is application specific. There can be different BLE+Matter (BLE Mesh is one example) applications that use custom platform in totally different ways. Some would require BLE after commissioning is done, some before commissioning. Others would require a different advertisement data and GATT database. In short, as per the application specific logic or usecase

costaudtest01, 2023-02-02:
https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html?highlight=esp32_custom#advanced-setup

Here is the description in the official doc.

wqx6, 2023-02-02:
The string `CONFIG_CHIP_EXTERNAL_PLATFORM_DIR` is passed to a gn file at last. So you should not use a CMake format( such as $ESP_MATTER_PATH) in it. I suggest you to use a absolute path for the external platform dir since we will not move it to other path very often. if you use a relative path, it be relative to the output gn file https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/CMakeLists.txt#L214

jonsmirl, 2023-02-02:
The environment variable in CONFIG_CHIP_EXTERNAL_PLATFORM_DIR was not needed once I figured out that CONFIG_CHIP_EXTERNAL_PLATFORM_DIR was not a relative path, instead it is a partial path.

However, the actual problem here is that you can not move the blemesh example out of the esp-matter tree and have it build. The platform directory is symlinked into the CHIP tree.  Move the example, don't copy it. If you copy it the copy will still use the symlinks in the original.

How about this? Move the example platform directory to the project root, and then inside of it make multiple ESP32_custom directories? For example ESP32_custom_blemesh.  

Do you see the problem from my side? I can't make a standalone tree for my code containing a platform directory because it would need symlinks into esp_matter and everyone has esp_matter in a different place. So I want to point CONFIG_CHIP_EXTERNAL_PLATFORM_DIR into esp-matter instead of my code, but I don't like pointing it into an example which could get moved, renamed, deleted, etc.

wqx6, 2023-02-10:
@jonsmirl We have just move the external platform of ble_mesh platform example out of the example. Can we close the issue now?

jonsmirl, 2023-02-10:
Should you flip the directory name in case there are more platforms?
blemesh_platform/platform  to platform/blemesh_platform

dhrishi, 2023-02-13:
Fixed with https://github.com/espressif/esp-matter/commit/52795c4b50d4ead4ead4a02c1e2562ed760c726d

closed 2023-02-13 by dhrishi