espressif/esp-matter #195: zap_light example error

MartinPArmstrong, 2023-02-02:
After building and flashing the zap_light example, the devices crashes and reboots continuously.
Here is a single iteration of the crash-reboot loop.

```
ELF file SHA256: afd86908c7c3bdf3

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375ae0
0x40375ae0: esp_restart_noos at /home/martin/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1664
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2e8c
entry 0x403c9954
I (29) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (29) boot: compile time 02:16:25
I (29) boot: chip revision: 0
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (69) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (84) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (92) boot:  4 ota_0            OTA app          00 10 00020000 001e0000
I (99) boot:  5 ota_1            OTA app          00 11 00200000 001e0000
I (107) boot:  6 fctry            WiFi data        01 02 003e0000 00006000
I (114) boot: End of partition table
I (118) esp_image: segment 0: paddr=00020020 vaddr=3c0e0020 size=305b8h (198072) map
I (162) esp_image: segment 1: paddr=000505e0 vaddr=3fc97ca0 size=04b24h ( 19236) load
I (167) esp_image: segment 2: paddr=0005510c vaddr=40374000 size=0af0ch ( 44812) load
I (178) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=dd258h (905816) map
I (341) esp_image: segment 4: paddr=0013d280 vaddr=4037ef0c size=08d90h ( 36240) load
I (349) esp_image: segment 5: paddr=00146018 vaddr=50000000 size=00010h (    16) load
I (357) boot: Loaded app from partition at offset 0x20000
I (358) boot: Disabling RNG early entropy source...

I (370) cpu_start: Pro cpu up.
I (370) cpu_start: Starting app cpu, entry point is 0x40375458
0x40375458: call_start_cpu1 at /home/martin/esp-idf/components/esp_system/port/cpu_start.c:148

I (348) cpu_start: App cpu up.
I (384) cpu_start: Pro cpu start user code
I (384) cpu_start: cpu freq: 160000000
I (384) cpu_start: Application information:
I (387) cpu_start: Project name:     zap_light
I (392) cpu_start: App version:      v1.0
I (397) cpu_start: Compile time:     Feb  2 2023 02:16:17
I (403) cpu_start: ELF file SHA256:  afd86908c7c3bdf3...
I (409) cpu_start: ESP-IDF:          v4.4.3-dirty
I (414) heap_init: Initializing. RAM available for dynamic allocation:
I (421) heap_init: At 3FCACAA8 len 0003CC68 (243 KiB): D/IRAM
I (428) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (434) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (441) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (448) spi_flash: detected chip: generic
I (452) spi_flash: flash io: dio
W (456) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (470) sleep: Configure to isolate all GPIO pins in sleep state
I (475) sleep: Enable automatic switching of GPIO sleep configuration
I (483) coexist: coexist rom version e7ae62f
I (488) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
[... 110 lines cut ...]

0x42019c85: emberAfColorControlClusterServerInitCallback(unsigned short) at /home/martin/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/color-control-server/color-control-server.cpp:2636

0x42013885: initializeEndpoint(EmberAfDefinedEndpoint*) at /home/martin/esp-matter/connectedhomeip/connectedhomeip/src/app/util/attribute-storage.cpp:352

0x420138b6: emAfCallInits() at /home/martin/esp-matter/connectedhomeip/connectedhomeip/src/app/util/attribute-storage.cpp:365

0x42015a76: emberAfInit(chip::Messaging::ExchangeManager*) at /home/martin/esp-matter/connectedhomeip/connectedhomeip/src/app/util/util.cpp:154

0x42012ed7: InitDataModelHandler(chip::Messaging::ExchangeManager*) at /home/martin/esp-matter/connectedhomeip/connectedhomeip/src/app/util/DataModelHandler.cpp:39

0x42012b9c: chip::Server::Init(chip::ServerInitParams const&) at /home/martin/esp-matter/connectedhomeip/connectedhomeip/src/app/server/Server.cpp:245

0x4200b39b: esp_matter::esp_matter_chip_init_task(int) at /home/martin/esp-matter/components/esp_matter/esp_matter_core.cpp:765

0x42053d8d: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/martin/esp-matter/examples/zap_light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.ipp:252

0x4205408e: chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/martin/esp-matter/examples/zap_light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:437
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /home/martin/esp-matter/examples/zap_light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:232

0x420540bd: chip::DeviceLayer::PlatformManager::RunEventLoop() at /home/martin/esp-matter/examples/zap_light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:362
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /home/martin/esp-matter/examples/zap_light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:257

0x40381f69: vPortTaskWrapper at /home/martin/esp-idf/components/freertos/port/xtensa/port.c:131
```

MartinPArmstrong, 2023-02-02:
The board in question is the ESP32-S3-DevKitC-1

wqx6, 2023-02-02:
Looks like the issue occurs because the reported temperature mireds value is 0, we will fix it asap.