espressif/esp-matter #196: Compilation of light example fails with "ModuleNotFoundError: No module named 'lark'"

benjaminnestler, 2023-02-02:
Hi,

I've issues during compilation and linking of the examples!

Following steps to reproduce:
- fresh checked out repos `esp-matter`and `esp-idf`(v4.4.3)
- ./install.sh in both repos
- activatation of both environments (first esp-idf, second esp-matter)

```
cd esp-matter/examples/ligth
idf.py set-target esp32c3
idy.py build
```
fails with following message trace:

```
./../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model/controller-clusters.matter
Traceback (most recent call last):
  File "/media/psf/Home/VM_Workspaces/Ubuntu_Matter_ESP/esp/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/codegen.py", line 28, in <module>
    from idl.matter_idl_parser import CreateParser
  File "/media/psf/Home/VM_Workspaces/Ubuntu_Matter_ESP/esp/esp-matter/connectedhomeip/connectedhomeip/scripts/idl/matter_idl_parser.py", line 7, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/media/psf/Home/VM_Workspaces/Ubuntu_Matter_ESP/esp/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/codegen.py", line 32, in <module>
    from idl.matter_idl_parser import CreateParser
  File "/media/psf/Home/VM_Workspaces/Ubuntu_Matter_ESP/esp/esp-matter/connectedhomeip/connectedhomeip/scripts/idl/matter_idl_parser.py", line 7, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
```
Full log is attached here [full_log_light_example.log](https://github.com/espressif/esp-matter/files/10566882/full_log_light_example.log)

Anybody an idea what happens?

wqx6, 2023-02-02:
Please following these steps to set up the python environment.
1. install and export IDF python environment by `cd esp-idf; ./install.sh; . ./export.sh`
2. install and export esp_matter requirement by `cd esp-matter; ./install.sh; . ./export.sh`

Then the error should be fixed.

benjaminnestler, 2023-02-02:
> Please following these steps to set up the python environment.
> 
>     1. install and export IDF python environment by `cd esp-idf; ./install.sh; . ./export.sh`
> 
>     2. install and export esp_matter requirement by `cd esp-matter; ./install.sh; . ./export.sh`
> 
> 
> Then the error should be fixed.

Sorry that's not the solution for my issue. Of course I activate the environments after installation process.

igrr, 2023-02-02:
Could you please capture the log of running `./install.sh` in esp-matter directory? This will help understand why the `lark` package didn't get installed.

(I got a similar issue recently, and in my case it was caused by the failure on ./install.sh due to a missing prerequisite. I haven't noticed that the script has failed and proceeded to the next step.)

shubhamdp, 2023-02-02:
Applications are built in idf python environment, so we have to run `source esp-idf/export.sh` before executing `esp-matter/install.sh`. This installs all the python dependencies in idf python environment.

shubhamdp, 2023-02-02:
[Documentation](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#getting-the-repositories) does mention it but in getting esp-if section. It should have been into getting esp-matter section.

benjaminnestler, 2023-02-02:
> Applications are built in idf python environment, so we have to run `source esp-idf/export.sh` before executing `esp-matter/install.sh`. This installs all the python dependencies in idf python environment.

Okay, that could be my problem. I follow the instructions on the esp documentation site (you've linked it already).

And also as you said, the installation steps are not correct.

I will try it again tomorrow and give a feedback.

MartinPArmstrong, 2023-02-03:
I had the same error while compiling yesterday, i fixed the error by installing Lark manually.

`pip3 install lark`

benjaminnestler, 2023-02-03:
> Applications are built in idf python environment, so we have to run `source esp-idf/export.sh` before executing `esp-matter/install.sh`. This installs all the python dependencies in idf python environment.

That was definitely the solution for my issue!

closed 2023-02-03 by benjaminnestler