espressif/esp-matter #201: Is there any way to change PRODUCT_NAME attribute of bridged endpoint?

YOGYUI, 2023-02-02:
test env.
- ESP32-WROOM-32E
- esp-matter repo: most recent commit
- CHIP repo: tag v1.0.0.2
- matter setting: vendor ID 0xFFF1, product ID 0x8000

recently, I found a way to set 'node label' attribute value of bridged endpoint like below.
(add NodeLabel attribute to BridgedDeviceBasic cluster and change characters)
```
// pseudo-code
char node_label[32] = "light_0(room1)";
cluster_t *cluster = cluster::get(m_endpoint, BridgedDeviceBasic::Id);
attribute::create(cluster, NodeLabel::Id, ATTRIBUTE_FLAG_NONVOLATILE | ATTRIBUTE_FLAG_WRITABLE, esp_matter_char_str(node_label, strlen(node_label)));
```
With Google Home App, I verified that NodeLabel attribute is well applied in UI.
[image]

But, in Test Suite of Google Home Developer Console, device names are all same as 'TEST PRODUCT', so i can't distinguish each other.
[image]

I tried to change ProductName(0x00000003) attribute value of BasicInformation(0x00000028) cluster or add BasicInformation clusters to bridged endpoint with other product name string, but all tries were not valid. (maybe, this is because this attribute is managed only internally)

Question) as i mentioned in title, **is there any way to change PRODUCT_NAME attribute of bridged endpoints**?

stefano055415, 2023-02-03:
@YOGYUI 

try to see [this](https://github.com/espressif/esp-matter/issues/148)

YOGYUI, 2023-02-03:
@stefano055415 
well, NodeLabel attribute of BridgeDeviceBasic cluster is not my concern, but thanks anyway.
(I already adopt node label attribute and was able to change 'display name' of each bridge device on Google Home App)

In my case, modification of NodeLabel attribute was not applied immediately (remove control bridge node from Google Home and after re-commissioning, changed names were displayed correctly)

I just followed Google Home Developer Console guide and succeed to start Test Suite like mentioned above.
(create developer project -> create company profile -> create Matter integration -> commission esp32 device via Googel Home App -> start Test)
sorry for not being helpful :(

dhrishi, 2023-02-03:
> as i mentioned in title, is there any way to change PRODUCT_NAME attribute of bridged endpoints?

@YOGYUI The bridged node device type (0x0013) that is present on each of the bridged endpoints contains a bridged device basic information cluster (0x0039). In that cluster there's a Product Name attribute (just like the node label). Although, this attribute, which is optional, is currently not handled in the data model. We will add that support soon.

Meanwhile on your side, if you want to try, you can add the attribute here: https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_attribute.cpp#L711 and then create it in your example just like you've created the node-label.

YOGYUI, 2023-02-03:
@jadhavrohit924 @dhrishi Sincerely thanks for your advice!
I found a special(?) way to solve this issue.
> just create **NodeLabel**(0x00000005) and **ProductName**(0x00000003) attributes to **BridgedDeviceBasic**(0x00000039) cluster after creating bridged_node endpoint (aggregator) in device initialization step.
```
// pseudo-code
endpoint_t *endpoint; // bridged endpoint
char name[] = "ROOM1 LIGHT1";
cluster_t *cluster = cluster::get(endpoint, BridgedDeviceBasic::Id);
attribute::create(cluster, BridgedDeviceBasic::Attributes::NodeLabel::Id, ATTRIBUTE_FLAG_WRITABLE, esp_matter_char_str(name, strlen(name)));
attribute::create(cluster, BridgedDeviceBasic::Attributes::ProductName::Id, ATTRIBUTE_FLAG_WRITABLE, esp_matter_char_str(name, strlen(name)));
```
as @jadhavrohit924 mentioned above, ProductName attribute is read-only, value cannot be modified after creating attribute but initialization with other string is possible (I guess).

Results after using above method are like below.
- In Google Home Developer Console - Test Suite
[image]
- In Google Home App
[image]

Now, I can design test plans with Test Suite for certification.

@dhrishi I didn't modified esp-matter source-code (esp_matter_attribute.cpp) but implemented above code in my own initialization code. I hope above code snippet can be adopted in the future :) 
(maybe variable of bridged_node::config_t?)

Thanks!

dhrishi, 2023-02-06:
@YOGYUI Can you please close the issue, if it is working as expected now?

closed 2023-02-06 by YOGYUI