espressif/esp-matter #202: zap_light example, Matter start failed: 259

MartinPArmstrong, 2023-02-03:
Hi currently experiencing this error when running the example on the ESP32 unit. Don't know what to make of it.

```
I (95) boot:  �ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1664
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2e8c
entry 0x403c9954
I (24) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (25) boot: compile time 02:00:20
I (25) boot: chip revision: 0
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (32) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (65) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (72) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (80) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (87) boot:  4 ota_0            OTA app          00 10 00020000 001e0000
I (95) boot:  5 ota_1            OTA app          00 11 00200000 001e0000
I (102) boot:  6 fctry            WiFi data        01 02 003e0000 00006000
I (110) boot: End of partition table
I (114) boot: No factory image, trying OTA 0
I (119) esp_image: segment 0: paddr=00020020 vaddr=3c0e0020 size=305b8h (198072) map
I (163) esp_image: segment 1: paddr=000505e0 vaddr=3fc97ca0 size=04b24h ( 19236) load
I (167) esp_image: segment 2: paddr=0005510c vaddr=40374000 size=0af0ch ( 44812) load
I (179) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=dd260h (905824) map
I (341) esp_image: segment 4: paddr=0013d288 vaddr=4037ef0c size=08d90h ( 36240) load
I (350) esp_image: segment 5: paddr=00146020 vaddr=50000000 size=00010h (    16) load
I (358) boot: Loaded app from partition at offset 0x20000
I (395) boot: Set actual ota_seq=1 in otadata[0]
I (395) boot: Disabling RNG early entropy source...
I (406) cpu_start: Pro cpu up.
I (406) cpu_start: Starting app cpu, entry point is 0x40375458
0x40375458: call_start_cpu1 at /home/martin/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (420) cpu_start: Pro cpu start user code
I (420) cpu_start: cpu freq: 160000000
I (420) cpu_start: Application information:
I (423) cpu_start: Project name:     zap_light
I (428) cpu_start: App version:      v1.0
I (432) cpu_start: Compile time:     Feb  3 2023 02:00:16
I (438) cpu_start: ELF file SHA256:  b3ab3a573693aeb8...
I (444) cpu_start: ESP-IDF:          v4.4.3-dirty
I (450) heap_init: Initializing. RAM available for dynamic allocation:
I (457) heap_init: At 3FCACAA8 len 0003CC68 (243 KiB): D/IRAM
I (463) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (470) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (476) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (483) spi_flash: detected chip: generic
I (487) spi_flash: flash io: dio
W (491) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (506) sleep: Configure to isolate all GPIO pins in sleep state
I (511) sleep: Enable automatic switching of GPIO sleep configuration
I (518) coexist: coexist rom version e7ae62f
I (524) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (651) led_driver_ws2812: Initializing light driver
I (651) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (661) pp: pp rom version: e7ae62f
I (661) net80211: net80211 rom version: e7ae62f
I (681) wifi:wifi driver task: 3fcedd4c, prio:23, stack:6656, core=0
[... 102 lines cut ...]
I (2691) chip[DIS]: Advertise operational node 8BFD1C887C902203-00000000C0BD13C2
I (2691) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 8BFD1C887C902203-00000000C0BD13C2.
I (2701) chip[DIS]: mDNS service published: _matter._tcp
I (3161) esp_netif_handlers: sta ip: 192.168.1.145, mask: 255.255.255.0, gw: 192.168.1.1
I (3161) chip[DL]: IP_EVENT_STA_GOT_IP
I (3161) chip[DL]: IPv4 address changed on WiFi station interface: 192.168.1.145/255.255.255.0 gateway 192.168.1.1
I (3171) chip[DL]: IPv4 Internet connectivity ESTABLISHED
I (3181) app_main: Interface IP Address Changed
I (3191) chip[DIS]: Updating services using commissioning mode 0
I (3201) chip[DIS]: CHIP minimal mDNS started advertising.
I (3211) chip[DIS]: Advertise operational node 8BFD1C887C902203-00000000C0BD13C2
I (3211) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 8BFD1C887C902203-00000000C0BD13C2.
I (3231) chip[DIS]: mDNS service published: _matter._tcp
I (3231) ROUTE_HOOK: Hook already installed on netif, skip...
I (3661) chip[DL]: IP_EVENT_GOT_IP6
I (3661) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fd04:4a6c:cbe1:f200:6255:f9ff:fef5:2694
I (3671) app_main: Interface IP Address Changed
I (3671) chip[DIS]: Updating services using commissioning mode 0
I (3671) chip[DIS]: CHIP minimal mDNS started advertising.
I (3701) chip[DIS]: Advertise operational node 8BFD1C887C902203-00000000C0BD13C2
I (3701) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 8BFD1C887C902203-00000000C0BD13C2.
I (3721) chip[DIS]: mDNS service published: _matter._tcp
I (3721) ROUTE_HOOK: Hook already installed on netif, skip...

```

MartinPArmstrong, 2023-02-03:
MCU used is ESP32-S3-DevKitC-1

MartinPArmstrong, 2023-02-03:
The error seems to be a non issue. I simply forgot to flash the MCU memory in advance.

closed 2023-02-03 by MartinPArmstrong

FMistrorigo, 2024-03-14:
I have the same problem in the same example (zap-light) and device (esp32s3).
I litterally use the example as it is, everything seems fine but i can't connect my device, when i use idf.py monitor i get the error "app_main: Matter start failed: 259" and the device reboot.
Is this a problem of the example zap-light or i'm doing something wrong?

PS. I already tryed to use id.py erase-flash and i'm also sure that the esp-matter environment works fine.

Fabio