espressif/esp-matter #203: Create Data Model for Custom Command

DonP2748, 2023-02-03:
Hi. I want to create custom cluster (manufacturer-specific) , i also refer to "components/esp_matter_rainmaker" , but its not enough for me. The command callback is called when the receiver take command. 
I understand that if i want to send a command between two nodes, i must use InvokeCommand API (src/controller/CHIPCluster.h) with the first arg is Data_Model ( ClusterName::Commands::CommandName::Type struct ) . I cannot find out where struct command data is defined.
How can i define my own Data model ?
Is there another way to send commands ?
Thanks.

jonsmirl, 2023-02-03:
Check out https://github.com/espressif/esp-matter/issues/187
Doing what you want is a complex process.  The answer is different for each platform: IOS, Android and C.

What you are looking for is in this directory. It is part of core CHIP, not esp-matter.
esp-matter/connectedhomeip/connectedhomeip/src/app/zap-templates/zcl/data-model

dhrishi, 2023-02-03:
>I understand that if i want to send a command between two nodes, 

@DonP2748 Do you want to implement a custom client cluster, using which you can send commands to the same custom server cluster on a different node? The RainMaker cluster implemented is a server cluster and is operated by or receives commands from a phone app that understands the data model of this cluster

jonsmirl, 2023-02-03:
Might need to implement Binding too.

DonP2748, 2023-02-04:
> Check out #187 
I completely agree with you. The matter has too little documentation.
> What you are looking for is in this directory. It is part of core CHIP, not esp-matter. esp-matter/connectedhomeip/connectedhomeip/src/app/zap-templates/zcl/data-model
i already have a cluster.xml file and it's added into that path, I also modified in all.xml but it doesn't' work. I still cannot generate my own data model.
> Doing what you want is a complex process. The answer is different for each platform: IOS, Android and C.
I understand now, it means that i also need CHIPTool (IOS Android) for binding custom cluster. Thank you @jonsmirl .

DonP2748, 2023-02-04:
@dhrishi the esp matter rainmaker example is not released yet so rainmaker is just another solution for me.

jonsmirl, 2023-02-04:
Add it into one of the XML files in the chip directory like the indentify one

DonP2748, 2023-02-06:
I finally found out where the datamodel struct is defined. It is  controller-cluster.matser (src/controller/data_model). But when i modified this file, so many errors appeared serially. I guess i must wait manufacturer-specific clusters from CHIP (not esp). So, thank you so much @jonsmirl.

closed 2023-02-06 by DonP2748