espressif/esp-matter #205: Enable Wi-Fi Access Point(AP) Mode after Matter Initialization?

YOGYUI, 2023-02-06:
I'm now developing HTTP Server & Web Socket Server with Matter on ESP32 platform to configure bridged endpoint modification for mobile application.

In the code [esp_matter_core.cpp](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.cpp),
Line 886 (function: **chip_init**)
```
    // ConnectivityMgr().SetWiFiAPMode(ConnectivityManager::kWiFiAPMode_Enabled);
```
I wonder why this line is annotated.
After removing annotaion and set CONFIG_ENABLE_WIFI_AP=y (using menuconfig), I could find Wi-Fi AP, name like "MATTER-XXX-VID-PID".

I think this code can be modified like below, but I'm not sure.
```
#if CHIP_DEVICE_CONFIG_ENABLE_WIFI_AP
    ConnectivityMgr().SetWiFiAPMode(ConnectivityManager::kWiFiAPMode_Enabled);
#endif
```
or, Is there any guide for enabling Wi-Fi AP with Matter on ESP32?

YOGYUI, 2023-02-06:
@dhrishi Oh, I see. I understood that configuring matter device should be done with CHIP protocol, not using other method like web, tcp, ble, etc. Is that correct?

dhrishi, 2023-02-06:
@YOGYUI By configuring, if you mean giving Wi-Fi connectivity then it is allowed. Basically there are two options for commissioning -
1. You use BLE and commission your device. One part of the commissioning process is to give IP connectivity to your device, termed as network commissioning.
2. If your device already has IP connectivity (using your own protocol or if it's an Ethernet device) then you can perform onnetwork commissioning over the IP network directly. Here, it will skip the part of network commissioning

YOGYUI, 2023-02-06:
@dhrishi Okay, Thanks very much. I'll dig in this issue.

closed 2023-02-06 by YOGYUI