espressif/esp-matter #206: An error occurred when installing esp-matter。

tonyVopo, 2023-02-08:
`Downloading and installing packages into local source directory:

  Setting up CIPD package manager...[|][proxychains] DLL init: proxychains-ng 4.16
[-][P1593 16:38:34.993 client.go:311 W] RPC failed transiently. Will retry in 1s    {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion\": dial tcp 103.226.246.99:443: i/o timeout", "host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository", "sleepTime":"1s"}
[\][P1593 16:39:06.023 client.go:311 W] RPC failed transiently. Will retry in 2s    {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion\": dial tcp 103.226.246.99:443: i/o timeout", "host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository", "sleepTime":"2s"}
[-][P1593 16:39:38.054 client.go:311 W] RPC failed transiently. Will retry in 4s    {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion\": dial tcp 103.252.115.53:443: i/o timeout", "host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository", "sleepTime":"4s"}
[/][P1593 16:40:12.082 client.go:311 W] RPC failed transiently. Will retry in 8s    {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion\": dial tcp 103.252.115.53:443: i/o timeout", "host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository", "sleepTime":"8s"}
[/][P1593 16:40:50.116 client.go:311 W] RPC failed transiently. Will retry in 16s   {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion\": dial tcp 103.252.115.53:443: i/o timeout", "host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository", "sleepTime":"16s"}
[\][P1593 16:41:36.154 client.go:311 W] RPC failed transiently. Will retry in 32s   {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion\": dial tcp 103.252.115.53:443: i/o timeout", "host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository", "sleepTime":"32s"}
[/][P1593 16:42:38.185 client.go:311 W] RPC failed transiently. Will retry in 1m4s  {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion\": dial tcp 31.13.90.19:443: i/o timeout", "host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository", "sleepTime":"1m4s"}
[-][P1593 16:44:12.230 client.go:311 W] RPC failed transiently. Will retry in 2m8s  {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion\": dial tcp 103.252.115.49:443: i/o timeout", "host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository", "sleepTime":"2m8s"}
[-][P1593 16:46:50.336 client.go:311 W] RPC failed transiently. Will retry in 4m16s {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion\": dial tcp 199.59.148.96:443: i/o timeout", "host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository", "sleepTime":"4m16s"}
[/][P1593 16:51:36.465 client.go:311 W] RPC failed transiently. Will retry in 8m32s {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion\": dial tcp 108.160.165.8:443: i/o timeout", "host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository", "sleepTime":"8m32s"}
[-][P1593 17:00:38.495 client.go:355 W] RPC failed permanently: rpc error: code = Internal desc = prpc: when sending request: Post "https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion": dial tcp 108.160.162.31:443: i/o timeout {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion\": dial tcp 108.160.162.31:443: i/o timeout", "host":"chrome-infra-packages.appspot.com", "method":"ResolveVersion", "service":"cipd.Repository"}
Error: prpc: when sending request: Post "https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ResolveVersion": dial tcp 108.160.162.31:443: i/o timeout.
CIPD selfupdate failed. Bootstrapping then retrying...
Bootstrapping cipd client for linux-amd64
[\][proxychains] Strict chain  ...  172.17.144.1:108[\] ...  timeout
Traceback (most recent call last):
  File "/mnt/f/ESP/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 381, in init
    selfupdate(client)
  File "/mnt/f/ESP/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 354, in selfupdate
    subprocess.check_call(cmd)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/mnt/f/ESP/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/cipd', 'selfupdate', '-version-file', '/mnt/f/ESP/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/.cipd_version', '-service-url', 'https://chrome-infra-packages.appspot.com']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/mnt/f/ESP/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 857, in <module>
    sys.exit(main())
  File "/mnt/f/ESP/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 849, in main
    return EnvSetup(**vars(parse())).setup()
  File "/mnt/f/ESP/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 501, in setup
    result = step(spin)
  File "/mnt/f/ESP/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 595, in cipd
    cipd_client = cipd_wrapper.init(
  File "/mnt/f/ESP/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 387, in init
    bootstrap(client, rosetta=rosetta)
  File "/mnt/f/ESP/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 328, in bootstrap
    tmp.write(client_bytes(rosetta))
  File "/mnt/f/ESP/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 252, in client_bytes
    conn.request('GET', path)
  File "/usr/lib/python3.10/http/client.py", line 1282, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.10/http/client.py", line 1328, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.10/http/client.py", line 1277, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.10/http/client.py", line 1037, in _send_output
    self.send(msg)
  File "/usr/lib/python3.10/http/client.py", line 975, in send
    self.connect()
  File "/usr/lib/python3.10/http/client.py", line 1447, in connect
    super().connect()
  File "/usr/lib/python3.10/http/client.py", line 941, in connect
    self.sock = self._create_connection(
  File "/usr/lib/python3.10/socket.py", line 845, in create_connection
    raise err
  File "/usr/lib/python3.10/socket.py", line 833, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused`

dhrishi, 2023-02-08:
@tonyVopo Please check the last part here: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#getting-the-repositories 

It's mostly a network related issue and a VPN is required for such cases.

dhrishi, 2023-02-13:
@tonyVopo Can you please confirm if your issue is solved using a VPN?

tonyVopo, 2023-02-13:
the issuse still exists. I cloned the esp-idf repository and the esp-matter repository using vpn, and successfully installed the esp-idf. I think the vpn works normally. But I don't know why the esp-matter installation failed

dhrishi, 2023-02-13:
Can you remove `esp-matter/connectedhomeip/connectedhomeop/.environmemt` folder and rerun install.sh (following the documentation)

tonyVopo, 2023-02-13:
[image]

chshu, 2023-02-13:
@tonyVopo It looks like a network connectivity:

> diag tcp xxx timeout

Could you double check your VPN connection to Google service? Or maybe try another VPN?

tonyVopo, 2023-02-13:
OK, I will try more VPN.  Ths

dhrishi, 2023-02-28:
@tonyVopo Please update if you got this to work using other VPN. If so, please close the issue

closed 2023-03-01 by tonyVopo