espressif/esp-matter #207: Communication between esp32 with Matter protocol

SergiSoler8, 2023-02-08:
Hi,

My question is if i can establish communication between two or more boards (with chip esp32) with Matter protocol.

Regards, thank you.

dhrishi, 2023-02-08:
@SergiSoler8 The short answer is yes. Device-to-device automation is a feature of Matter. A Matter client cluster device (like a light switch) can interact and control the corresponding server cluster device (like a light) after both of them are setup/commissioned into the Matter network.

We have [light](https://github.com/espressif/esp-matter/tree/main/examples/light) and [light_switch](https://github.com/espressif/esp-matter/tree/main/examples/light_switch) examples that can work with each other. You can flash them on 2 different boards and then commission them with chip-tool. After that, you can follow the steps [here](https://github.com/espressif/esp-matter/tree/main/examples/light_switch#21-bind-light-to-switch) so that they can communicate with each other. Then light_switch can send commands like on/off/toggle to the light in this case. If you need multiple devices, you can refer to the documentation just below that which talks about controlling a group of devices.

SergiSoler8, 2023-02-08:
@dhrishi Thank you very much for the quick answer.

I have two other questions. Can I use Windows 11 as a development host in Matter? Do you think this will ever be available for Arduino?

Regards, thank you.

dhrishi, 2023-02-08:
>Can I use Windows 11 as a development host in Matter? 

Upstream `connectedhomeip` does not support Windows yet. We have tested with Windows 10 using WSL. Please see the documentation [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#windows-10)

>Do you think this will ever be available for Arduino?

It should be. Although, I don't have any timelines as of now

closed 2023-02-08 by dhrishi