espressif/esp-matter #208: Test Test_TC_OO_1_1

stefano055415, 2023-02-08:
@dhrishi @shubhamdp 

Using the light example and running the test Test_TC_OO_1_1 breaks with this error

CHIP:TOO:  ***** Test Failure: featureMap value mismatch: expected 0 but got 1

PSONALl, 2023-02-08:
@stefan can you share the commit-id/branch upon which you are testing?
Also are you testing it with chip-tool or If you are testing using test-harness which version of Test-Harness you are using?
If you are using test harness please share your PICS file too

stefano055415, 2023-02-08:
@PSONALl 

i am using chip-tool

and the esp-matter commit is 3235490

dhrishi, 2023-02-08:
@stefano055415 Is it possible to paste the command and the output here?

stefano055415, 2023-02-08:
@dhrishi 

Command 
./chip-tool tests Test_TC_OO_1_1 --nodeId 0x1434

Output

CHIP:DL: KVS will be written to: file:///tmp/chip_kvs
CHIP:DL: _Init
CHIP:IN: UDP::Init bind&listen port=0
CHIP:IN: UDP::Init bound to port=58053
CHIP:IN: UDP::Init bind&listen port=0
CHIP:IN: UDP::Init bound to port=61536
CHIP:IN: BLEBase::Init - setting/overriding transport
CHIP:IN: TransportMgr initialized
CHIP:FP: Initializing FabricTable from persistent storage
CHIP:TS: Last Known Good Time: 2023-02-07T09:41:55
CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xE78B318F029FBF75, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
CHIP:ZCL: Using ZAP configuration...
CHIP:CTL: System State Initialized...
CHIP:CTL: Stopping commissioning discovery over DNS-SD
CHIP:CTL: Setting attestation nonce to random value
CHIP:CTL: Setting CSR nonce to random value
CHIP:IN: UDP::Init bind&listen port=5550
CHIP:IN: UDP::Init bound to port=5550
CHIP:IN: UDP::Init bind&listen port=5550
CHIP:IN: UDP::Init bound to port=5550
CHIP:IN: TransportMgr initialized
CHIP:DL: _OnPlatformEvent
CHIP:CTL: Stopping commissioning discovery over DNS-SD
CHIP:CTL: Setting attestation nonce to random value
CHIP:CTL: Setting CSR nonce to random value
CHIP:CTL: Generating NOC
CHIP:FP: Validating NOC chain
CHIP:FP: NOC chain validation successful
CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
CHIP:TS: Last Known Good Time: 2023-02-07T09:41:55
CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
CHIP:TS: Retaining current Last Known Good Time
CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
CHIP:TS: Committing Last Known Good Time to storage: 2023-02-07T09:41:55
CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: E78B318F029FBF75)
CHIP:IN: UDP::Init bind&listen port=5550
CHIP:IN: UDP::Init bound to port=5550
CHIP:IN: UDP::Init bind&listen port=5550
CHIP:IN: UDP::Init bound to port=5550
CHIP:IN: TransportMgr initialized
CHIP:TOO:  ***** Test Start : Test_TC_OO_1_1

CHIP:TOO:  ***** Test Step 0 : Wait for the commissioned device to be retrieved

CHIP:IN: Expiring all sessions for node <0000000000001434, 1>!!
CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000001434]
CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
CHIP:CTL: OperationalSessionSetup[1:0000000000001434]: State change 1 --> 2
CHIP:DIS: Resolving E78B318F029FBF75:0000000000001434 ...
CHIP:DIS: Resolve type=_matter._tcp name=E78B318F029FBF75-0000000000001434 interface=0
CHIP:DIS: Mdns : OnNewInterface hostname:C8F09E43BC40.local. fullname:E78B318F029FBF75-0000000000001434._matter._tcp.local. interface: 11 port: 5540 TXT:"T=1"
CHIP:DIS: Mdns: OnNewAddress interface: 11 ip:fe80::caf0:9eff:fe43:bc40
CHIP:DIS: Mdns: OnNewAddress interface: 11 ip:fdb8:8799:dc3b:4274:caf0:9eff:fe43:bc40
CHIP:DIS: Mdns: OnNewAddress interface: 11 ip:192.168.86.88
CHIP:DIS: Mdns: Resolve success on interface 11
CHIP:DIS: Node ID resolved for E78B318F029FBF75:0000000000001434
CHIP:DIS: 	Hostname: C8F09E43BC40
CHIP:DIS: 	IP Address #1: fe80::caf0:9eff:fe43:bc40
CHIP:DIS: 	IP Address #2: fdb8:8799:dc3b:4274:caf0:9eff:fe43:bc40
CHIP:DIS: 	IP Address #3: 192.168.86.88
CHIP:DIS: 	Port: 5540
CHIP:DIS: 	Mrp Interval idle: not present
CHIP:DIS: 	Mrp Interval active: not present
[... 136 lines cut ...]
CHIP:IN: SecureSession[0x600003b84000]: MarkForEviction Type:2 LSID:59373
CHIP:SC: SecureSession[0x600003b84000]: Moving from state 'kActive' --> 'kPendingEviction'
CHIP:IN: SecureSession[0x600003b84000]: Released - Type:2 LSID:59373
CHIP:FP: Forgetting fabric 0x1
CHIP:TS: Pending Last Known Good Time: 2023-02-07T09:41:55
CHIP:TS: Previous Last Known Good Time: 2023-02-07T09:41:55
CHIP:TS: Reverted Last Known Good Time to previous value
CHIP:CTL: Shutting down the commissioner
CHIP:CTL: Stopping commissioning discovery over DNS-SD
CHIP:CTL: Shutting down the controller
CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
CHIP:DMG: IM WH moving to [Uninitialized]
CHIP:DMG: IM WH moving to [Uninitialized]
CHIP:DMG: IM WH moving to [Uninitialized]
CHIP:DMG: IM WH moving to [Uninitialized]
CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
CHIP:BLE: CancelConnection
CHIP:FP: Shutting down FabricTable
CHIP:TS: Pending Last Known Good Time: 2023-02-07T09:41:55
CHIP:TS: Previous Last Known Good Time: 2023-02-07T09:41:55
CHIP:TS: Reverted Last Known Good Time to previous value
CHIP:DL: Inet Layer shutdown
CHIP:DL: BLE shutdown
CHIP:DL: System Layer shutdown
CHIP:TOO: Run command failure: ../examples/chip-tool/commands/tests/TestCommand.cpp:79: CHIP Error 0x000000AC: Internal error

dhrishi, 2023-02-08:
@stefano055415 As per the spec, the device type in the Light example has enabled the Lighting feature for On/Off cluster (ref: https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_endpoint.cpp#L176) and hence the feature map should show value 1.

Now, as far as the test case goes, IMO you should pass the correct PICS value for `OO.S.F00` so that this test step is skipped. See the test step2 [here](https://raw.githubusercontent.com/project-chip/connectedhomeip/v1.0.0.2/zzz_generated/chip-tool/zap-generated/test/Commands.h) under `class Test_TC_OO_1_1Suite : public TestCommand`  in `DoTestStep`. As you will see, step3 verifies the same feature map with lighting feature enabled and expects value 1.

You can see the tests README for details: https://github.com/project-chip/connectedhomeip/blob/v1.0.0.2/src/app/tests/suites/README.md#pics-usage

stefano055415, 2023-02-08:
@dhrishi 

Ok I understand, but if I try to change the type of light and create an on_off_light, the problem remains
[https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_endpoint.cpp#L101](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_endpoint.cpp#L101)

Could you give me an example for the pics file?
thanks

shubhamdp, 2023-02-10:
Hi @stefano055415, as per `4.1.5. Element Requirements`, `LT` is mandatory for On/Off Light device type. And esp-matter have implemented the standard device types.
What you are supposed to do is to change the PICS file as mentioned by @dhrishi in https://github.com/espressif/esp-matter/issues/208#issuecomment-1422661355.

stefano055415, 2023-02-10:
@shubhamdp 

Ok I understand now sorry

dhrishi, 2023-02-10:
>Ok I understand now sorry

Nothing to be sorry about. We are all here to help each other :-)

stefano055415, 2024-03-29:
@shubhamdp @dhrishi 

For the on_off_plugin_unit type, is it correct that the lighting feature enabled?

In the test the result I get is 1 and therefore to pass the test I enabled OO.S.F00 to true

shubhamdp, 2024-04-01:
@stefano055415 If you see the device library specification section: `5.1.5. Element Requirements` it states on/off's lighting feature is mandatory for on/off plug-in unit.
Same is implemented in esp-matter [SDK](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_endpoint.cpp#L498).

closed 2024-04-02 by stefano055415