espressif/esp-matter #210: Synchronize two clients to control one device

stefano055415, 2023-02-10:
i have two clients (Phone App) using the same account.

with the first I carry out commissioning with a device, I would like to synchronize the second via cloud so that I can use the device without having to repeat the commissioning

Could you tell me the correct flow?

it is sufficient to share the NOC with the second client

jonsmirl, 2023-02-10:
this is just a variation of this issue
https://github.com/espressif/esp-matter/issues/178

wqx6, 2023-02-14:
It is not suggested to share the NOC on the two client. Because we SHOULD use different Node-IDs for the two clients, which means that the NOCs in the two client SHOULD be different.

The suggested flow might be the following:
1. The NOCs, and ICACs(optional) in the two clients are issued by the same RCAC.
2. There is a same CASE Authenticated Tag(CAT) in the subject DN of the two NOCs.
3. When writing the ACL of the end devices at the first time, set the `subject` of the ACL to the CAT in client's NOC.
4. Then both the clients can control the end device if one of them has commissioned the end device.

falconeta, 2023-02-16:
Hi, I have the same problem but I found a key (called operatingCredentialsIssuerKeypair in the CHIP Android Wrapper code) that shared with multiple controllers allows them to control the nodes of the commissioned fabric.

Is it ok to share this key with the cloud?
Is it ok to generate this key from the cloud instead of the controller?

dhrishi, 2023-03-23:
@stefano055415 Can this be closed if your questions are answered?

@falconeta This looks like implementation specific question. If not resolved, can you please raise it in the upstream connectedhomeip repo so that appropriate people take a look and reply?

jonsmirl, 2023-03-23:
@falconeta  That key should be backed up somewhere. It is the key that allows controllers to alter your fabric. If you lose it you can't add any more devices to your fabric and you will have to rebuild the fabric using a new key. It doesn't matter where you generate it or where it is stored, just don't lose it due to reinstalling the app or something like that.

Note, you can use an intermediate CA in the cloud. The controllers can each have a different key if they all have the same intermediate CA. So that allows you to generate keys for each controller and then if they lose the key you can generate a new, different one.

See this issue: https://github.com/espressif/esp-matter/issues/178

BTW, I still have not gotten around to trying this yet.

dhrishi, 2023-04-17:
Closing this as the original question

closed 2023-04-17 by dhrishi