espressif/esp-matter #211: Setting Fixed Labels during the development process

jonsmirl, 2023-02-10:
I can use this in CMake to set various attributes during the development process
~~~
idf_build_set_property(COMPILE_OPTIONS "-DCHIP_DEVICE_CONFIG_DEVICE_VENDOR_ID=${CONFIG_DEVICE_VENDOR_ID}" APPEND)
idf_build_set_property(COMPILE_OPTIONS -DCHIP_DEVICE_CONFIG_DEVICE_VENDOR_NAME="${CONFIG_DEVICE_VENDOR_NAME}" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-DCHIP_DEVICE_CONFIG_DEVICE_PRODUCT_ID=${CONFIG_DEVICE_PRODUCT_ID}" APPEND)
idf_build_set_property(COMPILE_OPTIONS -DCHIP_DEVICE_CONFIG_DEVICE_PRODUCT_NAME="${CONFIG_DEVICE_PRODUCT_NAME}" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-DCHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING=\"${CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING}\"" APPEND)
etc...
~~~

How do I set fixed labels during the development process? I need to do this during development so that I can write the front end code which reads them.

BTW, in the latest IOS release they added Generic Switch support. Generic Switch has a Fixed Label. I want to set that label so that I can see what IOS does with it.  You might want to add it to the Generic Switch example. You need these labels because the switches are all identical and you can't tell them apart.

jonsmirl, 2023-02-14:
I have nvs initialized. What is the exact snippet of C code needed to write the Fixed Label into nvs? I can just do it every time I boot on development device. Something like this?
~~~
nvs_open("namespace?", NVS_READWRITE, &handle)
nvs_set_str(handle, "fl-sz/1", 2);
nvs_set_str(handle, "fl-k/1/1", "EP1LBL1");
nvs_set_str(handle, "fl-s/1/1", "myEP1LBL1");
nvs_set_str(handle, "fl-k/1/2", "EP1LBL2");
nvs_set_str(handle, "fl-s/1/2", "myEP1LBL2");
nvs_close(handle)
~~~

shubhamdp, 2023-02-14:
Below snippet adds two entries for fixed labels on endpoint 1.

```
nvs_handle_t handle;
nvs_open_from_partition(CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL, "chip-factory", NVS_READWRITE, &handle);
nvs_set_i32(handle, "fl-sz/1", 2);
nvs_set_str(handle, "fl-k/1/0", "EP1LBL1");
nvs_set_str(handle, "fl-v/1/0", "myEP1LBL1");
nvs_set_str(handle, "fl-k/1/1", "EP1LBL2");
nvs_set_str(handle, "fl-v/1/1", myEP1LBL2");
nvs_commit(handle);
nvs_close(handle);
```

> This could be added to the generic_switch example on an #ifdef DEBUG flag.

Let me check on this, and I'll get back.

closed 2023-03-01 by espressif-bot