espressif/esp-matter #212: activate.sh: line 234: pw: command not found

mydelta123, 2023-02-11:
where is the pw command

install esp-idf: success, try 4.4.4 and 4.4.2

install esp-matter:
https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#development-setup

the install.sh always report error
[image]

do I missing something?

antonio2904, 2023-02-11:
Try deleting the esp-matter/connectedhomeip/connectedhomeip/.environment folder manually and try install.sh again. It worked for me.

mydelta123, 2023-02-11:
> Try deleting the esp-matter/connectedhomeip/connectedhomeip/.environment folder manually and try install.sh again. It worked for me.

Oh my god it's working

1. rm -rf .environment
2. esp-matter/install.sh
3. close mac terminal app
4. open mac terminal app, and
cd esp-idf; source ./export.sh; cd ..
cd esp-matter; source ./export.sh; cd ..

after that, cd esp-matter/example/light ; idf.py flash monitor

i got error again, no module 'pw_build' 

```
~~ a log of logs ~~
[50/729] ACTION //third_party/connectedhomeip/src/controller/data_model:data_model_codegen_codegen(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
FAILED: gen/third_party/connectedhomeip/src/controller/data_model/app/PluginApplicationCallbacks.h gen/third_party/connectedhomeip/src/controller/data_model/app/callback-stub.cpp 
python ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../../../../connectedhomeip/connectedhomeip/config/esp32/ --current-path ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model --default-toolchain=//third_party/connectedhomeip/config/esp32/toolchain:esp32 --current-toolchain=//third_party/connectedhomeip/config/esp32/toolchain:esp32 --capture-output --python-dep-list-files gen/third_party/connectedhomeip/src/controller/data_model/data_model_codegen_codegen_metadata_path_list.txt -- ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/codegen.py --generator cpp-app --output-dir gen/third_party/connectedhomeip/src/controller/data_model --expected-outputs gen/third_party/connectedhomeip/src/controller/data_model/data_model_codegen.expected.outputs ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model/controller-clusters.matter
Traceback (most recent call last):
  File "../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py", line 34, in <module>
    from pw_build import gn_resolver
ModuleNotFoundError: No module named 'pw_build'

```

mydelta123, 2023-02-11:
I think i should have a pw_build directory but can't find the class?!?!

the connecthomeip neec compile and output something?
```
(base) xxxx@mac2020-3 pw_build % ls
BUILD.bazel				constraints				linker_script.gni			python_dist.gni
BUILD.gn				copy_from_cipd.gni			mirror_tree.gni				python_gn_args.gni
CMakeCache.txt				defaults.gni				module_config.gni			python_toolchain
CMakeFiles				docs.rst				multi_toolchain_group.gni		python_venv.gni
CMakeLists.txt				error.gni				pigweed.bzl				relative_source_file_names.gni
OWNERS					evaluate_path_expressions.gni		pigweed.cmake				selects.bzl
README.md				exec.gni				platforms				target_config.bzl
bazel_internal				facade.gni				pool					target_types.gni
cc_blob_library.cmake			generated_pigweed_modules_lists.gni	py					test_blob_0123.bin
cc_blob_library.gni			go.gni					python.gni				update_bundle.gni
cc_blob_library_test.cc			hil.gni					python.rst				zip.gni
cc_executable.gni			host_tool.gni				python_action.gni
cc_library.gni				input_group.gni				python_dist
(base) yougangkuo@mac2020-3 pw_build % pwd
/Users/yougangkuo/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/third_party/pigweed/repo/pw_build
(base) xxxxx@mac2020-3 pw_build % 
```

antonio2904, 2023-02-12:
Not sure if it's any different on MacOS. I suggest to try running `source ./export.sh` in esp-idf folder before running `./install.sh ` in esp-matter which installs the required modules in the correct path.

shubhamdp, 2023-02-13:
> I suggest to try running source ./export.sh in esp-idf folder before running ./install.sh  in esp-matter which installs the required modules in the correct path.

Did you try this? You have to source esp-idf/export.sh before running esp-matter/install.sh. Can you please try the following.
```
cd esp-idf
source ./export.sh
cd esp-matter
./install.sh
source ./export.sh
cd light
idf.py build
```

Let me know if this helps.

mydelta123, 2023-02-14:
Try to remove them all and try again
rm -rf .espressif 
rm -rf esp-idf 
rm -rf esp-matter 
brew update ; brew upgrade
**manual remove anaconda**
(remove dirs, remove .zshrc about anaconda, etc)
reboot

fellow the instruction again:
https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#development-setup
everything works!
my anaconda python == 3.6
intel base macbook

shubhamdp, 2023-02-14:
@mydelta123 If its working, can you please close the issue, thanks!

closed 2023-02-15 by dhrishi