espressif/esp-matter #213: Detecting multiple endpoints in Alexa app

antonio2904, 2023-02-11:
Is there a way to register multiple devices in Alexa app? I have 2 endpoints both of which are on_off_plugin_unit. It is controllable from the device console. But only the first endpoint is shown in Alexa app when commissioned.

jonsmirl, 2023-02-14:
Check and see what it does in both Google and Apple Home. Both Google and Apple handle devices with multiple endpoints.  If it works in Google and Apple Home complain to Amazon. If it doesn't work on Goog/Apple, something is not right on your end.

VaishaliAvhale, 2023-02-14:
We checked with Alexa and see the same problem. While other ecosystem apps (Google, Apple) support multiple endpoints.

dhrishi, 2023-02-18:
@antonio2904 Can we close this as this seems to be an issue with a specific commissioner phone app?

closed 2023-02-18 by antonio2904